Koenkk/zigbee2mqtt #25625: Language settings not saved.. Its getting me crazy!

Barry40, 2025-01-06:
### What happened?

Hi, i do a lot of dev work, and for that i need to clean cache and cookies, but the z2m getting me crazy.
Everytime when opening the frontend its in italian, whil i'm a dutchie but i want to have it in english as main, i do NOT want to use other languages. 
I can't find a way to set the language steady to english, and thats getting me crazy :-(
Thanks! 

### What did you expect to happen?

Language should saved by settings and not by a cookie or other...

### How to reproduce it (minimal and precise)

Always...

### Zigbee2MQTT version

 1.42.0 commit: 861cba6

### Adapter firmware version

7.4.4 [GA]

### Adapter

ember

### Setup

Linux

### Debug log

No log

m0wlheld, 2025-01-06:
What timezone do you use on the machine/container running Z2M?

Barry40, 2025-01-06:
Time on server is: TZ Europe/Amsterdam

Barry40, 2025-01-07:
I changed the TZ timezone to Etc/UTC as a test, but that didnt do anything... What to do? 
Thanks

Barry40, 2025-01-09:
Anybody?

closed 2025-03-18 by github-actions[bot]

bmw-biker, 2026-01-17:
Thanks for the confirmation. This is similar to what I guessed.

Yes, I'm using latest z2m with windfront.
And I'll have a look at the repository you proposed.