Koenkk/zigbee2mqtt #25626: [Feature request]: Possibility to bring back the "Permit Join forever" option

daloki83, 2025-01-06:
### Is your feature request related to a problem? Please describe

Some sensors (like water detection (https://www.zigbee2mqtt.io/devices/SJCGQ11LM.html#aqara-sjcgq11lm)) can't find the way back to my network. Only fix was to let that option enabled in V <2.x

### Describe the solution you'd like

As before, let the user decide to let their network open or not. Maybe place a warning or something

### Describe alternatives you've considered

n/a - i have to dismount the inspection shaft in the ceiling and trigger the pairing manually

### Additional context

nope, i think i told you everything :-)

daloki83, 2025-01-06:
Just a few thoughts about this:
- Enabling only via config file + showing warning in the UI
- Option to enable joining (without timelimit) until all known devices have joined.

I just switched back to 1.42.0 to get that device back, but maybe your idea via an automation could also "fix" that (manually triggered the join button for like 2h without success). I'll try that in the next days. But i also got your point to protect users. On the other hand for more experienced users that option shouldn't be eliminated completely. Maybe just hided or modified like one of the above options.

KaSroka, 2025-01-06:
This also breaks the Livolo light switches that I have. They are randomly re-joining the network and without this option they are out of the network after a few days or a power outage (they always re-join after power outage). I don't like the idea of forcing this on users, I know it is not safe and requires proper use of block/pass lists but we should re-introduce it in some way. Special automation every 4mins is not a reliable solution to this problem.

olgru, 2025-01-07:
Please re-active this option. Maybe disabled by default, maybe with an additional security switch, maybe with some more selectable durations; but please - You don't need to babysit the users. Let them decide how secure they need their network... Tec folks decide because they know, none tec folks may not even see the re-implemented option if you keep this away from the gui...

jonnycastaway, 2025-01-07:
Same at my side. Some of the Xiaomi Door/Window Sensors leaving the Network and try to come back with a rejoin.
Without Permit Join forever they falling off the Network and won't rejoin.
As a little security improvement i use the passlist so no other Devices can join the network.
Please, re-add the possibility to permit join forever.

m0wlheld, 2025-01-07:
I think the devices you mention are broken, not Zigbee2MQTT. My Hue, Blitzwolf, Innr and Ikea devices work flawless after initial join. 

I'd vote against permanent permission to join.

jonnycastaway, 2025-01-07:
> I think the devices you mention are broken, not Zigbee2MQTT. My Hue, Blitzwolf, Innr and Ikea devices work flawless after initial join.
> 
> I'd vote against permanent permission to join.

I never said that z2m is broken! It seems the FW of the Devices is the problem. But End-Devices on Battery are mostly not updatable. 
What would you do with 10+ of these devices and this problem? You use a workaround. And permanent Permit Join was the Workaround for years. For little safety with passlist.
And cutting this workaround let many devices behind useless.
So, this is a feature request and not a bug report.

m0wlheld, 2025-01-07:
You could continue to work-around, if you use Home Assistant. The Zigbee2MQTT Bridge Device (should be available in the MQTT integration) does have a "Permit Join" switch (entity id `switch.zigbee2mqtt_bridge_permit_join`) that can be set 'on' by a cron-like automation.

Even without Home Assistant it should be possible to set a cronjob, sending an MQTT message to the appropriate topic (see https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request)

KennethLavrsen, 2025-01-07:
I appreciate your problem and it surprises me that devices are that bad.

The issue with having Zigbee networks with permanent permit join is that you create a problem that I have tried in real life.

My neighbor opposite me had bought a Philips lamp and probably not known what Zigbee is and just use a physical switch and rely on default on behaviour. Each time I put my Zigbee network in Permit join his lamp would join my network if the lamp was on. And then I had to kick it off.

It is not a good thing to do this but OK, if you live far from neighbors the problem is not really there.

Would it be possible and maybe also more effective if the system can be used so that a disconnect is detected and if that happens, an automation turns on permit join which times out in a new minutes. Or will that not always bring the device on the network?

Kenneth

jonnycastaway, 2025-01-07:
> Would it be possible and maybe also more effective if the system can be used so that a disconnect is detected and if that happens, an automation turns on permit join which times out in a new minutes. Or will that not always bring the device on the network? Kenneth

I will try this  in the near future.

mundschenk-at, 2025-01-07:
The technical reasons for the removal where discussed [here](https://github.com/Koenkk/zigbee-herdsman/issues/940).

wizake, 2025-01-08:
I have issues with one Livolo switch that disconnects from the network if its not permanently in 'permit join' mode, si I have an automation that turns on the 'permit join'  for 4 mins  every 15 minutes.
Still with this automation in place, after the 2.0 update, the switch keeps dropping off. I know its probably the switch's fault (  [TI0001-dimmer](https://www.zigbee2mqtt.io/devices/TI0001-dimmer.html#livolo-ti0001-dimmer ), but I still hope for a way of keeping it on the network

realthk, 2025-01-11:
It is not only about devices with broken Zigbee connection, but maintenance: I'm now in the middle of re-pairing all my devices after a coordinator replacement, and it is a huge PITA to always going back to the desktop computer every 4 minutes to re-enable joining.

jeugen1, 2025-01-16:
+1 of keeping this option. I  use Livolo switches too and they not working without this option.

didenkopp, 2025-01-24:
+1 the same livolo devices and the same issue, vote to get back this option.

dkras, 2025-01-26:
+1 for having this option available in expert settings.
Had to revert back to 1.4.2 due to a number of issues with Aqara H1M switches, they misbehave unless permit join is always on.

closed 2025-01-29 by daloki83