Koenkk/zigbee2mqtt #25627: Miss the old Illuminate sensor, the new lux value is not comparable

Typ1er, 2025-01-06:
### What happened?

after renaming to HA you can already see that with the old illuminance sensor you can let the blinds move nicely from 10,000, with the lux value this does not work, here the lux value rises much later (about 60 minutes later)

Is there a way to get the old value back?
[image]

### What did you expect to happen?

Would it be possible to optionally activate the old value and also display it under a new name?

### How to reproduce it (minimal and precise)

it affects all Illuminace sensors

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20220219

### Adapter

zStack SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Add-on on Home Assistant on Proxmox VM

m0wlheld, 2025-01-06:
Question is: Why is the value of former `illuminance_lux` not almost identical to more recent value of `illuminance` (in lx)? The graph for illuminance show values which are about 1/3 of the Lux-values to expect in winterly mid-europe, according to https://de.wikipedia.org/wiki/Beleuchtungsst%C3%A4rke#Beispiele_typischer_Beleuchtungsst%C3%A4rken (German) . And other tickets (https://github.com/Koenkk/zigbee-herdsman-converters/pull/8304#issuecomment-2571535471) show even more deviation.

Koenkk, 2025-01-06:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/25379

> Question is: Why is the value of former illuminance_lux not almost identical to more recent value of illuminance (in lx)?

@m0wlheld this should be the case, `illuminance_lux` has been renamed to `illuminance`.

m0wlheld, 2025-01-06:
> Let's continue in #25379
> 
> > Question is: Why is the value of former illuminance_lux not almost identical to more recent value of illuminance (in lx)?
> 
> @m0wlheld this should be the case, `illuminance_lux` has been renamed to `illuminance`.

I know, but the comparing graphs in https://github.com/Koenkk/zigbee2mqtt/issues/25627#issue-2771073092 show major differences. 

Could be some HA conversion of course.

Koenkk, 2025-01-07:
For me the frontend exposes matched the HA one:

[image]
[image]

Nikolay-Ch, 2025-01-07:
I have the same problem... Old Z2M exports the lux aqara sensor as **sensor_id_lux**
But after updating to version 2.0 this sensor changes its name to **sensor_id**, it turns out the **_lux** postfix is ​​lost...
And I lost the entire history of this sensor in HA. :(

mundschenk-at, 2025-01-07:
@Nikolay-Ch Rename the new sensor entity to the old name (with `_lux`), wait a bit, then rename it back. This should recover the history unless you removed the statistics data via the dev tools already.

m0wlheld, 2025-01-08:
> I have the same problem... Old Z2M exports the lux aqara sensor as **sensor_id_illuminance_lux** But after updating to version 2.0 this sensor was not supported... And I lost the entire history of this sensor in HA. :(

That is not the problem, illuminance_lux has been renamed to illuminance. See #24198 for details. 

But - why do the values of illuminance differ from former illuminance_lux? The name changed, but the value should remain identical.

m0wlheld, 2025-01-08:
> For me the frontend exposes matched the HA one:
> 
> <img alt="Screenshot 2025-01-07 at 20 12 01" width="559" src="https://private-user-images.githubusercontent.com/2892853/400889986-94bf4277-2861-4820-b9e3-5e4b54c9e138.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYzNDE2NDksIm5iZiI6MTczNjM0MTM0OSwicGF0aCI6Ii8yODkyODUzLzQwMDg4OTk4Ni05NGJmNDI3Ny0yODYxLTQ4MjAtYjllMy01ZTRiNTRjOWUxMzgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDhUMTMwMjI5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTZiMGZkYWFmMmY0MTA1NjVkZGQ5N2FiYjFhZDcyOGY4ODc4NmNiNzVlMDc1Mjg4MTRiZGQ0OWVmNWE4NTcxNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.omXciDyhWxor4bEqVHtxvng879-oRc-PQMCqtLYITrU"> <img alt="Screenshot 2025-01-07 at 20 12 29" width="402" src="https://private-user-images.githubusercontent.com/2892853/400890107-432f5c6e-ed8b-4a4f-8bdd-1b49e372c877.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYzNDE2NDksIm5iZiI6MTczNjM0MTM0OSwicGF0aCI6Ii8yODkyODUzLzQwMDg5MDEwNy00MzJmNWM2ZS1lZDhiLTRhNGYtOGJkZC0xYjQ5ZTM3MmM4NzcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDhUMTMwMjI5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjdkZTA0NzNjOGY0OWRhZGMzNmQ4YmEyYjE5NTlmMzE2NDgzOTNhNWQ3NTVmZTU2YTA2ZWFhYWY3NDVjYTkwNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.uqitH54xTfjYKfhVzqoRAtcsIcs2Rx_3lAo5MYRYrkw">

They do (of course). But the values differ. From my understanding the values of former illuminance_lux should be identical to the new illuminances values.

closed 2025-03-17 by github-actions[bot]