Koenkk/zigbee2mqtt #25631: Remotes not sending data to NodeRed

locoroko, 2025-01-06:
### What happened?

After Z2M update, ikea remote's don't publish messages in mqtt. Other devices do. No error messages. 

### Zigbee2MQTT version

2.0.2

### Adapter firmware version

20210708

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle

### Setup

IntelNuc, VM in Unraid. 

### Debug log

[45df7312_zigbee2mqtt_2025-01-06T18-58-50.007Z.log](https://github.com/user-attachments/files/18323472/45df7312_zigbee2mqtt_2025-01-06T18-58-50.007Z.log)

mundschenk-at, 2025-01-06:
If you are using the legacy action sensors, you need to [re-enable them](https://www.zigbee2mqtt.io/guide/configuration/homeassistant.html) explicitly after the update to 2.0 by setting `legacy_action_sensor: true` in the `homeassistant` topic. See https://github.com/Koenkk/zigbee2mqtt/discussions/24198 for more details.

playerfreak, 2025-01-06:
After updating to 2.0. you need to tick this.
[image]
[image]
[image]

ThePatricide, 2025-01-07:
What @playerfreak describes works for me as well. For me it is called 'legacy action sensors' though, on the status topic 'homeassistant/status'. 

I did need to re-pair at least two of my trhee remotes. The battery seemed to be dead too, so it might be a coincidence. I'll keep an eye on this topic, as I have all my remote actions defined in Node-Red, so I might need to change the entity at some point.

closed 2025-01-07 by Koenkk