Koenkk/zigbee2mqtt #25633: Перенос автоматизаций после обновления до версии 2.0

nemen89, 2025-01-06:
### What happened?

Добрый день. После перехода на версию 2.0 перестали работать автоматизации связанные с переходом от sensor к event
В процессе изучения выявилась проблема очень медленного и не корректного (не с первого нажатия) отклика на нажатие кнопки. Особенно если статус события не менялся и нужен повторный запуск.
Автоматизация простая внешняя кнопка без фиксации триггер на запуск таймера обратного отсчета для выключения и включения канала реле в положение ON , и соответственно при повторном нажатии выключение и закрытие таймера.
Перенос корректно не удался. Прошу помощи.  В ветке telegramm много схожих проблем. Не через event, не через GUI адекватно пересобрать с нуля не получилось.

Прилагаю код действующей автоматизации, до перехода на новую версию. Буду благодарен за помощь в переносе кода

`- alias: toilet_switch_fan_on
  initial_state: true
  trigger:
    - platform: state
      entity_id: sensor.switch_vent_action
      attribute: action 
      to: 'left'
  condition: 
    - condition: state
      entity_id: switch.aqara_wireless_relay_vent_l1
      attribute: state_l1 
      state: 'OFF'
  action:
    - service: timer.start
      entity_id: timer.vent_toilet
    - service: switch.turn_on
      entity_id: switch.aqara_wireless_relay_vent_l1
      
- alias: toilet_switch_fan_off
  initial_state: true
  trigger:
    - platform: state
      entity_id: sensor.switch_vent_action
      attribute: action 
      to: 'left'
    - platform: event
      event_type: timer.finished
      event_data:
        entity_id: timer.vent_toilet
  condition: 
    - condition: state
      entity_id: switch.aqara_wireless_relay_vent_l1
      attribute: state_l1 
      state: 'ON'
  action:
    - service: timer.cancel
      entity_id: timer.vent_toilet
    - service: switch.turn_off
      entity_id: switch.aqara_wireless_relay_vent_l1`

### Zigbee2MQTT version

2.0

### Adapter firmware version

20200327

### Adapter

zStack30x

### Setup

Pi 4B, Supervision Docker

m0wlheld, 2025-01-06:
Cannot read your issue text, but the YaML is an Home Assistant automation. Please forward your request to the Home Assistant Community Forum.

eastvan, 2025-01-06:
Как временное решение проблемы предлагаю открыть форточку в туалете :)

m0wlheld, 2025-01-06:
Btw: 

```yaml

trigger:
  - platform: state
    entity_id: sensor.switch_vent_action
    attribute: action
    to: 'left'
```

these „action sensors“ have been removed with Zigbee2MQTT 2.0.0.  Follow the instructions in #24198 how to change to MQTT device triggers

lensherm, 2025-01-07:
> Как временное решение проблемы предлагаю открыть форточку в туалете :)

That's neither very funny, nor helpful when someone's genuinely asking for assistance.

lensherm, 2025-01-07:
Sorry for switching languages here guys and gals. I'll try to help this person out in their language.

В новой версии удаленны action сенсоры, и теперь нужно использовать MQTT триггеры. 

Как посоветовал  @m0wlheld, нужно следовать инструкциям в его ссылке.

Например, у меня другой пульт/выключатель, но смысл тот же:
```yaml
triggers:
  - trigger: mqtt
    topic: zigbee2mqtt/TRADFRI remote/action
    payload: toggle
```

Чтобы узнать что происходит в MQTT брокере с топиками и пейлоадами, очень удобно использовать программу под названием MQTT Explorer
[https://mqtt-explorer.com/](url)

m0wlheld, 2025-01-07:
> Sorry for switching languages here guys and gals. I'll try to help this person out in their language.

Appreciated, but it is still the wrong spot. Maybe you should open a thread at https://community.home-assistant.io/.

mrmaximas, 2025-01-07:
Please write in English. Thanks.

semyonchetvertnyh, 2025-01-24:
@nemen89 вам нужно изменить автоматизацию следующим образом:
```
triggers:
  - trigger: state
    entity_id:
      - event.switch_vent_action # вместо sensor.switch_vent_action
    not_from: unavailable
conditions:
  - condition: template
    value_template: "{{ trigger.to_state.attributes.event_type == 'left' }}" # ранее это был attribute
actions:
...
```

closed 2025-04-03 by github-actions[bot]