Koenkk/zigbee2mqtt #25636: Light status after a total power outage is not correct

KennethLavrsen, 2025-01-06:
### What happened?

After having moved around 100 devices from Deconz to Zigbee2MQTT I decided to turn off all power to my house for around 1 minute to provoke the network to optimize the routing.

All my lights, most of then Philips Hue bulbs, turn on after a power outage as I want it (so you can use old mechanical switches if the smart home dies).

I noticed that more than half the lights were displayed as off even if they were on. This did not happen with Deconz as I remember. There was no problem with all the battery powered sensors. They showed the value as before the power outage.

I would expect the lights to show the right on/off state.

After turning a light on and off they are in sync and shows correct status.

### What did you expect to happen?

I expect mains powered lights to be reported correctly after power comes back. At least after a while. I had a light showing off even if it was on many hours later

### How to reproduce it (minimal and precise)

See above

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

202307xx

### Adapter

Sonoff P

### Setup

Docker installation on Intel NUC

### Debug log

I did not capture log, sorry

KennethLavrsen, 2025-01-06:
One important info I forgot. The computer running Zigbee2MQTT is on a UPS so it did not reset.

I just tried with single lights.

And if a light is off, and then I turn off the wallswitch and turn it on again, the lights turn on, as they should, but the state of the light remains off in the Zigbee2MQTT dashboard.

So maybe this is an impossible thing to fix as I cannot tell if the Philips bulbs sends a status message when it wakes up and polling all lights all the time would be silly and destroy the network.

I may remember the Deconz behavior wrong.

KennethLavrsen, 2025-01-06:
More updates. Some light do update when you repower them. Some don't. Seems to be firmware dependent.

m0wlheld, 2025-01-07:
> One important info I forgot. The computer running Zigbee2MQTT is on a UPS so it did not reset.

I don‘t see how the network should optimize, if you leave the coordinator on, but power off any client device. Besides - the network continuously optimises, not only once.

closed 2025-03-16 by github-actions[bot]