Koenkk/zigbee2mqtt #25637: Z2M 2.0.0 fails to start (Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR)

ialdalaimi, 2025-01-06:
### What happened?

> First of all, I want to express my gratitude to the contributors of this repository for their incredible work. The effort you've put into developing and maintaining this project is truly commendable, and it’s clear that a lot of thought and expertise have gone into it.

### Background Info
After upgrading Zigbee2MQTT (Z2M) to version `2.0.0` using the SONOFF Dongle-P, I encountered consistantly crashes on startup. After an evening of debugging, I decided to:

1. Delete the `coordinator_backup.json` and `state.json` files.
2. Clean up the `configuration.json` file.
3. Switch to my new SkyConnect (ZBT-1) USB dongle, as I would need to rejoin all my devices anyway.

I flashed the SkyConnect dongle with firmware version `7.4.0` using the following file: [NabuCasa_SkyConnect_EZSP_v7.4.0.0_ncp-uart-hw_115200.gbl](https://github.com/NabuCasa/silabs-firmware/blob/main/EmberZNet/beta/NabuCasa_SkyConnect_EZSP_v7.4.0.0_ncp-uart-hw_115200.gbl). After updating the adapter setting to `ember`, I successfully started Z2M version `2.0.0` and reconnected approximately 25 devices. Everything appeared to be working perfectly (yay!).

### Issue
After making some adjustments to settings, I restarted my Intel NUC, which runs Z2M, Mosquitto, and Home Assistant in a Docker Compose setup. Upon restarting, Z2M failed to start and logged the following error:

```log
Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR
```

The last configuration change I remember making was updating the `mqtt base_topic` through the frontend UI. Before restarting, I reverted this change to its original setting in the `configuration.yaml` file. However, I don't believe this is related to the error.

### Versions
- **SkyConnect**: `7.4.0`
- **Z2M**: `2.0.0`

---

### Z2M Logs (with debug logging enabled)
```log
Starting Zigbee2MQTT without watchdog.
[2025-01-06 22:32:53] debug: 	z2m: Removing old log directory '/app/data/log/2025-01-06.22-30-21'
[2025-01-06 22:32:53] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-06 22:32:53] debug: 	z2m: Loaded state from file /app/data/state.json
[2025-01-06 22:32:53] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a)
[2025-01-06 22:32:53] debug: 	z2m: sd-notify loaded
[2025-01-06 22:32:53] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-06 22:32:53] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"rtscts\":true,\"path\":\"/dev/ttyUSB1\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-06 22:32:53] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS10"},{"path":"/dev/ttyS11"},{"path":"/dev/ttyS12"},{"path":"/dev/ttyS13"},{"path":"/dev/ttyS14"},{"path":"/dev/ttyS15"},{"path":"/dev/ttyS16"},{"path":"/dev/ttyS17"},{"path":"/dev/ttyS18"},{"path":"/dev/ttyS19"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS20"},{"path":"/dev/ttyS21"},{"path":"/dev/ttyS22"},{"path":"/dev/ttyS23"},{"path":"/dev/ttyS24"},{"path":"/dev/ttyS25"},{"path":"/dev/ttyS26"},{"path":"/dev/ttyS27"},{"path":"/dev/ttyS28"},{"path":"/dev/ttyS29"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyS30"},{"path":"/dev/ttyS31"},{"path":"/dev/ttyS4"},{"path":"/dev/ttyS5"},{"path":"/dev/ttyS6"},{"path":"/dev/ttyS7"},{"path":"/dev/ttyS8"},{"path":"/dev/ttyS9"},{"path":"/dev/ttyUSB0","manufacturer":"Nabu Casa","serialNumber":"5ad25cea513aef1192e2321455516304","pnpId":"usb-Nabu_Casa_Home_Assistant_Connect_ZBT-1_5ad25cea513aef1192e2321455516304-if00-port0","vendorId":"10c4","productId":"ea60"},{"path":"/dev/ttyUSB1","manufacturer":"FTDI","serialNumber":"A19J3WB5","pnpId":"usb-FTDI_FT232R_USB_UART_A19J3WB5-if00-port0","vendorId":"0403","productId":"6001"}]
[2025-01-06 22:32:53] debug: 	zh:adapter:discovery: Unable to match USB adapter: ember | /dev/ttyUSB1
[2025-01-06 22:32:53] info: 	zh:ember: Using default stack config.
[2025-01-06 22:32:53] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"rtscts":true,"path":"/dev/ttyUSB1","adapter":"ember"},"adapter":{"disableLED":false},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}'
[2025-01-06 22:32:53] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-06 22:32:53] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-06 22:32:53] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-06 22:32:53] debug: 	zh:ember:uart:ash: Opening serial port with {"path":"/dev/ttyUSB1","baudRate":115200,"rtscts":true,"autoOpen":false,"parity":"none","stopBits":1,"xon":false,"xoff":false}
[2025-01-06 22:32:53] info: 	zh:ember:uart:ash: Serial port opened
[2025-01-06 22:32:53] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-06 22:32:53] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2025-01-06 22:32:53] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2025-01-06 22:32:53] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2025-01-06 22:32:53] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2025-01-06 22:32:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2025-01-06 22:32:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2025-01-06 22:32:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2025-01-06 22:32:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2025-01-06 22:32:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2025-01-06 22:32:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2025-01-06 22:32:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2025-01-06 22:32:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2025-01-06 22:32:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1100/2500
[2025-01-06 22:32:54] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2025-01-06 22:32:55] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
[2025-01-06 22:32:55] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
[2025-01-06 22:32:55] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
[2025-01-06 22:32:55] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
[2025-01-06 22:32:55] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
[2025-01-06 22:32:55] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
[2025-01-06 22:32:55] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
[2025-01-06 22:32:55] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
[... 401 lines cut ...]
[2025-01-06 22:33:05] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2025-01-06 22:33:05] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1300/2500
[2025-01-06 22:33:05] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1400/2500
[2025-01-06 22:33:05] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1500/2500
[2025-01-06 22:33:05] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1600/2500
[2025-01-06 22:33:05] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1700/2500
[2025-01-06 22:33:05] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1800/2500
[2025-01-06 22:33:05] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1900/2500
[2025-01-06 22:33:05] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2000/2500
[2025-01-06 22:33:06] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2100/2500
[2025-01-06 22:33:06] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2200/2500
[2025-01-06 22:33:06] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2300/2500
[2025-01-06 22:33:06] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 2400/2500
[2025-01-06 22:33:06] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-06 22:33:06] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-06 22:33:06] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-06 22:33:06] error: 	z2m: Exiting...
[... cut ...]
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
```

svenstaro, 2025-01-07:
I'm using the Sonoff and I had the same problem. I was also using `adapter: ember` but then switched to `adapter: zstack` and that seems to be working just fine.

RetepNamenots, 2025-01-07:
> I'm using the Sonoff and I had the same problem. I was also using `adapter: ember` but then switched to `adapter: zstack` and that seems to be working just fine.

I hadn't specified an adapter in my config, but I can confirm that adding the line `adapter: zstack` resolved the issue for me.

ialdalaimi, 2025-01-07:
I solved my problem by deleting the following files:

1. Delete the `database.db`, `coordinator_backup.json` and `state.json` files.
2. Clean up the configuration.json file.
3. Deleted the `zigbee2mqtt` topic in MQTT (using MQTT Explorer)
4. Restart my Intel Nuc

closed 2025-01-07 by ialdalaimi

f22raptorroland, 2025-04-16:
Just to document it here:

I have the hacked [Lidl / Silvercrest gateway](https://github.com/Koenkk/zigbee2mqtt/discussions/22520#discussioncomment-12828203), and got this exact same error repeatedly.

Restarting the gateway fixed the issue.