Koenkk/zigbee2mqtt #25640: Auara WXKG11LM button has no action in state

steku, 2025-01-07:
### What happened?

Aquara WXKG11LM button has no action in its state. The payload is sent as follows

{
    "battery": 100,
    "device_temperature": 27,
    "last_seen": "2025-01-06T21:07:37-05:00",
    "linkquality": 163,
    "power": 1,
    "power_outage_count": 48,
    "voltage": 3005
}

### What did you expect to happen?

I have 4 of these buttons and all have lost the action in the state. I believe this to be the result for upgrading the HA Addon to the lastest build 2.0.0-2

### How to reproduce it (minimal and precise)

I have 4 of these buttons and all have no longer have the action in the state since my upgrade 

### Zigbee2MQTT version

 2.0.0-2

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus, TI CC2652P + CP2102N Zigbee

### Setup

Add-on,  x86-64

hellboyfishman, 2025-01-07:
I'm having the same issue with a SLZB-06P7 adapter on firmware 20240716.   

All 5 of my buttons no longer have an action state entity exposed.   I've tried resetting a couple of them which will then show a double, single, or hold action within the Z2M add-on, but I will do not have an action state entity in home assistant for them.

m0wlheld, 2025-01-07:
Close issue, please?

johnjoemorgan, 2025-01-07:
@steku cheers mate - at last someone explained it simply and succinctly.I have a whole load of different switches using "action" and this is the only post/info that actually explains how to get the problem fixed for all types of sensors.  Not to fix the illum/lux sensor problem.

m0wlheld, 2025-01-07:
> @steku cheers mate - at last someone explained it simply and succinctly.I have a whole load of different switches using "action" and this is the only post/info that actually explains how to get the problem fixed for all types of sensors. Not to fix the illum/lux sensor problem.

It's all in the 2.0 release notes and the referenced breaking changes (see #24198). Including instructions how to migrate.

I-G-1-1, 2025-01-07:
I use .yaml to configure my automation.
Following zigbee2mqtt docs/wiki it seemed I had to change all my automation involving buttons in this way:

```
    triggers:
      - trigger: device
        domain: mqtt
        device_id: b34dd512c426ef17f15edb6ec2f31b1c
        type: action
        subtype: 'single_left'
```

but i really don't like the use of "device_id", because is a uuid generated by HomeAssistant for each button that I don't know how to find.

So thanks to [hellboyfishman](https://github.com/hellboyfishman), now I can simply use the name I gave to the button in zigbee2mqtt:
```
    triggers:
      - trigger: mqtt
        topic: zigbee2mqtt/wall_switch_camera/action
        payload: 'single_left'
```

> It's all in the 2.0 release notes and the referenced breaking changes (see #24198). Including instructions how to migrate.

I didn't find the instruction on how to migrate my automation in that topic, so again thanks to [hellboyfishman](https://github.com/hellboyfishman)

steku, 2025-01-07:
> It's all in the 2.0 release notes and the referenced breaking changes (see #24198). Including instructions how to migrate.

This feels like a huge step backwards in usability for non-techie people and a PITA to find the IDs for the triggers, compared to just using an entity in HA.

https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended

Jinra, 2025-01-07:
> For anyone who finds this: The easiest way ~to replicate the pre 2.X breaking change way~ of creating a trigger via GUI is to follow [this](https://community.home-assistant.io/t/aqara-buttons-all-in-one-zigbee2mqtt/268717/45?u=drafted)

This is how I originally created the triggers in the GUI, but now I see they're all malformed and broken as shown below. However, recreating them seems to work.

[image]

Koenkk, 2025-01-07:
This is one of the Zigbee2MQTT 2.0.0 breaking changes, from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

> - All action sensors are now disabled by default (`sensor.*_action` entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your `configuration.yaml`.
>   ```yaml
>   homeassistant:
>     legacy_action_sensor: true
>   ```

closed 2025-01-07 by Koenkk

Bastiencc, 2025-01-10:
Hi !

I can't convert this automation to 2.0, can you help me?

```
- id: chambre_amis_switch_mur_on_press_on
  triggers:
    - trigger: mqtt
      topic: 'zigbee2mqtt/chambre_amis_switch_chevet'
  conditions:
    - condition: or
      conditions:
        - condition: template
          value_template: '{{ "on-press" == trigger.payload_json.action }}'
        - condition: template
          value_template: '{{ "off-press" == trigger.payload_json.action }}'
    - condition: template
      value_template: '{{ 1 == trigger.payload_json.counter|int }}'
  actions:
    (...)
```

I-G-1-1, 2025-01-10:
> can you help me?

I think that this should work:

```
- id: chambre_amis_switch_mur_on_press_on
  triggers:
    - trigger: mqtt
      topic: 'zigbee2mqtt/chambre_amis_switch_chevet/action'
      payload: 'on-press'
  actions:
    (...)
```

If it's not working try to change `payload: 'on-press'` whit the correct payload published by your device

I-G-1-1, 2025-01-10:
> and about this ?

I don't know if there is a better and cleaner way, but personally I would create a mqtt sensor and then use that sensor in the condition:

for the sensor:
```
mqtt:
  sensor:
    - name: "chambre_amis_switch_mur_counter"
      unique_id: bb0c5bd6110d41968f986ad6b1c57c87
      state_topic: "zigbee2mqtt/chambre_amis_switch_chevet"
      value_template: "{{ value_json.counter | lower }}"
```
for the automotaion:
```      
- id: chambre_amis_switch_mur_on_press_on
  triggers:
    - trigger: mqtt
      topic: 'zigbee2mqtt/chambre_amis_switch_chevet/action'
      payload: 'on-press'
  conditions:
    - condition: template
      value_template: '{{ 1 == states.sensor.chambre_amis_switch_mur_counter.state | int }}'      
  actions:
    (...)
```

Bastiencc, 2025-01-10:
> I don't know if there is a better and cleaner way, but personally I would create a mqtt sensor and then use that sensor in the condition:

the sensor works well in 1.4 but no more in 2.0 :(

I-G-1-1, 2025-01-10:
> seems that there is not more access to this topic

maybe you could check with [MQTT-Explore](http://mqtt-explorer.com) if/what topic is publishing zigbee2mqtt v2.0

Bastiencc, 2025-01-10:
> > seems that there is not more access to this topic
> 
> maybe you could check with [MQTT-Explore](http://mqtt-explorer.com) if/what topic is published by zigbee2mqtt v2.0

yeah... I checked, and the problem is that counter is not more updated in 2.0

I-G-1-1, 2025-01-10:
> yeah... I checked, and the problem is that counter is not more updated in 2.0

try to use experimantal [event-entity](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-home-assistant-event-entity-experimental) instead of [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended).

You have to enable it in zigbee2mqtt configuration (see [here](https://www.zigbee2mqtt.io/guide/configuration/homeassistant.html)): 
```
experimental_event_entities: true
```

ellisjm74, 2025-01-27:
> I use .yaml to configure my automation. Following zigbee2mqtt docs/wiki it seemed I had to change all my automation involving buttons in this way:
>
> ```
>     triggers:
>       - trigger: device [...]

BIG THANKS to @I-G-1-1 & @hellboyfishman for this solve. Simple, doesn't require a blueprint, can be done with YAML or GUI, and doesn't require turning any legacy options on.  I agree with the disdain for device_id, would much rather have the device name, which this does.

BTW: GUI-wise it's Add Trigger/Other Triggers/MQTT.  Topic is zigbee2mqtt/_name of device in Z2M_/action, Payload is single (or whatever you used before z2m 2.0 for the action)

Here's what I ended up with for my Aqara WXKG11LM (yes, I know the name has a space in it and uses caps, it's going to be OK, I promise):
```
  - trigger: mqtt
    topic: zigbee2mqtt/Switch Kitchen/action
    payload: single
```