Koenkk/zigbee2mqtt #25641: Zigbee2MQTT does not start after Update 2.0.0-2

AndreL79, 2025-01-07:
### What happened?

After Update to Version 2.0.0-2 Zigbee2MQTT does not start

Z2M:
Modell: | SLZB-06 |  
                                  | Radio           | Kern
CPU-Modell:              | CC2652P     | ESP32-D0WD-V3
CPU-Kerne und Takt: | 1 @ 48Mhz | 2 @ 240 Mhz
Flash-Speicher:          | 352 kB        | 16Mb
RAM-Speicher:          | 88 Kb          | 119/292KiB
SoC-Temperatur:       | 88.15°C       | 90.56°C
Firmware Version:      | 20240710    | v2.5.8
Aktueller FW-Check:  | 12/11/2024 | 12/11/2024

Config:
socat:
enabled: false
master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
options: "-d -d"
log: false

serial: 
port: tcp://192.168.178.35:6638
baudrate: 115200
disable_led: false
advanced:
  transmit_power: 20

Backup Restore to Version 1.42.0-2, its works fine

Sorry I'm new in HA World, i don't know what can i do

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

Proxmox VM on Dell x64

### Debug log

`
[2025-01-07 05:15:46] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-07 05:15:46] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-07 05:15:46] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-07 05:15:46] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-07 05:15:46] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-07 05:15:46] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-07 05:15:46] error: 	z2m: Exiting...
[2025-01-07 05:15:46] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
`

djurny, 2025-01-07:
Hi @AndreL79,
You write you have a SLZB-06? If so, you need to add the `adapter: zstack` setting in zigbee2mqtt's `configuration.yaml`. 

Try with the following content:
```
serial:
  port: tcp://192.168.178.35:6638
  baudrate: 115200
  disable_led: false
  adapter: zstack
advanced:
  transmit_power: 20
```
See the following topics for more info:
- https://github.com/Koenkk/zigbee2mqtt/discussions/24364#discussion-7328423
- https://github.com/Koenkk/zigbee2mqtt/discussions/24198

Groetjes,

AndreL79, 2025-01-07:
Hi @djurny 
thanks a lot for your quick help.
Now its running

Happy new Year

closed 2025-01-07 by AndreL79