Koenkk/zigbee2mqtt #25644: Problems with external converters

customstudioyandexru, 2025-01-07:
### What happened?

I have Z2M 2.0 installed as an integration, I created a folder
/homeassistant/zigbee2mqtt/external_converters/
I put the external converters in there. There are a lot of errors in the logs. Contacted the device developers. They fixed
the converters and I updated them. Now there are no errors in the logs, but the devices are not working. Devices are not supported.

### What did you expect to happen?

I thought the updates would fix the problems, not add them.

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

1000

### Adapter

DarkStar lan

### Setup

ordinary

### Debug log

[13:42:34] INFO: Preparing to start...
[13:42:34] INFO: Socat not enabled
[13:42:34] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.

m0wlheld, 2025-01-07:
Duplicate of #25510 , please close this issue and follow instructions in the referenced issue.

customstudioyandexru, 2025-01-07:
> Дубликат #25510, пожалуйста, закройте эту проблему и следуйте инструкциям в указанной проблеме.
I followed exactly this instruction, but it doesn't work!
I was creating folders
addons_config/45df7312_zigbee2mqtt/zigbee2mqtt/external_converters
/homeassistant/zigbee2mqtt/external_converters/
/homeassistant/zigbee2mqtt/data/external_converters/
it doesn't help with my problem. I have been rebooting the Home Assistant and Z2m

m0wlheld, 2025-01-07:
I think the external_converters folder must be created in the same directory as the configuration.yaml file.

customstudioyandexru, 2025-01-08:
> I think the external_converters folder must be created in the same directory as the configuration.yaml file.

I also tried the root folder.

dhc25, 2025-01-08:
I have the same problem as @customstudioyandexru 
I have tried all the folder paths, and cannot get external converters to load.

closed 2025-01-08 by customstudioyandexru

John-Rookie, 2025-01-31:
I have exactly the same problem. Is there a solution yet?