Koenkk/zigbee2mqtt #25646: SONOFF ZBDongle-E with Router Firmware - z2m does not start

kickartz, 2025-01-07:
### What happened?

I wanted to install Zigbee2MQTT on my Home Assistant. I read on this [site](https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/Router) that I first need to flash the router firmware via the browser application. This worked, but after that, I encountered the following errors when starting Zigbee2MQTT: once with the ezsp driver and once with the z-stack driver. Then, I tried to flash the newest firmware to my Sonoff stick, but the browser application does not retrieve the current firmware information and fails. What can I do?

ezsp: `[11:51:01] INFO: Preparing to start...
[11:51:01] INFO: Socat not enabled
[11:51:02] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-07 11:51:13] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-07 11:51:13] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-07 11:51:13] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-07 11:51:13] warning: 	zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2025-01-07 11:51:23] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":-1} after 10000ms
[2025-01-07 11:51:23] error: 	zh:ezsp:ezsp: Connection attempt 1 error: Error: Failure to connect
[2025-01-07 11:51:38] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":-1} after 10000ms
[2025-01-07 11:51:38] error: 	zh:ezsp:ezsp: Connection attempt 2 error: Error: Failure to connect
[2025-01-07 11:51:58] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":-1} after 10000ms
[2025-01-07 11:51:58] error: 	zh:ezsp:ezsp: Connection attempt 3 error: Error: Failure to connect
[2025-01-07 11:52:03] error: 	z2m: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'close')
    at /app/lib/extension/frontend.ts:132:52
    at new Promise (<anonymous>)
    at Frontend.stop (/app/lib/extension/frontend.ts:132:15)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.stop (/app/lib/controller.ts:233:9)
    at stop (/app/index.js:172:5)
    at process.handleQuit (/app/index.js:180:13))
[2025-01-07 11:52:03] error: 	z2m: Not connected to MQTT server!
[2025-01-07 11:52:03] error: 	z2m: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: '{"state":"offline"}
[2025-01-07 11:52:03] info: 	z2m: Disconnecting from MQTT server
[2025-01-07 11:52:03] info: 	z2m: Stopping zigbee-herdsman...
[2025-01-07 11:52:03] info: 	z2m: Stopped zigbee-herdsman
[2025-01-07 11:52:03] info: 	z2m: Stopped Zigbee2MQTT`

zstack `[12:15:45] INFO: Preparing to start...
[12:15:45] INFO: Socat not enabled
[12:15:45] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-07 12:15:49] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-07 12:15:49] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-07 12:15:49] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-07 12:15:49] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_ae017e628a12ef11868075b8bf9df066-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-01-07 12:15:49] info: 	zh:zstack:znp: Serialport opened
[2025-01-07 12:15:49] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-07 12:15:50] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-01-07 12:16:09] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-07 12:16:09] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-07 12:16:09] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-07 12:16:09] error: 	z2m: Exiting...
[2025-01-07 12:16:09] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)`

### What did you expect to happen?

I want to use Z2M

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

Z3RouterUSBDonlge_EZNet6.10.3_V1.0.0.gbl

### Adapter

SONOFF ZBDongle-E

### Setup

HomeAssistant OS in Proxmox, USB Passthrought full Port

clockbrain, 2025-01-07:
You need the coordinator firmware (NCP), not the router firmware. See instructions here https://www.zigbee2mqtt.io/guide/adapters/emberznet.html

cuipito, 2025-01-07:
i see that you are still using the ezsp driver, which is only supported for firmware 7.4.4.0 and lower.
it could be very well that the itead firmware software is higher then the ezsp driver can support ( i could be wrong as i am not familiar with itead firmware versions )

My recommendation try to flash the firmware manually. 

https://www.zigbee2mqtt.io/devices/ZBDongle-E.html

Download a tool that supports Xmodem like "secureCRT"
https://www.vandyke.com/cgi-bin/releases.php?product=securecrt

Download the proper firmware from darkxst

https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee

I would recommend to try version 7.4.4.0 first as it seems the latest firmware 8.0.0 and 8.0.1 have some issues in them.

ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl

and switch your stack to ember in your z2m
```

serial:
    adapter: ember

```

kickartz, 2025-01-07:
> well that the itead firmware software is higher then the ezsp driver c

i tried it with the Silabs Flasher Addon. What am i doing wrong?
-----------------------------------------------------------
 Add-on: Silicon Labs Flasher
 Silicon Labs firmware flasher add-on
-----------------------------------------------------------
 Add-on version: 0.3.2
 You are running the latest version of this add-on.
 System: Home Assistant OS 14.1  (amd64 / qemux86-64)
 Home Assistant Core: 2025.1.1
 Home Assistant Supervisor: 2024.12.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service banner successfully started
s6-rc: info: service universal-silabs-flasher: starting
[13:12:20] INFO: Starting universal-silabs-flasher with /dev/ttyUSB0 (bootloader baudrate 115200)
2025-01-07 13:12:21.963 core-silabs-flasher universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=2, sdk_version='2024.6.2', ezsp_version='8.0.2.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.ZIGBEE_NCP: 'zigbee_ncp'>, fw_variant='sw_flow', baudrate=115200)
2025-01-07 13:12:21.964 core-silabs-flasher universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-01-07 13:12:23.979 core-silabs-flasher universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2025-01-07 13:12:28.995 core-silabs-flasher universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2025-01-07 13:12:33.319 core-silabs-flasher universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 115200 baud
2025-01-07 13:12:37.648 core-silabs-flasher universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 230400 baud
2025-01-07 13:12:41.980 core-silabs-flasher universal_silabs_flasher.flasher INFO Probing ApplicationType.SPINEL at 460800 baud
Error: Failed to probe running application type
[13:12:46] INFO: universal-silabs-flasher-up script exited with code 1
s6-rc: warning: unable to start service universal-silabs-flasher: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service banner: stopping
s6-rc: info: service banner successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

m0wlheld, 2025-01-07:
```yaml
serial:
    adapter: ember
```

No need for a screwdriver, just set this in your `configuration.yaml`.

m0wlheld, 2025-01-07:
> Anyways, adding devices is giving me headache. the adapter crashes often and causes z2m to reboot

[image]

(Source: https://www.zigbee2mqtt.io/guide/getting-started/)

WombatHollow, 2025-02-26:
I have a similar problem, Z2M crashes the RasPi (requires power removal to restart) when I run it using either as a systemd service or `pnpm start`.    But if I start the Raspi, then turn off USB ports, run `sudo systemctl start zigbee2mqtt.service` then turn on the USB ports it runs.  After that I can stop and start z2m service without turning off USB, until the Raspi is rebooted or power cycled.
To help verify and find a workaround I created a simple clean Raspi set up from scratch to be sure no other programs caused issues.
### System Configuration
- RasPi 3B+, new 128GB SD card, 4.8A available USB power (draws 0.27A, peaks up to 0.51A)
- OS: Bookworm Raspberry Pi OS Lite up to date
- z2M: v2.1.1 commit 1accb8b
- Herdsman: 3.2.5
- Herdsman Converters : 21.31.0
- z2m frontend : 0.9.4
- Coordinator model : {"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20220714170928","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220714170928-if00","vendorId":"1a86","productId":"55d4"} => ember: 4
- Coordinator type: EmberZNet
- Coordinator revision : 7.4.4 [GA] (installed using [Silabs Firmware Flasher](https://darkxst.github.io/silabs-firmware-builder/))
no other software installed other than that required for Z2M.

Z2M installed as per the Linux instructions at zigbee2MQTT.io but when I run using pnpm (or npm) it produces this output and the Raspi crashes requiring power cycling to restart.  The output was as follows
`pi@raspPi2:~ $ cd /opt/zigbee2mqtt
pi@raspPi2:/opt/zigbee2mqtt $ pnpm start

> zigbee2mqtt@2.1.1 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.
[2025-02-26 14:41:05] info:     z2m: Logging to console, file (filename: log.log)
[2025-02-26 14:41:05] info:     z2m: Starting Zigbee2MQTT version 2.1.1 (commit #1accb8b)
[2025-02-26 14:41:05] info:     z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-26 14:41:06] info:     zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20220714170928","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220714170928-if00","vendorId":"1a86","productId":"55d4"} => ember: 4
[2025-02-26 14:41:06] info:     zh:ember: Using default stack config.
[2025-02-26 14:41:06] info:     zh:ember: ======== Ember Adapter Starting ========
[2025-02-26 14:41:06] info:     zh:ember:ezsp: ======== EZSP starting ========
[2025-02-26 14:41:06] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-02-26 14:41:06] info:     zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
` 
The front end has not started, no MQTT messages are sent and no log.log file is created, you cannot logon remotely and the Raspi does not respond to ping.

Likewise if I run it as a systemd service `sudo systemctl start zigbee2mqtt.service` it will crash.  If I enable the systemd service to run on startup, it crashes on startup.  But the Raspi will start if you unplug the ZBDongle-E.  If z2M is not enabled as a systemd service the RasPi will start even with the ZBDongle plugged in.

### The Workaround
 
I have found a workaround that will allow Z2M and this dongle to run.  

1. disable zigbee2MQTT.service so it does not start on reboot or power up
2. On powerup or restart disable USB Ports using `echo '1-1' | sudo tee /sys/bus/usb/drivers/usb/unbind`
3. then start zigbee2MQTT.service using `sudo systemctl start zigbee2mqtt.service` , it will return an error
4. then enable the USB ports using `sudo tee /sys/bus/usb/drivers/usb/bind`
and z2M will then startup and run.  Here is the output:

`Linux raspPi2 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-0                                                             8) aarch64
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Feb 26 14:57:06 2025 from 10.0.0.164
pi@raspPi2:~ $ systemctl status zigbee2mqtt.service
○ zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; disabled; preset: enabled)
     Active: inactive (dead)
pi@raspPi2:~ $ echo '1-1' | sudo tee /sys/bus/usb/drivers/usb/unbind
1-1
pi@raspPi2:~ $ sudo systemctl start zigbee2mqtt.service
Job for zigbee2mqtt.service failed because the control process exited with error code.
See "systemctl status zigbee2mqtt.service" and "journalctl -xeu zigbee2mqtt.service" for details.
pi@raspPi2:~ $ echo '1-1' | sudo tee /sys/bus/usb/drivers/usb/bind
1-1
pi@raspPi2:~ $ systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; disabled; preset: enabled)
     Active: activating (start) since Wed 2025-02-26 14:59:14 AEDT; 11s ago
   Main PID: 624 (node)
      Tasks: 12 (limit: 767)
        CPU: 16.501s
     CGroup: /system.slice/zigbee2mqtt.service
             ├─624 /usr/bin/node index.js
             └─660 udevadm info -e

Feb 26 14:59:14 raspPi2 systemd[1]: Starting zigbee2mqtt.service - zigbee2mqtt...
pi@raspPi2:~ $ systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; disabled; preset: enabled)
     Active: active (running) since Wed 2025-02-26 14:59:29 AEDT; 9s ago
   Main PID: 624 (node)
      Tasks: 11 (limit: 767)
        CPU: 19.873s
     CGroup: /system.slice/zigbee2mqtt.service
             └─624 /usr/bin/node index.js

Feb 26 14:59:14 raspPi2 systemd[1]: Starting zigbee2mqtt.service - zigbee2mqtt...
Feb 26 14:59:29 raspPi2 systemd[1]: Started zigbee2mqtt.service - zigbee2mqtt.
pi@raspPi2:~ $
`

At least this workaround can be implemented in a batch file run on startup, or run by Node-red on Raspi start

[configuration.yaml.txt](https://github.com/user-attachments/files/18976977/configuration.yaml.txt)

closed 2025-05-05 by github-actions[bot]