Koenkk/zigbee2mqtt #25654: Yokis MTV300EUP not detected when trying to pair

LMProduction67, 2025-01-07:
### What happened?

Hello everyone,

I recently purchased a Sonoff Zigbee 3.0 Dongle-P (CC2652P) and noticed that the MTV300EUP from yokis is now supported by Z2M.

I’ve installed the Home Assistant add-on with the MQTT broker, and my Zigbee2MQTT setup is running.

However, I’m unable to detect my device. I’ve unlocked it with 23 taps and set it to pairing mode, but it doesn’t appear in Z2M.

Below are the Z2M logs for reference:

`[17:19:38] INFO: Preparing to start...
[17:19:38] INFO: Socat not enabled
[17:19:38] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-07 17:19:39] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-07 17:19:39] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-07 17:19:39] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-07 17:19:39] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0a16d2ff5b52ef1182bc3253cc32aab1-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-01-07 17:19:39] info: 	zh:zstack:znp: Serialport opened
[2025-01-07 17:19:39] info: 	z2m: zigbee-herdsman started (resumed)
[2025-01-07 17:19:39] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
[2025-01-07 17:19:39] info: 	z2m: Currently 0 devices are joined.
[2025-01-07 17:19:39] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-01-07 17:19:39] info: 	z2m: Connected to MQTT server
[2025-01-07 17:19:39] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-01-07 17:19:39] info: 	z2m: Started frontend on port 8099
[2025-01-07 17:19:39] info: 	z2m: Zigbee2MQTT started!
[2025-01-07 17:19:44] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x00124b002f89b786/connection_state/config', payload '{"device":{"hw_version":"zStack3x0 20210708","identifiers":["zigbee2mqtt_bridge_0x00124b002f89b786"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"2.0.0"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"2.0.0","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0x00124b002f89b786_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
[2025-01-07 17:20:25] info: 	z2m: Zigbee: allowing new devices to join.
[2025-01-07 17:20:26] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"lcx1a-1"}'`
Let me know if you'd like me to adjust anything!

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee Dongle P CC2652P chip

### Setup

add-on

mrmaximas, 2025-01-07:
I highly recommend to flash this firmware https://github.com/Koenkk/Z-Stack-firmware

LMProduction67, 2025-01-07:
Thanks for your reply. I am with the latest firmware now (zStack3x0 version 20240710) but still not working.

LaurentChardin, 2025-02-02:
Hi

On my MTV, i didn't need to press the 23 times in order to pair it. There are several ways to activate the pairing process.
But i highly recommend you to update the firmware to the latest version using BLE: at the moment, OTA is not supported so you want to use their mobile app for that (the app is UP2PRO on ios).

Then make sure the pairing process is activated (blinking light). For me, i just needed to press the push-button once after powering up to start the pairing process (at power up, the light is permanent red, meaning no network configured).

Also, power cycling 5 times the device will reset the configuration.

(FYI i am the one that created the Yokis definition for Z2M, with support from the Yokis team).

Best,

closed 2025-04-11 by github-actions[bot]