Koenkk/zigbee2mqtt #25656: Zigbee2MQTT not detecting Ewelink Zigbee E27 15W Bulb (Model: C3009)

megustaeliphone, 2025-01-07:
### What happened?

I have a Ewelink Zigbee E27 15W Smart LED Bulb (Model: C3009), but it is not being detected by Zigbee2MQTT. The bulb is powered on and functioning normally, but it doesn't show up in Zigbee2MQTT's device list.
Although the box mentions "WiFi," the specifications printed on the packaging clearly indicate support for Zigbee. Additionally, the bulb itself has "Zigbee" written on it, which confirms that it should be compatible with a Zigbee network. I have attached photos of the box and the bulb for reference.
[image]
[image]
[image]
[image]

### What did you expect to happen?

I expected the Ewelink Zigbee E27 15W Smart LED Bulb (Model: C3009) to be detected automatically by Zigbee2MQTT and show up in the device list once powered on and in pairing mode.

### How to reproduce it (minimal and precise)

Power on the Ewelink Zigbee E27 15W Smart LED Bulb (Model: C3009).
Put the bulb into pairing mode (usually by turning it on and off multiple times).
Open Zigbee2MQTT and check the device list.
The bulb does not appear in the list, despite being powered on and in pairing mode.

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20210708

### Adapter

Sonoff zigbee 3.0 usb dongle plus

### Setup

Add-on on Home Assistant OS

### Debug log

There are no relevant entries in the debug logs because the bulb is not detected at all by Zigbee2MQTT. I checked the logs while attempting to pair the device, but no messages or errors appear.

m0wlheld, 2025-01-07:
Could it be a fake product? I don't find a single smart bulb when filtering the list of [supported devices](https://www.zigbee2mqtt.io/supported-devices/) by "ewelink". There's also no appropriate entry at https://zigbee.blakadder.com/search.html.

Besides, the company or product line uses <q>eWeLink</q>, not <q>Ewelink</q> or <q>ewelink</q>

mrmaximas, 2025-01-07:
How about press Permit join (All) in Zigbee2MQTT?

megustaeliphone, 2025-01-07:
Thank you for the suggestion! I did press "Permit join (All)" in Zigbee2MQTT, but unfortunately, the bulb still isn’t detected. Maybe if I press the button harder next time, it will work? 😉

Honestly, I’m not sure if the product might be fake. I bought it on AliExpress, so there’s always that possibility. I’m also unsure if it’s a supported device, but my main issue is that it’s not even being detected by Zigbee2MQTT. Without that, I just have a bulb that randomly changes to a different color every time I turn it on.

It’s a shame Christmas is over—I could’ve used it for the tree with all its random color changes. 🎄

megustaeliphone, 2025-01-08:
I finally managed to pair it by turning it on and off countless times. The manufacturer is Tuya, and the model is CK-BL702-AL-01.

closed 2025-01-08 by megustaeliphone

Nemesis24, 2025-01-11:
I've got he same and mine are recognize with difficulty as TS0505B_1_1 but don't work

megustaeliphone, 2025-01-11:
For me, it works, but the signal quality is not great. Even at just 1 meter from the coordinator, the signal is lost frequently.

Nemesis24, 2025-01-15:
after trying to pair again and again i've got some working, but it's very inconsistent and it seems causes problems with other iot for me !!

megustaeliphone, 2025-01-15:
For me as well. It was difficult to pair, and once I managed to do it, the connection quality was poor—it disconnects frequently, and everything works worse than without them. I’ve disconnected them. However, this doesn’t make my Zigbee network work decently, and I can’t afford to spend money on 20 plugs and 10 more bulbs just to see if that makes everything work normally.

pie-rs, 2025-02-24:
I have three of these, but in an e14 candle format—none of them pair—for the SEO. Don't buy. It's a false economy.