Koenkk/zigbee2mqtt #25661: Cannot pair EER50000

answer35, 2025-01-07:
### What happened?

I am trying to pair my EER50000 climate device with Z2M using a conbee II. If you put the device to pair mode by pressing once the set button, led starts flashing orange. When holding reset button for +5s then it switch green then red and go back to fix orange. When I follow pairing notes by holding the set button +5s nothing happens. Anything more I need to know to have it to connect ?

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

0x26720700

### Adapter

    ConBee2/RaspBee2

### Setup

Z2M HA Addon

m0wlheld, 2025-01-07:
> Anything more I need to know to have it to connect ?

Enabled permit join in Zigbee2MQTT?

answer35, 2025-01-07:
Yes of course, multiple times already

answer35, 2025-01-07:
Is there a specific channel to use ? I am currently on 11 that is needed for Legrand devices that I have already connected

answer35, 2025-01-07:
Actually after some more test, to complete how the device react, when I hold the reset button while led is blinking orange, after aprox 10s, it starts to blink few times green then it starts blinking orange again and as soon as I release the reset button it blinks red and then fix orange. I don't understand how to pair it without wiser gateway. 5s holding set button doesn't seems to do anything

Evan1086, 2025-01-08:
Hi, I have exactly the same behavior with my Wiser devices dated from 2014. 
But it works as expected for the ones dated from 2015 and 2017 (pairing activated by holding 5s the set key). The FW date indicated by Zigbee2MQTT is 20130521 on these 2 devices (H-relay produced in 2015 and L-relay produced in 2017).
I have the same issue with Wiser H-relay (EER50000), L-relay (EER42000) and smart plugs EER40000 (dated from 2013 and 2015).

I would really appreciate if someone can help to have the old devices to work.

answer35, 2025-01-08:
Oh if I understand it right, mine might be from 2013. Can you tell me how it reacts on 2015+ ? Is it fix orange when you hold 5s ? is the led starts to blink before or after you stop holding it ? or does it stays fix ?

Evan1086, 2025-01-09:
The LED is fix orange before I press the Set button, stay fix orange while I keep the button pressed and starts to blink as soon as I release the button.
If you open the white box you will see a sticker with the manufacturing date. Could you please let me know what is the date of your device ?

When I try to pair a 2014 L-Relay, I got the following logs :
error 2025-01-09 23:08:55z2m: Publish 'set' 'state' to 'L-Relay 0x00124b00040a024a' failed: 'Error: ZCL command 0x00124b00040a024a/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

davealtijdthuis, 2025-01-10:
I've upgraded to Zigbee2MQTT version 2.0.0 and can't pair any devices anymore. Previous I used Zigbee2MQTT version 1.x.x and all worked well. My Zigbee stick is an CONBEE II version 4. Is it discovered by Zigbee2MQTT but pairing is not working. 

_Starting Zigbee2MQTT without watchdog.
[2025-01-11 00:11:23] info:     z2m: Logging to console, file (filename: log.log)
[2025-01-11 00:11:24] info:     z2m: Starting Zigbee2MQTT version 2.0.0 (commit #060ae99c)
[2025-01-11 00:11:24] info:     z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-11 00:11:24] info:     zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE2465733","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2465733-if00","vendorId":"1cf1","productId":"0030"} => deconz: path=/dev/ttyACM0, score=4
[2025-01-11 00:11:24] info:     z2m: zigbee-herdsman started (resumed)
[2025-01-11 00:11:24] info:     z2m: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
[2025-01-11 00:11:24] info:     z2m: Currently 0 devices are joined.
[2025-01-11 00:11:24] info:     z2m: Connecting to MQTT server at mqtt://localhost
[2025-01-11 00:11:24] info:     z2m: Connected to MQTT server
[2025-01-11 00:11:24] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-01-11 00:11:24] info:     z2m: Zigbee2MQTT started!_

yaml config:
homeassistant:
  enabled: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost
serial:
  port: /dev/ttyACM0
version: 4

answer35, 2025-01-22:
Your device are from 2015 or later, right ?
Still cannot manage to set my device from 2013 in pairing mode. It goes in pairing mode when i briefly push the set button do it can pair with Wiser gateway but when I hold set button, nothing happens, it never start pairing. There is no log on Z2M side about it.

davealtijdthuis, 2025-01-22:
@answer35 Yes, all devices are relatively new from 2020 to 2024. After a few hours searching for solutions, trying different configs and even upgraded the conbee II, I've downgraded to version 1.42.0 and everything works like a charm again.

closed 2025-03-31 by github-actions[bot]