Koenkk/zigbee2mqtt #25664: NOUS D4Z is flodding the bus with payload (12 per second). no interval setting in reporting tab

imLeeat, 2025-01-07:
### What happened?

NOUS D4Z is constantly sending information on the bus.
In the Tab " reporting" there is no intervals set. and also no interval setting possible because of missing cluster
here is screenshot from DZ4 Reporting Tab:
[image]

compared to A6Z, reporting interval can be set:

[image]

### What did you expect to happen?

i expect to have also interval settings for DZ4

### How to reproduce it (minimal and precise)

pair D4Z

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

Home assistant

imLeeat, 2025-01-15:
i recognized that this issue is also existing with other devices:

1. NOUS D4Z no cluster with attributes available
2. eMylo (TUYA) SPM02V2.5 no cluster with attributs available
3. NOUS D2Z no default cluster with attributes (and wrongly paired as Tuya TS011F_plug_3)
4. NOUS A4Z no default cluster with attributes

imLeeat, 2025-01-15:
and this causes a lot of traffic (payload)

J108560, 2025-02-14:
NOUS D1Z also wrongly paired as Tuya TS011F_plug_3

It must be TS011F_with_threshold

AlexanderLeporiday, 2025-02-20:
> NOUS D1Z also wrongly paired as Tuya TS011F_plug_3
> 
> It must be TS011F_din_smart_relay_polling
> 
> ```
>     fingerprint: tuya.fingerprint('TS011F', ['_TZ3000_qeuvnohg', '_TZ3000_6l1pjfqe']),
>     model: 'TS011F_din_smart_relay_polling',
>     description: 'Din smart relay (with power monitoring via polling)',
> ```

I've just had two pull requests approved for the Nous D2Z.
To add Nous D1Z I'd need to know the Zigbee Manufacturer as reported on the device page.
[image]
Example for the D2Z above.

AlexanderLeporiday, 2025-02-20:
D4Z converter has minimum configuration resulting in missing endpoints.
Without endpoints you can't automatically or manually configure the reporting settings.
Changes need to be made [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/659bf7d6ff54e4b9a9abf55735c44b0bb5e3ef98/src/devices/tuya.ts#L9964)

J108560, 2025-02-20:
For D1Z Manufakturer is _TZ3000_2iiimqs9

AlexanderLeporiday, 2025-02-20:
> For D1Z Manufakturer is _TZ3000_2iiimqs9
Ah this will be a problem then and I've just registered this as D2Z.
It means that functionally the D1Z, D2Z and D3Z are the same device with some hardware changes to trip at higher currents.

Perhaps we'll need to go with a model of DxZ, however D4Z is a different. So it'll get confusing.
For now, after the update the D1Z, D2Z and D3Z will likely all appear as D2Z but they all share the same functionality.

closed 2025-04-29 by github-actions[bot]

bagasse, 2026-01-12:
Hello, same issue here, I've multiple D4Z devices that are flooding the zigbee network. I didn't understand if the device firmware is missing some configuration regarding  the "interval" of each message or it's a lack of  attribute binding on the z2m side.

xardas-eu, 2026-02-14:
Hi! Why is this not planned? It is still an issue for people using this device

@AlexanderLeporiday  maybe you could help?