Koenkk/zigbee2mqtt #25665: Perte d'action SH-SC07

icagn, 2025-01-07:
### What happened?

Hello, the 'Action (enum)' field is no longer available

### Zigbee2MQTT version

21.12.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

SH-SC07

### Setup

Raspberry PI4 HA

Hypnos-69, 2025-01-08:
Hi,

I upgraded to Z2M 2.0.0 commit 060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a on Monday and since then my following remote controls have not worked properly on Home Assistant:
 - Loratap SS6100ZB 1 button (Tuya TS0041).
 - Loratap SS600ZB 3 buttons (Tuya TS0043).
 - SONOFF SNZB-01 1 button.

For my 5 remotes, the Action sensor appears as unavailable with the following message :
"This entity is no longer provided by the mqtt integration. If the entity is no longer used, delete it from the settings."

[image]

huperaisan, 2025-01-08:
Same issue with SH-SC07. The Action field no longer exists, and the only entities remaining are "Battery" and "Link quality". I didn't see anything in the breaking changes list about this.