Koenkk/zigbee2mqtt #25666: experimental_event_entities problem with events

antst, 2025-01-07:
### What happened?

Maybe it is how it is supposed to be, but..

let assume you have some button, which reports press-release states, then you can use events in automations, you pressed - it delivers "press" you released - it deliver "release", then you press again and it is "press" again. So, second event.

But if I have some another button, which delivers "single" , "double" etc, then no luck.
You click once, it delivered "single" and, obviously, stays "single". Then I press again. MQTT delivers to HA "single". But is is "single" already, so...nothing changed from HA point of view, so, no automation triggered. Unless I do something like double click etc, to change "single" state to something else. Then, after such "reset", single click will work again one time.

So, kind of sucks! :)

I'd suggest that, by default, all such buttons would follow "single" with "release" message maybe.

### What did you expect to happen?

some way to actually use events for a lot of devices, for which it does not work right now.

### How to reproduce it (minimal and precise)

any button, switch etc with "single", "double" etc events, but without explicit "press" event.

### Zigbee2MQTT version

2.0

### Adapter firmware version

8.1.0 [GA]

### Adapter

EmberZNet

### Setup

Supervised HA with Z2M add-on.

ctomazini, 2025-01-07:
Hello!
I may not fully understand the issue you're facing, but in my automations, I typically set the trigger as a state trigger and use event.entity_action for the entity ID. Then, I create conditions based on each button type or press. It works for me, even if I press the same button (two times a single, for example) twice in a row. Just a note, do not add any "from" or "to" attribute for the trigger.

Let me know if this aligns with what you're looking for!
Here's an example from an automation I use for a 4 buttons moe's switch.

```
alias: Nova Automação
description: ""
triggers:
  - trigger: state
    entity_id:
      - event.moes_botoeira01_action
conditions: []
actions:
  - choose:
      - conditions:
          - condition: state
            entity_id: event.moes_botoeira01_action
            attribute: event_type
            state: 1_single
        sequence:
          - if:
              - condition: time
                after: "23:00:00"
                before: "06:00:00"
                weekday:
                  - sun
                  - mon
                  - tue
                  - wed
                  - thu
                  - fri
                  - sat
            then:
              - metadata: {}
                data:
                  color_name: darkgoldenrod
                  brightness_pct: 10
                target:
                  entity_id:
                    - light.abajur_charles
                action: light.toggle
            else:
              - metadata: {}
                data:
                  color_name: darkgoldenrod
                  brightness_pct: 100
                target:
                  entity_id:
                    - light.abajur_charles
                action: light.toggle
      - conditions:
          - condition: state
            entity_id: event.moes_botoeira01_action
            attribute: event_type
            state: 1_double
        sequence:
          - if:
              - condition: time
                after: "23:00:00"
                before: "06:00:00"
                weekday:
                  - sun
                  - mon
                  - tue
                  - wed
                  - thu
                  - fri
                  - sat
            then:
              - metadata: {}
                data:
                  color_name: darkgoldenrod
                  brightness_pct: 10
                target:
                  entity_id: light.abajur_carine
                action: light.toggle
            else:
              - metadata: {}
                data:
                  color_name: darkgoldenrod
                  brightness_pct: 100
                target:
                  entity_id: light.abajur_carine
                action: light.toggle
      - conditions:
          - condition: state
            entity_id: event.moes_botoeira01_action
            attribute: event_type
            state: 2_single
        sequence:
          - type: toggle
            device_id: 1d9ea071ec7957d93a0e2430faa3cb98
            entity_id: 5ffd138e86157f0997ff8627bf934596
            domain: light
      - conditions:
          - condition: state
            entity_id: event.moes_botoeira01_action
            attribute: event_type
            state: 3_single
        sequence:
          - data: {}
            target:
              entity_id: light.todas_as_luzes_2
            action: homeassistant.turn_off
      - conditions:
          - condition: state
            entity_id: event.moes_botoeira01_action
            attribute: event_type
            state: 4_single
        sequence:
          - data: {}
            target:
              device_id: 7108fb98caa6d24bd4e013ae9e46ff95
            action: media_player.media_play_pause
      - conditions:
          - condition: state
            entity_id: event.moes_botoeira01_action
            attribute: event_type
            state: 4_double
        sequence:
          - data:
              entity_id: media_player.lg_webos_tv_uj6565
              button: ENTER
            action: webostv.button
mode: single

```

antst, 2025-01-08:
But then it is a hell, if for such simple thing we need to use state trigger with condition!
And, particular hell, when you have multiple triggers by multiple devices.
But I see, in your case, it is trigger-device-based automation. So, you have automation per trigger device, kind of, where you describe all actions of such device.

In my case, typically i have automation with multiple triggers (by different devices) with one or more actions on such trigger.

antst, 2025-01-08:
so, it was
```
trigger: state
entity_id: sensor.study_button_action
enabled: true
to: single
```
then become 
```

then become 

```
domain: mqtt
device_id: a6e65827e8df0b0725d86b1384138489
type: action
subtype: single
trigger: device
enabled: true
```

and I need now to attach complex condition stuff to multi-trigger automations (so I will need to check state and device combinations) or rewrite ad trigger-device-based....

mundschenk-at, 2025-01-08:
Unfortunately, this is a [HA issue, see this WTH](https://community.home-assistant.io/t/wth-are-event-entities-not-triggered-a-second-time-when-using-the-attribute-event-type-in-a-state-trigger/804428).

Koenkk, 2025-01-08:
> The new event entities do not support this functionality though. If the same event comes through a second time, the event is essentially ignored because the event_type attribute does not change.

That's seem like a really weird design decision from HA which looks like a deal breaker. The MQTT device trigger doesn't have this behaviour.

mundschenk-at, 2025-01-08:
I think this was not really a design decision, but emergent behavior from how the HA state engine works. The event is not technically ignored, the timestamp that is the `state` of the entity is updated, but the value of the `attribute` `event_type` does not change. The difference is subtle, and the behavior is annoying, but it is consistent with how all entities and their attributes work.

I did not notice this behavior during testing (which led to the incorrect/not general enough example in the docs) because the Hue dimmer has so many actions that each press-hold-release cycle always changes the `event_type` value. As a possible workaround, we could publish an empty string or a default value (e.g. `idle`) after an action that does not naturally cycle through values.

mundschenk-at, 2025-01-08:
@ctomazini You mean you had two separate automations for "single button press A" and "single button press B" and when you pressed both at the same time, both automations ran separately? If so, that's not happening with the new way because you set the wrong automation `mode`, needs to be `parallel`, see https://www.home-assistant.io/docs/automation/modes/ for details.

dnedeltchev, 2025-01-09:
Apologies if this is not the right place but I am not able to find anywhere where to post this question. Does anyone know how to use these new event triggers in node-red? I've reverted back to the legacy actions but if they will be removed in the future... or point me in the right direction?

closed 2025-03-18 by github-actions[bot]