Koenkk/zigbee2mqtt #25667: [New device support]: NOUS E6

niloofr, 2025-01-07:
### Link

https://www.domadoo.fr/fr/peripheriques/6184-capteur-de-temperature-et-d-humidite-lcd-zigbee-30-nous-e6-5907772033975.html

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0xa4c1386f3c6da830","nwkAddr":43211,"manufId":4417,"manufName":"_TZE200_wtikaxzs","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1736289486479}

### Zigbee2MQTT version

2.0.0-2

### Comments

Hello,

I have a fresh install on a Raspberry 5, these components : 
 - Home Assistant (2025.1.1)
 - Zigbee2MQTT (2.0.0-2)
 - Sonoff Zigbee 3.0 USB Dongle Plus V2

When I pair a temperature and humidity sensor (vendor NOUS and model E6), I have this message : 
 - Zigbee Model     TS0601
 - Zigbee Manufacturer     _TZE200_wtikaxzs
 - Support status     Not supported (how_to_add_support)

I see this page about this device : https://www.zigbee2mqtt.io/devices/E6.html
But I don't understand why it's displayed not supported.
May be the device id is not the same.

Please, could you help me ?

Thank you.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_wtikaxzs',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I don't know how I can set/edit external definition, sorry.

Koenkk, 2025-01-08:
Will be added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/8574

Koenkk, 2025-02-01:
`_TZE200_wtikaxzs` is supported in z2m 2.1.0

closed 2025-02-01 by Koenkk