Koenkk/zigbee2mqtt #25669: Bad gateway after update to 2.0.0

Friedi1970, 2025-01-08:
### What happened?

I tried 2.0.0-1 as well as 2.0.0-2 and I also changed the config.yaml as described, but after reboot nothing works and when I try to start the addon I just get "bad gateway".

### What did you expect to happen?

I'd expect that the version 2 is working as 1.

### How to reproduce it (minimal and precise)

Update to 2.0.0-2

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20210708

### Adapter

SONOFF Dongle Plus

### Setup

Addon on Home Assistant

akitaylor, 2025-01-08:
Check the zigbee2mqtt log and attach it here. In this form your issue is absolutely imprecise, nothing can be established about the root cause.

Friedi1970, 2025-01-08:
That is what I got from the logs:

```
[2025-01-08 07:42:49] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-08 07:42:49] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-08 07:42:49] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-08 07:42:49] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-08 07:42:49] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-08 07:42:49] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-08 07:42:49] error: 	z2m: Exiting...
[2025-01-08 07:42:49] error: 	z2m: Error: USB adapter discovery error (No valid USB adapter found). Specify valid 'adapter' and 'port' in your configuration.
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:523:15)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:33)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[2025-01-08 07:42:49] error: 	z2m: If this happens after updating to Zigbee2MQTT 2.0.0, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364
```
The thing is that I didn't change anything and I have no idea what happened.

Danieldiazi, 2025-01-08:
> https://github.com/Koenkk/zigbee2mqtt/discussions/24364

Into your logs there are a link "https://github.com/Koenkk/zigbee2mqtt/discussions/24364"  and it says do you need to add adapter line in zigbee2mqtt 2.0.0. Are you added that line?

Danieldiazi, 2025-01-08:
> > I tried 2.0.0-1 as well as 2.0.0-2 and I also changed the config.yaml as described,
> 
> No, you didn‘t. You did not specify the adapter type in the serial section, as described in #24198

It seems that this is the issue, as I also explained in my previous message. Let's wait for [Friedi1970](https://github.com/Friedi1970) answer...

Koenkk, 2025-01-08:
This is one of the Zigbee2MQTT 2.0.0 breaking changes, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364 on how to fix this.

closed 2025-01-08 by Koenkk