Koenkk/zigbee2mqtt #25670: Tuya RB-SRAIN01 reports lux, should be mW

JarnoBoks, 2025-01-08:
### What happened?

The  Tuya RB-SRAIN01 reports illuminance in lx on three channels (illuminance/illuminance avg 20 / illuminance max), but to my knowledge this should be mV instead of lx.

### What did you expect to happen?

I would expect a conversion or the unit mV so I can do my conversion. 

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20220219

### Adapter

CC2531

### Setup

Plain on x86064 debian

### Debug log

None

lux73, 2025-01-08:
ehm, no...

https://en.m.wikipedia.org/wiki/Lux

JarnoBoks, 2025-01-08:
ehm, I don't understand your answer. 

The unit reports ~5300lx while my calibrated light meter reports 680lx. And in this review the values are shown as mV too. 
https://smarthomescene.com/reviews/tuya-zigbee-solar-rain-sensor-rb-srain01-review/

closed 2025-03-17 by github-actions[bot]

dgibbs64, 2025-11-30:
I have re raised this issue as per the manual it is measuring mV not lux

hipp0o, 2025-12-07:
How can illuminance be messured in "mV"? - this is bs even If the "manual" says something else.
So it was expected that z2mqtt throws an error now...

> Logger: homeassistant.components.mqtt.entity
> Quelle: components/mqtt/entity.py:159
> Integration: MQTT (Dokumentation, Probleme)
> Erstmals aufgetreten: 00:23:16 (1 Vorkommnis)
> Zuletzt protokolliert: 00:23:16

`Error 'The unit of measurement `mV` is not valid together with device class `illuminance`' when processing MQTT discovery message topic: 'homeassistant/sensor/0xa4c1380a3c17d253/illuminance/config', message: '{'availability': [{'topic': 'zigbee2mqtt0/bridge/state', 'value_template': '{{ value_json.state }}'}, {'topic': 'zigbee2mqtt0/Regensensor/availability', 'value_template': '{{ value_json.state }}'}], 'availability_mode': 'all', 'default_entity_id': 'sensor.regensensor_illuminance', 'device': {'hw_version': 1, 'identifiers': ['zigbee2mqtt_0xa4c1380a3c17d253'], 'manufacturer': 'Tuya', 'model': 'Solar rain sensor', 'model_id': 'RB-SRAIN01', 'via_device': 'zigbee2mqtt_bridge_0x84b4dbfffebc5ee1', 'name': 'Regensensor'}, 'device_class': 'illuminance', 'enabled_by_default': True, 'object_id': 'regensensor_illuminance', 'origin': {'support_url': 'https://www.zigbee2mqtt.io', 'sw_version': '2.7.1', 'name': 'Zigbee2MQTT'}, 'state_class': 'measurement', 'state_topic': 'zigbee2mqtt0/Regensensor', 'unique_id': '0xa4c1380a3c17d253_illuminance_zigbee2mqtt0', 'unit_of_measurement': 'mV', 'value_template': '{{ value_json.illuminance }}'}'`

Pls revert changes. Or change device class or whatever.