Koenkk/zigbee2mqtt #25672: device_options must be object

barii86, 2025-01-08:
### What happened?

after updating to version 2.0.0-2 the add-on did not stop starting

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

??

### Adapter

zigbee cc2652p

### Setup

Intel NUC HA Generic x86x64 Instalation

### Debug log

configuration:
[...]
  availability_blocklist: []
  availability_passlist: []
device_options:
  legacy: false
blocklist: []
passlist: []
[...]

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
[09:13:19] INFO: Preparing to start...
[09:13:19] INFO: Socat not enabled
[09:13:19] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
            READ THIS CAREFULLY
Refusing to start because configuration is not valid, found the following errors:
- device_options must be object
If you don't know how to solve this, read https://www.zigbee2mqtt.io/guide/configuration
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

frittenlab, 2025-01-08:
Check your configuration.yaml the device options should look similar to this:

```
device_options:
  legacy: false
```

barii86, 2025-01-08:
> Check your configuration.yaml the device options should look similar to this:
> 
> ```
> device_options:
>   legacy: false
> ```

that's exactly what I have

barii86, 2025-01-08:
[image]

m0wlheld, 2025-01-08:
How about removing `device_options` section?

m0wlheld, 2025-01-08:
Btw: `homeassistant: true` is also incompatible to Zigbee2MQTT 2.x

closed 2025-01-08 by barii86