Koenkk/zigbee2mqtt #25674: Error: Device. xxx has no input cluster manuSpecificLumi (Aqara QBKG20LM)

hongliangl, 2025-01-08:
### What happened?

After restarting z2m, I always get the logs like the following:

```
[2025-01-08 17:58:31] error: 	z2m: Failed to configure '卧室 灯开关', attempt 1 (Error: Device 0x54ef441000600681 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:59:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:144:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:207:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:27)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
[2025-01-08 17:58:31] error: 	z2m: Failed to configure '客厅 灯开关', attempt 1 (Error: Device 0x54ef441000603d80 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:59:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:144:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:207:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:27)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
[2025-01-08 17:58:31] error: 	z2m: Failed to configure '玄关 灯开关', attempt 1 (Error: Device 0x54ef4410005d1ce9 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:59:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:144:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:207:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:27)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
[2025-01-08 17:58:31] error: 	z2m: Failed to configure '厨房 灯开关', attempt 1 (Error: Device 0x54ef4410005d39ff has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:59:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:144:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:207:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:27)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
[2025-01-08 17:58:32] error: 	z2m: Failed to configure '客厅 灯开关', attempt 2 (Error: Device 0x54ef441000603d80 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:59:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:144:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:207:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:27)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
[2025-01-08 17:59:01] error: 	z2m: Failed to configure '厨房 灯开关', attempt 2 (Error: Device 0x54ef4410005d39ff has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:59:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:144:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:207:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:27)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at Immediate.<anonymous> (/app/lib/extension/configure.ts:69:17))
[2025-01-08 17:59:11] error: 	z2m: Failed to configure '客厅 灯开关', attempt 3 (Error: Device 0x54ef441000603d80 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:59:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:144:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:207:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:27)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at Immediate.<anonymous> (/app/lib/extension/configure.ts:69:17))
[2025-01-08 17:59:21] error: 	z2m: Failed to configure '次卧 灯开关', attempt 1 (Error: Device 0x54ef44100060406e has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:59:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:144:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:207:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:27)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at Immediate.<anonymous> (/app/lib/extension/configure.ts:69:17))
[2025-01-08 17:59:28] error: 	z2m: Failed to configure '卧室 灯开关', attempt 2 (Error: Device 0x54ef441000600681 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:59:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:144:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:207:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:27)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
[2025-01-08 17:59:31] error: 	z2m: Failed to configure '玄关 灯开关', attempt 2 (Error: Device 0x54ef4410005d1ce9 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:59:15)
[... 37 lines cut ...]
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:27)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))

```

### What did you expect to happen?

The corresponding devices work normally, but I'm curious about that why there are such logs. There is no reason to have them.

### Zigbee2MQTT version

 2.0.0

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Add-on on Home Assistant OS on PVE

Koenkk, 2025-01-08:
Could you try to force remove the device, restart z2m and re-pair the device? If that doesn't help, please share the data/databse.db entry of this device.

hongliangl, 2025-01-09:
> Could you try to force remove the device, restart z2m and re-pair the device? If that doesn't help, please share the data/databse.db entry of this device.

Sure, will do.

hongliangl, 2025-01-09:
I picked one device followed the steps you posted, but I got the error logs again. Here are the items from `database.db`

```json
{"id":13,"type":"Router","ieeeAddr":"0x54ef441000603d80","nwkAddr":19885,"manufId":4151,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.b2nacn01","epList":[1,2,21,31,41,42,51,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9,1794,2820],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.b2nacn01","appVersion":28,"zclVersion":3,"dateCode":"07-22-2024"}},"manuSpecificLumi":{"attributes":{"220":{"type":"Buffer","data":[246,201,240,46,173,77,0,0,234,174,255,233]},"223":{"type":"Buffer","data":[22,33,0,0,23,33,0,0,12,33,34,0,24,33,206,203,30,33,109,0,31,33,18,233,32,33,195,52,34,33,51,38,35,33,0,0,36,32,0,37,35,195,218,183,81,254,35,1,0,0,0]},"229":{"type":"Buffer","data":[25,17,8,86,32,37,104,3,220,37,98,216,148,37,110,162,25,37,82,88,236,37,103,221,78,37,91,5,189,37,42,225,213,37,75]},"247":{"type":"Buffer","data":[100,16,1,101,16,1,3,40,31,152,57,100,187,149,66,149,57,108,114,51,67,150,57,0,96,11,69,151,57,142,192,162,67,5,33,26,0,154,32,0,8,33,28,1,9,33,7,22,11,32,0,10,33,0,0,12,32,1]},"670":0}},"genOnOff":{"attributes":{"245":117440537,"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dedcb9","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"245":117440542,"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dedcb9","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"21":{"profId":260,"epId":21,"devId":256,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":72.16000366210938}}},"binds":[],"configuredReportings":[],"meta":{}},"31":{"profId":260,"epId":31,"devId":256,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"41":{"profId":260,"epId":41,"devId":256,"inClusterList":[18],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"42":{"profId":260,"epId":42,"devId":256,"inClusterList":[18],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"51":{"profId":260,"epId":51,"devId":256,"inClusterList":[18],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":28,"stackVersion":2,"hwVersion":1,"dateCode":"07-22-2024","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1736399279684}
{"id":14,"type":"Router","ieeeAddr":"0x54ef44100060406e","nwkAddr":13380,"manufId":4151,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.b2nacn01","epList":[1,2,21,41,42,51,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9,1794,2820],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"stackVersion":2,"hwVersion":1,"modelId":"lumi.switch.b2nacn01","appVersion":28,"zclVersion":3,"dateCode":"07-22-2024"}},"manuSpecificLumi":{"attributes":{"220":{"type":"Buffer","data":[188,91,4,47,216,196,68,52,0,0,255,233]},"223":{"type":"Buffer","data":[22,33,0,0,23,33,0,0,12,33,7,0,24,33,63,84,30,33,71,0,31,33,115,234,32,33,64,28,34,33,201,37,35,33,0,0,36,32,0,37,35,127,5,159,81,254,35,1,0,0,0]},"229":{"type":"Buffer","data":[17,0,17,225,213,37,67,205,117,37,94,221,78,37,95,101,7,37,118,61,0,37,81,144,22,37,50,216,196,37,119,0,0,36,114,143,81,37,89,173,77,37,89,216,148,37,50,174,237,37,58,219,249,37,50,194,179,37,68,5,189,37,41,3,220,37,65,228,28,37,54]},"247":{"type":"Buffer","data":[100,16,0,101,16,0,3,40,31,152,57,0,0,0,0,149,57,21,90,218,64,150,57,0,0,12,69,151,57,0,0,0,0,5,33,11,0,154,32,0,8,33,28,1,9,33,0,20,11,32,0,10,33,0,0,12,32,1]},"252":0,"512":0,"515":0,"670":0}},"genOnOff":{"attributes":{"245":117440584,"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dedcb9","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"245":117440728,"onOff":1}},"manuSpecificLumi":{"attributes":{"512":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dedcb9","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"21":{"epId":21,"inClusterList":[],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":22.841999053955078}}},"binds":[],"configuredReportings":[],"meta":{}},"41":{ [... cut ...] e":1}],"meta":{}},"21":{"epId":21,"inClusterList":[],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":70.80000305175781}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":28,"stackVersion":2,"hwVersion":1,"dateCode":"07-22-2024","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1736399342876}
```

Koenkk, 2025-01-10:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

hongliangl, 2025-01-13:
Thanks for fixing this!

closed 2025-01-13 by hongliangl