Koenkk/zigbee2mqtt #25675: WXKG15LM Action "hold_left" is missing

oli737, 2025-01-08:
### What happened?

hold_lefft action its missing now.

### What did you expect to happen?

updated to 2..0

### How to reproduce it (minimal and precise)

Its still missing

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

 8.0.2 [GA]

### Adapter

Sonoff Dongle E

### Setup

HAOS on Proxmox

adynis, 2025-01-08:
In HA I've selected automation - Trigger: Device, and then ... I have it :
[image]

oli737, 2025-01-08:
I can add in yaml than it show up as expected but was not listed in section before.
Right after manual edit unknown it was shown an "unknown".
Once pressed the hold left button its was in the selection from graphical editor.

[image]

closed 2025-01-08 by oli737