Koenkk/zigbee2mqtt #25682: Tuya TS0225 | z2m: Exception while calling fromZigbee converter: Value '5' is not allowed, expected one of 0,1,2,3}

Kekbach, 2025-01-08:
### What happened?

The error Accures every second or so..

### How to reproduce it (minimal and precise)

On Movement

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

Pi 4

### Debug log

info 2025-01-08 18:39:34z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:34+01:00","linkquality":152,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"large","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:36+01:00","linkquality":152,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"large","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:36+01:00","linkquality":152,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"small","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:37+01:00","linkquality":152,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"small","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:37+01:00","linkquality":152,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"large","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:41+01:00","linkquality":152,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"large","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:41+01:00","linkquality":152,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"small","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:44+01:00","linkquality":148,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"small","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:44+01:00","linkquality":148,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"large","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:47+01:00","linkquality":152,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"large","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:47+01:00","linkquality":152,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"small","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:49+01:00","linkquality":128,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"small","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:49+01:00","linkquality":132,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"large","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:51z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:51+01:00","linkquality":140,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"large","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:57z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:57+01:00","linkquality":140,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"large","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:57z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:57+01:00","linkquality":136,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"small","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:39:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:39:58+01:00","linkquality":140,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"small","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:40:04z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:40:04+01:00","linkquality":140,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"small","presence":true,"small_detection_distance":4.5,"sma [... cut ...] e":4.5,"small_detection_sensitivity":4}'
info 2025-01-08 18:40:56z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Tuya Presense', payload '{"fading_time":10,"illuminance":2353,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":3,"last_seen":"2025-01-08T18:40:56+01:00","linkquality":132,"medium_motion_detection_distance":6,"medium_motion_detection_sensitivity":5,"motion_state":"small","presence":true,"small_detection_distance":4.5,"small_detection_sensitivity":4}'

Koenkk, 2025-01-09:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Kekbach, 2025-01-09:
[log.log](https://github.com/user-attachments/files/18366950/log.log)

Koenkk, 2025-01-10:
This is the motion state, do you have any clue what value 5 means here? Here are the current mapped values:

```
{none: tuya.enum(0), large: tuya.enum(1), medium: tuya.enum(2), small: tuya.enum(3)}
```

Kekbach, 2025-01-10:
Where can i see the raw Data in the web interface my first thought is that maybe the Level of movement is 0-5 because if i move fast the error accurs and if i move slow the sensor is not throwing any errors

Kekbach, 2025-01-10:
Error 5 = Much Movement
Error 4 = a little less movement

Kekbach, 2025-01-10:
FYI Here i have ordered it 
https://de.aliexpress.com/item/1005007605874437.html?spm=a2g0o.order_list.order_list_main.4.bfe85c5fjGMMoU&gatewayAdapt=glo2deu

Koenkk, 2025-01-11:
Added them!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Kekbach, 2025-01-11:
Still There...

[log.log](https://github.com/user-attachments/files/18386786/log.log)

[image]

[image]

Koenkk, 2025-01-12:
Sorry I did not merge https://github.com/Koenkk/zigbee-herdsman-converters/pull/8590 yet, will be there once it is merged.

Koenkk, 2025-01-13:
The values are `'none', 'small', 'medium', 'large', 'mini', 'micro']` now, what would you recommend instead?

Kekbach, 2025-01-13:
i think it is 
'none', 'micro', 'mini', 'small', 'medium', 'large', 

because i think its the motion intensity from no movement to much movement

Kekbach, 2025-01-14:
yes

Kekbach, 2025-01-14:
Yes Currently it is outputting

None = No Movement
small = if i dont move 
large = if i move more
Mini = more movement than large 
micro = more movement than Mini 

medium did not show maybe it is a too little change

but it does not make any sense because then it is in no order i think but i cant see the raw values

Koenkk, 2025-01-15:
I don't think there is a need to see the raw values, just let me know what name you want to use instead of `micro` and `mini` and I will update it accordingly.

Kekbach, 2025-01-15:
Mini = Huge
micro = Gigantic

maybe?

Koenkk, 2025-01-17:
Updated!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Feriman22, 2025-02-27:
On my end, I can confirm that the issue has been resolved. Thank you very much!

closed 2025-03-01 by Koenkk