Koenkk/zigbee2mqtt #25684: Bad gateway after update to 2.0.0-2

wzmigrod, 2025-01-08:
### What happened?

after update from 2.0.0-1 to 2.0.0-2 Z2M seem not to start fully - opening the Zigbee2Mqtt addon end up up with the Bad Gateway error. 
[image]
[image]
Restoring 2.0.0-1 does not help
After restoring from backup HA with Z2M 2.0.0-1 Z2M has been working OK until the automatic udpate to 2.0.0.-2 - that resulted with Bad Gateway error again
System infomation: 
Core 2024.12.5
Supervisor 2024.12.3
Operating System 14.1
Frontend 20241127.8

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

zStack3x0 20220219 firmware 1.42.0

### Adapter

Sonoff ZigBee 3.0 USB CC2652P

### Setup

Add on on Home Assistant; HA installed on Proxmox

### Debug log

[Z2M log.txt](https://github.com/user-attachments/files/18352025/Z2M.log.txt)

chrishoerm, 2025-01-08:
I have the same issue. Most people forgot to add "adapter: zstack" in the config. Maybe, this will help in your case as well.

wzmigrod, 2025-01-08:
Already added "adapter : zstack" in the config - but this has not fixed the issue. And in fact in the log one can see that the zigbee adapter is recignized properly and zigbee devices are connected

chrishoerm, 2025-01-09:
I've the same issue. Same error log.

TanerCRB, 2025-01-09:
Also similar behaviour on my side for the HA addon, I described the problem in another issue and added the logs so maybe this will help to analyse what is wrong, I need to go back to 1.42.0-2 to have zigbee2mqtt work again. 

[https://github.com/Koenkk/zigbee2mqtt/issues/25622#issuecomment-2577096180](url)

levanLuongnt, 2025-01-09:
how to update to 2.0.0-2. i using docker.

Daximino, 2025-01-12:
> Already added "adapter : zstack" in the config - but this has not fixed the issue. And in fact in the log one can see that the zigbee adapter is recignized properly and zigbee devices are connected

But if i'm usig ember adapter which adapter i specify in the configuration?

Koenkk, 2025-01-12:
@Daximino 

> But if i'm usig ember adapter which adapter i specify in the configuration?

`adapter: ember`

closed 2025-02-02 by wzmigrod