Koenkk/zigbee2mqtt #25689: z2m Timeout

pinkfloydFR, 2025-01-08:
### What happened?

Hi,
Since update, i have a lot off problem. I control my light with wireless switch to order relay. But now i need to press multiple time my switch for the order to be send

All my automation are on node red.

I currently disabled all my get state action node to replace it by mqtt reader

Note, when i publish mqtt message is well received but no action, i need to send the message multiple time for my relay toggle too.

[image]

```
devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant:
  enabled: true
  legacy_action_sensor: true
mqtt:
  base_topic: zigbee2mqtt
  user: addons
  password: ******
  server: mqtt://core-mosquitto:1883
serial:
  port: /dev/ttyACM0
  adapter: deconz
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
device_options: 
  legacy: false
blocklist: []
passlist: []
queue: {}
frontend:
  enabled: true
  port: 8099
availability:
  enabled: false
version: 4
```

### What did you expect to happen?

No timeout, action directly send

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

Add on HA OS rasp 4

### Debug log

[45df7312_zigbee2mqtt_2025-01-08T21-16-14.194Z.log](https://github.com/user-attachments/files/18353203/45df7312_zigbee2mqtt_2025-01-08T21-16-14.194Z.log)

pinkfloydFR, 2025-01-09:
As seen on the doc : 

[image]

So i downgraded with deCONZ_ConBeeII_0x26580700.bin file, no more luck :(

xupefei, 2025-01-09:
I have the same issue with ConBee2 and Hue Smart Plug. The of/off command is actually being send to the device but Z2M shows “timeout”.

Nerivec, 2025-01-09:
The timeout (10s) would be because the device in question did not reply to confirm, after Z2M sent the command.

Any chance something in the environment around changed that could create more interference than before? With commands sometimes sending, and responses sometimes received, it sounds like something is messing with the Zigbee network (or possibly the USB port).

Have you tried the deconz-specific `adapter_delay` option to see if it has any effect on this?
https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#advanced-configuration

closed 2025-01-12 by pinkfloydFR