Koenkk/zigbee2mqtt #25695: Bosch thermostats II ( BTH-RA, BTH-RM) do not create a temperature sensor in Home Assistant / Missing Entität

Kai9555, 2025-01-09:
### What happened?

The BTH-RA and BTH-RM thermostats display the temperature under MQTT devices, but no sensor is created. This means I cannot assign an external sensor of the wall thermostat to the thermostats as a group.

### Zigbee2MQTT version

2.00-2

### Adapter firmware version

7.4.5 [GA]

### Adapter

SLZB-06M

### Setup

HACS/Addon on HA 2025.1

Kai9555, 2025-01-09:
[image]
[image]

Kai9555, 2025-01-09:
[image]

Kai9555, 2025-01-09:
[image]
[image]

h1nnak, 2025-01-14:
Hey, 

I also noticed this problem and have solved it by myself. Maybe it will help you, put this in:
/homeassistant/sensors.yaml

# Temperatur-Attribut als Extra-Wert für Bosch Thermostate
# Flur
template:
  - sensor:
      - name: "Thermostat Flur Temperatur"
        unique_id: "th_flur_temperature" #change to your name
        unit_of_measurement: '°C'
        state_class: "measurement"
        state: >
            "{{ state_attr('climate.th_flur', 'current_temperature') }}"

drufener, 2025-03-21:
Hi there
any news to this?
I'm "fighting" with the same problem

Thanks Daniel

dierochade, 2025-04-11:
> Hi there any news to this? I'm "fighting" with the same problem
> 
> Thanks Daniel

Hi Daniel, what do you mean by news? It seems the temperature is only exposed as an attribute of the device. 

If you want to have a sensor entity, just create a template. You can do this with yaml (see example above) or in the UI too (settings→ devices/entites/etc→create helper → template. Then select the thermostat in the device dropdown and enter under template "{{ state_attr('REPLACE_WITH_YOUR_DEVICE_ENTITY', 'current_temperature') }}"

drufener, 2025-04-11:
> > Hi there any news to this? I'm "fighting" with the same problem
> > Thanks Daniel
> 
> Hi Daniel, what do you mean by news? It seems the temperature is only exposed as an attribute of the device.
> 
> If you want to have a sensor entity, just create a template. You can do this with yaml (see example above) or in the UI too (settings→ devices/entites/etc→create helper → template. Then select the thermostat in the device dropdown and enter under template "{{ state_attr('REPLACE_WITH_YOUR_DEVICE_ENTITY', 'current_temperature') }}"
> 
> Be aware though that the measurement will be quite off the room temperature, cause its next to the heater...

Many thanks for that
it works well for my application.
The sensor is a standalone sensor, located in the middle of the room. So the measurement will be ok for me.
Thanks again!

closed 2025-06-19 by github-actions[bot]