Koenkk/zigbee2mqtt #25699: Cannot update Nodon SIN-4-1-21

PtiBed0, 2025-01-09:
### What happened?

Hello
Z2m offers an update of the Nodon SIN-4-1-21 module and I have an invalid CRC-32 error
[image]
Many Thanks

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P + CP2102N

### Setup

HAOS

### Debug log

[2025-01-09 10:18:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Chauffe Eau', payload '{"energy":7.6,"impulse_mode_configuration":null,"last_seen":"2025-01-09T09:18:10.492Z","linkquality":87,"power":0,"power_on_behavior":"on","state":"ON","switch_type":"auto_detect","update":{"installed_version":771,"latest_version":773,"state":"available"}}'
[2025-01-09 10:18:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'Chauffe Eau' failed (Image CRC-32 is invalid)","status":"error","transaction":"ged1t-1"}'
[2025-01-09 10:18:11] error: 	z2m: Update of 'Chauffe Eau' failed (Image CRC-32 is invalid)

Gramboa, 2025-01-09:
Same issue for me.

zulufoxtrot, 2025-01-10:
I have almost the same issue. Same error when performing OTA.

Device: https://www.zigbee2mqtt.io/devices/SIN-4-FP-21.html

Zigbee2MQTT version
    2.0.0

Coordinator type
    EmberZNet

Coordinator revision
    7.4.4 [GA]

Frontend version
    0.9.4

zigbee-herdsman-converters version
    21.12.0

zigbee-herdsman version
    3.2.1

Note: I had no issue updating another device of the same model, but that was with an older version of Z2M.

technoman63, 2025-01-10:
I have the same issue

Chartreusito, 2025-01-10:
As far as i know the problem is on the manufacturer's side, update is not possible even using their app and connecting using bluetooth

steve41200, 2025-01-12:
I have same issue, I have do this update in 1.42 with 3 of device it works.
I'm waiting before update all to be careful. I have update z1m in 2.0.2 directly and I have the same error when I try to update them.
I think it's link with 2.0.2, may be the image of device change ?

PtiBed0, 2025-01-12:
Hello,
Some change !
there is no more error Image CRC-32 is invalid but only (No new image available)
Good Progress ;-)
thanks in advance

steve41200, 2025-01-12:
same error here always

dFxed, 2025-01-15:
Same error (crc32), 
z2m 2.0.2
Nodon modules :
SIN-4-RS-20
SIN-4-FP-21

steve41200, 2025-01-15:
I've try to reload a Zm2Edge older (1.40) and the OTA is working

XAV59213, 2025-01-17:
Même problème de mon côté sur les modules SIN-4-FP-21

Chartreusito, 2025-01-17:
I'm at 2.0.0 z2m version. Using the "check all" button clears the possibility of update on problematic modules.

Lucky-Strike1695, 2025-01-19:
Same issue.
I have 2 SIN-4-RS-20 on 3.3.0-1.3.0
At first, both were marked for update, but both failed with CRC-32 error.
After hitting "search for update" again, now : 
- One is marked for update, but keeps giving CRC-32 error
- The other says "firmware is newer than update image"

This has no sense :)

XAV59213, 2025-01-19:
since 1 hour he has successfully carried out the update

tobzsc, 2025-01-20:
I am having the same issue using latest Z2M 2.0.0-2.

```
info 2025-01-20 12:03:23zhc:ota: [0x286847fffe2e0524 | SIN-4-RS-20] Getting latest image, meta: {"fileName":"128b-0009-0305-700_nodon_sin_4_rs_20_fw_V0305.zigbee","fileVersion":773,"fileSize":398752,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/NodOn/128b-0009-0305-700_nodon_sin_4_rs_20_fw_V0305.zigbee","imageType":9,"manufacturerCode":4747,"sha512":"88bdfeadcbae70c3fe541f98af1dfbd9b437600ede7d7d27ca8e21a5b6e86dc8fa6ffbf9921cafda9a0f19809e909d1126b71cc221b53e39f5f62d252e4e407e","otaHeaderString":"nodon_sin_efr32_ota\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}
info 2025-01-20 12:03:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Büro Rolladen', payload '{"calibration_rotation_run_time_down":null,"calibration_rotation_run_time_up":null,"calibration_vertical_run_time_down":null,"calibration_vertical_run_time_up":null,"cover_mode":null,"linkquality":96,"position":100,"state":"OPEN","tilt":0,"update":{"installed_version":768,"latest_version":773,"state":"available"}}'
info 2025-01-20 12:03:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Büro Rolladen/mode', payload '{"cover":null}'
info 2025-01-20 12:03:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'Büro Rolladen' failed (Image CRC-32 is invalid)","status":"error","transaction":"kcve2-21"}'
error 2025-01-20 12:03:23z2m: Update of 'Büro Rolladen' failed (Image CRC-32 is invalid)
```

bvalat, 2025-01-21:
I have the same problem  for a brand new Nodon SIN-4-FP-21

```
info 2025-01-21 19:32:36z2m: Updating 'Radiateur Cuisine' to latest firmware
info 2025-01-21 19:32:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Radiateur Cuisine', payload '{"energy":0.163,"last_seen":"2025-01-21T19:32:36+01:00","linkquality":132,"pilot_wire_mode":"comfort_-2","power":0,"state":"ON","update":{"installed_version":771,"latest_version":773,"state":"available"}}'
info 2025-01-21 19:32:38z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Radiateur Cuisine', payload '{"energy":0.163,"last_seen":"2025-01-21T19:32:38+01:00","linkquality":128,"pilot_wire_mode":"comfort_-2","power":0,"state":"ON","update":{"installed_version":771,"latest_version":773,"state":"available"}}'
info 2025-01-21 19:32:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Radiateur Cuisine', payload '{"energy":0.163,"last_seen":"2025-01-21T19:32:38+01:00","linkquality":128,"pilot_wire_mode":"comfort_-2","power":0,"state":"ON","update":{"installed_version":771,"latest_version":773,"state":"available"}}'
info 2025-01-21 19:32:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'Radiateur Cuisine' failed (Image CRC-32 is invalid)","status":"error","transaction":"yvlbi-2"}'
error 2025-01-21 19:32:39z2m: Update of 'Radiateur Cuisine' failed (Image CRC-32 is invalid)
```

allistermaguire, 2025-01-23:
I have the same issue but for different devices; mine is with Schneider Electric. I have added some since upgrading to v2.0.0 and trying to update I get the Image CRC-32 is invalid. I added some a couple of weeks ago while running 1.42.0, and they updated fine.

furmek, 2025-01-23:
> I have the same issue but for different devices; mine is with Schneider Electric. I have added some since upgrading to v2.0.0 and trying to update I get the Image CRC-32 is invalid. I added some a couple of weeks ago while running 1.42.0, and they updated fine.

@allistermaguire - thank you!
I was trying to update one of those SE CH10AX switches with no luck.
Your reply gave me an idea: run [1.42.0](https://hub.docker.com/layers/koenkk/zigbee2mqtt/1.42.0/images/sha256-a929a7a9c157a7dcc8ae0bfa955ff810a7e46c056e82ee62c59b478ed20ec3a3) in docker and update from there - that worked!
I know this doesn't scale well, now I need to join that switch to the main zigbee network... manually but at least it is updated.

steve41200, 2025-01-23:
Leave network and join the main after update device is better way no?

jbjardine, 2025-02-02:
Same issue here with 2.1.0 version (OTA update for SIN-4-FP-21) :
```
info 2025-02-02 13:55:26z2m: Updating 'Radiateur_bureau' to latest firmware
info 2025-02-02 13:55:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Radiateur_bureau', payload '{"energy":18.24,"linkquality":200,"pilot_wire_mode":"off","power":0,"state":"ON","update":{"installed_version":771,"latest_version":773,"state":"available"},"update_available":false}'
info 2025-02-02 13:55:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'Radiateur_bureau' failed (Image CRC-32 is invalid)","status":"error","transaction":"h0ym3-3"}'
error 2025-02-02 13:55:28z2m: Update of 'Radiateur_bureau' failed (Image CRC-32 is invalid)
info 2025-02-02 13:55:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Radiateur_bureau', payload '{"energy":18.24,"linkquality":200,"pilot_wire_mode":"off","power":0,"state":"ON","update":{"installed_version":771,"latest_version":773,"state":"available"},"update_available":false}'
```

nerdiges, 2025-02-02:
Same issue.
I have 6 SIN-4-RS-20 on 3.3.0-1.1.2.

tobzsc, 2025-02-04:
I can confirm. Still no fix. Question is what the issue exactly means. Is the update image corrupt or is the CRC used for checking simply wrong.

PtiBed0, 2025-03-11:
Hello,

For information, my Nodon SIN-4-1-21 update successful today (Z2M version 2.1.3-1)

Sincerely

Rykian, 2025-03-11:
Same here, updating works since the last Z2M update.

Nardol, 2025-03-11:
Same here, I mentioned it in the opened issue on zigbee-OTA repository but not here, OTA works now.

jbjardine, 2025-03-15:
Same here, updating works again.

closed 2025-05-23 by github-actions[bot]