Koenkk/zigbee2mqtt #25700: 502: Bad Gateway - not working after updating to 2.0.0-2

PGLDK, 2025-01-09:
### What happened?

After updating to 2.0.0-2 the addon dashboard says 502: Bad Gateway ( The add-on seems to not be ready, it might still be starting. Do you want to try again?"

Everything is starting correctly and the breaking changes has been followed.

### What did you expect to happen?

Frontend starting. due to the debug log every attached zigbee devices are connected, but listed as unavailable in HA

zigbee2mqtt/configuration.yaml file

`devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  user: mqtt_user
  password: mqtt
  server: mqtt://core-mosquitto:1883
serial:
  adapter: zstack
  port: /dev/ttyUSB0
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
  pan_id: 6754
  channel: 25
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  transmit_power: 20
device_options:
  legacy: false
blocklist: []
passlist: []
queue: {}
frontend:
  enabled: true
  port: 8099
availability:
  enabled: false
version: 4
`

### Zigbee2MQTT version

2.0.0-2 

### Adapter firmware version

unknown (cant find the build)

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Add-on in home assistant with mqtt plugin Generic x86-64

### Debug log

    at /app/lib/extension/frontend.ts:132:52
    at new Promise (<anonymous>)
    at Frontend.stop (/app/lib/extension/frontend.ts:132:15)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.stop (/app/lib/controller.ts:233:9)
    at stop (/app/index.js:172:5)
    at process.handleQuit (/app/index.js:180:13))
[10:22:06] INFO: Preparing to start...
[10:22:06] INFO: Socat not enabled
[10:22:07] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-09 10:22:15] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-09.10-00-47'
[2025-01-09 10:22:15] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-09 10:22:15] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-09 10:22:15] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-09 10:22:15] debug: 	z2m: sd-notify loaded
[2025-01-09 10:22:15] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-09 10:22:15] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[25],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_36ec0893c119ec11be667fe5f01c6278-if00-port0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false,\"transmitPower\":20}}"'
[2025-01-09 10:22:15] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"36ec0893c119ec11be667fe5f01c6278","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_36ec0893c119ec11be667fe5f01c6278-if00-port0","vendorId":"10c4","productId":"ea60"}]
[2025-01-09 10:22:15] debug: 	zh:adapter:discovery: Unable to match USB adapter: zstack | /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_36ec0893c119ec11be667fe5f01c6278-if00-port0
[2025-01-09 10:22:15] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_36ec0893c119ec11be667fe5f01c6278-if00-port0","adapter":"zstack"},"adapter":{"disableLED":false,"transmitPower":20},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-09 10:22:15] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_36ec0893c119ec11be667fe5f01c6278-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-01-09 10:22:15] info: 	zh:zstack:znp: Serialport opened

PGLDK, 2025-01-09:
Just updated firmware from 20220219 ->> 20240710 - no luck....
No errors in log file....
Backup of previous version is working fine.

djurny, 2025-01-09:
Hi @PGLDK,
What dongle are you using? And can you post a screenshot or properly formatted markdown of your `configuration.yaml` file?
Grt,

PGLDK, 2025-01-09:
> Hi @PGLDK, What dongle are you using? And can you post a screenshot or properly formatted markdown of your `configuration.yaml` file? Grt,

Hey.  it is a Sonoff_Zigbee_3.0_USB_Dongle_Plus

`devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  user: mqtt_user
  password: mqtt
  server: mqtt://core-mosquitto:1883
serial:
  port: /dev/ttyUSB0
  adapter: zstack
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: error
  pan_id: 6754
  channel: 25
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  transmit_power: 20
device_options:
  legacy: false
blocklist: []
passlist: []
queue: {}
frontend:
  enabled: true
  port: 8099
availability:
  enabled: false
version: 4
`

PGLDK, 2025-01-09:
```
devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  user: mqtt_user
  password: mqtt
  server: mqtt://core-mosquitto:1883
serial:
  port: /dev/ttyUSB0
  adapter: zstack
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: error
  pan_id: 6754
  channel: 25
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  transmit_power: 20
device_options:
  legacy: false
blocklist: []
passlist: []
queue: {}
frontend:
  enabled: true
  port: 8099
availability:
  enabled: false
version: 4

```

djurny, 2025-01-09:
> Hey. it is a Sonoff_Zigbee_3.0_USB_Dongle_Plus

Hi @PGLDK,
Following logging might suggest that the wrong `adapter` type is configured:
```
[2025-01-09 10:22:15] debug: zh:adapter:discovery: Unable to match USB adapter: zstack | /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_36ec0893c119ec11be667fe5f01c6278-if00-port0
```

For your dongle, `Sonoff_Zigbee_3.0_USB_Dongle_Plus` can either be a
- Sonoff Zigbee 3.0 USB Dongle Plus-**P** or 
- Sonoff Zigbee 3.0 USB Dongle Plus-**E**

For the **P** version, you need `adapter: zstack` and for the **E** version, you need `adapter: ember`.

Grt,

Nerivec, 2025-01-09:
The firmware versions mentioned above are for zStack, so, `zstack` should be correct.
It seems it's hanging when first trying to talk to the adapter (right after the serial port is opened).
Can you try rebooting HA (full reboot)?
If it still doesn't work after, shutdown the Z2M add-on (if still trying to start), unplug the adapter for a few minutes, and then re-plug the adapter and try to start Z2M again.

PGLDK, 2025-01-10:
> Hi @PGLDK, Following logging might suggest that the wrong `adapter` type is configured:
>
> ```
> [2025-01-09 10:22:15] debug: zh:adapter:discovery: Unable to match USB adapter: zstack | [...]

Just tried that, but the "Ember" adapter just crashes.
`[2025-01-10 19:15:08] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-10 19:15:08] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-10 19:15:08] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-10 19:15:08] error: 	z2m: Exiting...
[2025-01-10 19:15:09] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)`

PGLDK, 2025-01-10:
> The firmware versions mentioned above are for zStack, so, `zstack` should be correct. It seems it's hanging when first trying to talk to the adapter (right after the serial port is opened). Can you try rebooting HA (full reboot)? If it still doesn't work after, shutdown the Z2M add-on (if still trying to start), unplug the adapter for a few minutes, and then re-plug the adapter and try to start Z2M again.
> 
> If it still doesn't work, please post the content of the configuration on the HA add-on configuration page (settings there take precedence over the same configs in the yaml file).

Thanks for your reply, i tried rebooting several times, no luck. Running with no adaptor at all gives the following error:

`[2025-01-10 19:19:01] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-10 19:19:01] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-10 19:19:01] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-10 19:19:01] error: 	z2m: Exiting...
[2025-01-10 19:19:01] error: 	z2m: Error: Error: No such file or directory, cannot open /dev/ttyUSB0`

Unplugging the adaptor and starting Z2M again - same, not starting. This is really weird because version (1.42.0-2) is just working like a charm..........

[image]

djurny, 2025-01-10:
Hi @PGLDK,
Seems that you are getting different errors; once it appears to start but the add-on states a 502 when trying to access the web UI, second it seems herdsman cannot locate `/dev/ttyUSB0`. Based on @Nerivec comment and your results, `adapter` should be `zstack`.

To avoid mixing up failure modes, can you set the `log_level` to `debug` and try again?
Please post the log and both `configuration.yaml` and add-on screenshot.
Thanks,
Groetjes,

PGLDK, 2025-01-11:
After trying to update for what feels like 20 times, it suddenly works? Even with the old `configuration.yaml` without specifying an adapter or anything... which is really strange. The only changes I made before updating this time were adding a Zigbee device back after a reset (it had disappeared from my Z2M) and updating a Hue device that had available updates. I have no idea why it suddenly works or if there's any connection. So, I’ll close the thread again. Thank you very much for your help!

closed 2025-01-11 by PGLDK