Koenkk/zigbee2mqtt #25701: Aqara RTCGQ11LM Motion sensor no reporting lux correctly

Systemedic, 2025-01-09:
### What happened?

Since the update to 2.0 and using the new entity "illuminance" instead of "illuminance_lux" the lux value is not reported correctly anymore. The current lux value is set to zero, even when there is lightning around the device. I tried repairing the sensor but this doesn't work. It also seems like that there is huge delay in the reporting status of the device. How can I fix this?

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20210120

### Adapter

Electrolama  zzh Multiprotocol RF Stick (CC2652R1)

### Setup

Add-on on Home Assistant

mrmaximas, 2025-01-09:
Hi! Please read this https://github.com/Koenkk/zigbee2mqtt/discussions/24198

Systemedic, 2025-01-09:
@mrmaximas Done. Like I said in my message; I already changed the entity to use the correct one.
[image]

But there is a very long delay in showing the correct value

closed 2025-03-18 by github-actions[bot]