Koenkk/zigbee2mqtt #25702: [New device support]: Tuya TS0601 Curtain Motor _TZE204_nladmfvf Not supported

oig123, 2025-01-09:
### Link

https://www.alibaba.com/product-detail/TYSH-Zigbee-Wifi-Smart-Curtain-Motor_1600738285450.html?spm=a2756.order-detail-ta-ta-b.0.0.5a2df19coSJFBr

### Database entry

Device '0xa4c1384a02487b3c' with Zigbee model 'TS0601' and manufacturer name '_TZE204_nladmfvf' is NOT supported

### Zigbee2MQTT version

Zigbee2MQTT version 2.0.0 

### Comments

I can't add it, please help me.
[image]
[image]
[image]
I contacted the seller and he told me this module is in the motor.
[image]

### External definition

```shell
const definition = {     zigbeeModel: ['TS0601'],     model: 'TS0601',     vendor: '_TZE204_nladmfvf',     description: 'Automatically generated definition',     extend: [],     meta: {}, };  module.exports = definition;
```

### What does/doesn't work with the external definition?

Works well on Tuya's Zigbee 3.0 hub

Chartreusito, 2025-01-10:
Please have a look at the documentation here to try adding it yourself : https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html
We won't be able to help you as we don't own this specific device.
I encourage you to try as you'll be able to add any other unsupported device yourself.
You can find inspiration in the main tuya converter if there is already similar devices existing : https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts

oig123, 2025-01-11:
> გთხოვთ, გადახედოთ აქ დოკუმენტაციას, რომ სცადოთ მისი დამატება: https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html ჩვენ ვერ დაგეხმარებით, რადგან ჩვენ არ გვაქვს ამ კონკრეტულ მოწყობილობას. გირჩევთ, სცადოთ, რადგან თქვენ თავად შეძლებთ დაამატოთ ნებისმიერი სხვა მხარდაჭერილი მოწყობილობა. თქვენ შეგიძლიათ იპოვოთ შთაგონება მთავარ tuya კონვერტორში, თუ უკვე არსებობს მსგავსი მოწყობილობები: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts

[https://github.com/Koenk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts) There is no "_TZE204_nladmfvf" here, so I can't add it?

Chartreusito, 2025-01-11:
Please read the generic documentation : https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

You have to create an external converter with which you'll be able to test your device. Take inspiration in the main tuya.js with similar devices

martinmts, 2025-02-16:
Hi, have you managed to get it resolved?
I have exactly the same device and I don't know how to add it. 
I have no programming background. I think I should be able to make use of existing template and change keywords such as model name, vendor etc., but I can't find a good template yet. The `tuya.ts` file is over my head unless I can see some video tutorials on how to make use of it.

Where should I put [the external definition](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition) file?
Where should I put the [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters)? folder?

I have tried to put them in `zigbee2mqtt` or `zigbee2mqtt/data`, but nothing happened...

oig123, 2025-02-21:
> Hi, have you managed to get it resolved? I have exactly the same device and I don't know how to add it. I have no programming background. I think I should be able to make use of existing template [...]

I couldn't connect
I don't know and no one helped me.

martinmts, 2025-03-09:
Hi, I'm here to report that the following curtain motor is now shown as supported and I can indeed control it from HA.
```
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE204_nladmfvf
```

[image]

closed 2025-05-17 by github-actions[bot]