Koenkk/zigbee2mqtt #25704: Ikea E2123 leaving network after successfull pairing

senna1992, 2025-01-09:
### What happened?

I tried to pair my new remote, but it leaves right after being detected and configured successfully.

### What did you expect to happen?

Stay connected

### How to reproduce it (minimal and precise)

Pair the remote

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

8.0.2

### Adapter

HA yellow

### Setup

HA Yellow

### Debug log

Device has left the network

Chartreusito, 2025-01-10:
> Fix: pressing button "play" multiple times while interviewing. Might be worth adding to wiki if others experience the same issue / solution?

You can edit that yourself [right here](https://github.com/Koenkk/zigbee2mqtt.io/edit/master/docs/devices/E2123.md) ;)

senna1992, 2025-01-10:
> > Fix: pressing button "play" multiple times while interviewing. Might be worth adding to wiki if others experience the same issue / solution?
> 
> You can edit that yourself [right here](https://github.com/Koenkk/zigbee2mqtt.io/edit/master/docs/devices/E2123.md) ;)

Thanks, didn't know that.
Did add my comment :)

closed 2025-03-19 by github-actions[bot]