Koenkk/zigbee2mqtt #25710: Docker pull latest certificate error

nottledim, 2025-01-09:
### What happened?

```docker pull koenkk/zigbee2mqtt:latest```

```
Error response from daemon: Get "https://registry-1.docker.io/v2/": tls: failed to verify certificate: x509: certificate is valid for 76a02b8f34b0a7a5a5dcd164de1ae295.gr7.us-east-1.eks.amazonaws.com, ip-172-16-104-219.ec2.internal, kubernetes, kubernetes.default, kubernetes.default.svc, kubernetes.default.svc.cluster.local, not registry-1.docker.io
```
I understand what's wrong.   This situation is not improved by full system update and reboot.  This is ArchLinux.

### What did you expect to happen?

I expected no error

### How to reproduce it (minimal and precise)

See above

### Zigbee2MQTT version

n/a

### Adapter firmware version

n/a

### Adapter

Sonoff

### Setup

Docker

### Debug log

See above

Nerivec, 2025-01-09:
Looks like an outage of some kind
https://www.dockerstatus.com/

closed 2025-01-09 by Nerivec

nottledim, 2025-01-10:
Thanks for responding especially since it's not a z2m issue.  It's not a docker issue either as I can pull from other systems on my network.  The difference seems to be that this particular system is connected to internet via 3 mobile (Three UK). I've no real proof but I'm wondering if they have some sort of proxy or NAT which is preventing setting up the https connection to docker.  Not your problem - My quest goes on.  Thanks again.

Nerivec, 2025-01-10:
> [Investigating] We're investigating reports of failed pushes and pulls to Docker Registry
> [Monitoring] We found an issue with a broken certificate store. This has been rolled back and error rates have returned to normal. We will continue to investigate and monitor.

The resolved issue on dockerstatus was pretty spot on with the error though, weird coincidence?