Koenkk/zigbee2mqtt #25711: [New device support]: Hue White and Color Ambiance Bulb

MonsyDK, 2025-01-09:
### Link

https://www.philips-hue.com/da-dk/p/hue-white-and-color-ambiance-kerte-e14-pare-1-pak/8719514356610#specifications

### Database entry

[2025-01-09 20:26:43] info: 	zh:controller: Interview for '0x001788010d915421' started [2025-01-09 20:26:43] info: 	z2m: Starting interview of '0x001788010d915421' [2025-01-09 20:26:43] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010d915421","ieee_address":"0x001788010d915421","status":"started"},"type":"device_interview"}' [2025-01-09 20:27:34] info: 	z2m: Zigbee: allowing new devices to join. [2025-01-09 20:27:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"719pm-6"}' [2025-01-09 20:27:43] error: 	zh:controller: Interview failed for '0x001788010d915421 with error 'Error: Interview failed because can not get node descriptor ('0x001788010d915421')' [2025-01-09 20:27:43] error: 	z2m: Failed to interview '0x001788010d915421', device has not successfully been paired

### Zigbee2MQTT version

2.0.0

### Comments

I don't know how this works, but tried my best on getting all info noted.

### External definition

```shell
const definition = {
    zigbeeModel: [''],
    model: '',
    vendor: '',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

It doesn't give any information

MonsyDK, 2025-01-09:
model no. 929002294204

gvlekke, 2025-02-25:
maybe related? https://github.com/Koenkk/zigbee2mqtt/issues/25438

thepiecesofme, 2025-05-02:
I’m having the same issue with some of my Hue White & Color Bulbs. They simply won’t complete the pairing process. Others seem to work without fail. Using the latest firmware version from Philips. Have tried multiple times to pair. Reset each time and paired again with Hue hub in order to be able to later delete it (which resets the bulb and puts it into pairing mode). 

Using ember driver with SLZB-06Mg24.

closed 2025-07-09 by github-actions[bot]