Koenkk/zigbee2mqtt #25713: Since update to V 2.0 it no longer works with Zigbee coordinator (CC2652P7) from cod.m

FloX5E70, 2025-01-09:
### What happened?

Error while starting Zigbee2MQTT-Addon in Home Assistant witch LAN/Zigbee Gateway with TI CC2652P7 . No problems before last major update. IP/Port-Config correct, no changes.

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20240710

### Adapter

cod.m zigbee coordinator, ESP32/CC2652P7

### Setup

VM, Synology, Home Assistant

### Debug log

`Starting Zigbee2MQTT without watchdog.
[2025-01-09 22:37:44] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-09 22:37:44] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-09 22:37:44] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-09 22:37:44] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-09 22:37:44] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-09 22:37:44] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-09 22:37:44] error: 	z2m: Exiting...
[2025-01-09 22:37:44] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)`

geni0602, 2025-01-09:
what´s your adapter config? Did you try to add **adapter: zstack** in config.yaml?
For example
serial:
  port: tcp://IP
  adapter: zstack

FloX5E70, 2025-01-09:
Perfect, thanks! "adapter: zstack" works. Was not needed in previous versions...

closed 2025-01-09 by FloX5E70

xbmcnut, 2025-01-13:
N.B This affects the ZigStar UZG-01 too (and all others I'm assuming)