Koenkk/zigbee2mqtt #25715: Failed to call 'Frontend'

mcneilc, 2025-01-10:
### What happened?

Restarted previously working Zigbee2MWTT add on in Home Assistant. Z2M either halts on the error message below or hard crashes

### What did you expect to happen?

Z2M to load up without errors

### How to reproduce it (minimal and precise)

Start Z2M. No changes were made to a previously working set up

### Zigbee2MQTT version

2.0.0.2

### Adapter firmware version

2.6.8.dev26

### Adapter

SLZB-06

### Setup

Home Assistant Add On

### Debug log

[07:59:36] INFO: Preparing to start...
[07:59:36] INFO: Socat not enabled
[07:59:36] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-10 08:00:52] error: 	z2m: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'close')
    at /app/lib/extension/frontend.ts:132:52
    at new Promise (<anonymous>)
    at Frontend.stop (/app/lib/extension/frontend.ts:132:15)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.stop (/app/lib/controller.ts:233:9)
    at stop (/app/index.js:172:5)
    at process.handleQuit (/app/index.js:180:13))

mcneilc, 2025-01-10:
According to the debugs, Z2M is running and collecting data from zigbee devices but it's not passing the data on to Home Assistant. Whether this is related to the Frontend issue, I don't know.

hemandk, 2025-01-11:
I have the same issue, Z2M add-on is running and connected to my Zigstar, but frontend fails to load and it is not possible to access any Zigbee devices in HA.

Koenkk, 2025-01-11:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

mcneilc, 2025-01-11:
[45df7312_zigbee2mqtt_2025-01-11T13-57-46.767Z.log](https://github.com/user-attachments/files/18386077/45df7312_zigbee2mqtt_2025-01-11T13-57-46.767Z.log)

Koenkk, 2025-01-12:
Could you provide the debug log directly from starting z2m?

mcneilc, 2025-01-12:
I could’ve sworn those logs included that info. Unfortunately I don’t have access to the logs anymore as I restored to a backup then ended up installing the Z2M Edge version of the Home Assistant Add On. Happily for me, this fixed the issue

hemandk, 2025-01-13:
I can install 2.0 again and add full debug log later today if that helps?

manuelkue, 2025-01-14:
Ok, after enabling logs in `homeassistant/zigbee2mqtt/configuration.yaml` it worked. Enabling the logs in the frontend configuration did not do anything o.O

Logs containing the error `z2m: Failed to call 'Frontend' 'stop'`:
[log.log](https://github.com/user-attachments/files/18409944/log.log)

maku-eso, 2025-01-15:
Same issue on my updated z2m install.
`Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'close')`

manuelkue, 2025-01-16:
@mcneilc can you try removing the entries of your zigbee database.db device by device and look when/if the z2m frontend starts again?

Possibly you also have a device type in your database that prevents the frontend from starting like in my case the Popp thermostats.

dommolHA, 2025-01-16:
Hi Folks,
Same problem here - I have 4 such devices [Danfoss Ally](https://www.zigbee2mqtt.io/devices/014G2461.html#danfoss-014g2461) (one of the clone of Popp) and after removing 1 of them from database.db FE became accessible.

What I noticed? In the same time, when the problems started, this device switched to mounting state (M on device) and lost connection.

manuelkue, 2025-01-16:
@dommolHA Oh nice, at your setup it works with the 3 other Danfoss thermostats, though? So only one of the 4 devices caused z2m to break?
Could you compare the database-entries and find differences in them? Maybe it's some malformed data or some attribute that's incompatible.
Does z2m break when you copy this deviations to the non-breaking thermostats?

dommolHA, 2025-01-16:
@manuelkue yes, it works with these 3 Danfoss thermostats

I wasn't able to find differences, these database-entries look similar for me:
- problematic one:
```
{"id":24,"type":"EndDevice","ieeeAddr":"0xMACmacMAC","nwkAddr":33800,"manufId":4678,"manufName":"Danfoss\u0000","powerSource":"Battery","modelId":"eTRV0103","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,10,32,513,516,2821],"outClusterList":[0,25],"clusters":{"genBasic":{"attributes":{"modelId":"eTRV0103","manufacturerName":"Danfoss\u0000","powerSource":3,"zclVersion":3,"appVersion":28,"stackVersion":8,"hwVersion":69,"dateCode":"20241126","swBuildId":"00.28.0008 00.28"}},"genPollCtrl":{"attributes":{"checkinInterval":1200}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":170}},"hvacThermostat":{"attributes":{"danfossLoadEstimate":529,"localTemp":2121,"pIHeatingDemand":28,"danfossPreheatStatus":0,"danfossWindowOpenFeatureEnable":1,"danfossWindowOpenExternal":0,"danfossDayOfWeek":4,"danfossTriggerTime":660,"danfossAlgorithmScaleFactor":1,"danfossHeatAvailable":1,"danfossMountedModeControl":0,"danfossMountedModeActive":0,"danfossExternalMeasuredRoomSensor":-8000,"danfossRadiatorCovered":0,"danfossLoadBalancingEnable":1,"danfossLoadRoomMean":-8000,"danfossAdaptionRunControl":0,"danfossAdaptionRunSettings":1,"systemMode":4,"danfossWindowOpenInternal":1,"occupiedHeatingSetpoint":2100,"danfossHeatRequired":1,"danfossAdaptionRunStatus":2,"setpointChangeSource":0}},"hvacUserInterfaceCfg":{"attributes":{"keypadLockout":0}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0xMACmacMAC","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xMACmacMAC","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0xMACmacMAC","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0},{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":8,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":16402,"minRepIntval":60,"maxRepIntval":65000,"repChange":1,"manufacturerCode":4678},{"cluster":513,"attrId":16384,"minRepIntval":60,"maxRepIntval":3600,"repChange":1,"manufacturerCode":4678},{"cluster":513,"attrId":16433,"minRepIntval":60,"maxRepIntval":3600,"repChange":1,"manufacturerCode":4678},{"cluster":513,"attrId":16405,"minRepIntval":60,"maxRepIntval":65000,"repChange":1,"manufacturerCode":4678},{"cluster":513,"attrId":16461,"minRepIntval":60,"maxRepIntval":3600,"repChange":1,"manufacturerCode":4678},{"cluster":513,"attrId":16463,"minRepIntval":60,"maxRepIntval":65000,"repChange":1,"manufacturerCode":4678}],"meta":{}}},"appVersion":28,"stackVersion":8,"hwVersion":69,"dateCode":"20241126","swBuildId":"00.28.0008 00.28","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1977182463},"lastSeen":1736098332039,"checkinInterval":300}
``` 
- non-problematic ones:
```
{"id":25,"type":"EndDevice","ieeeAddr":"0xMACmacMAC","nwkAddr":62015,"manufId":4678,"manufName":"Danfoss\u0000","powerSource":"Battery","modelId":"eTRV0103","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,10,32,513,516,2821],"outClusterList":[0,25],"clusters":{"genPollCtrl":{"attributes":{"checkinInterval":1200}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":158}},"hvacThermostat":{"attributes":{"danfossMountedModeActive":0,"localTemp":2188,"danfossWindowOpenFeatureEnable":1,"danfossWindowOpenExternal":0,"danfossDayOfWeek":4,"danfossTriggerTime":660,"danfossAlgorithmScaleFactor":1,"danfossHeatAvailable":1,"danfossMountedModeControl":0,"danfossExternalMeasuredRoomSensor":-8000,"danfossRadiatorCovered":0,"danfossLoadBalancingEnable":1,"danfossLoadRoomMean":-8000,"danfossAdaptionRunControl":0,"danfossAdaptionRunSettings":1,"systemMode":4,"danfossLoadEstimate":556,"danfossHeatRequired":1,"danfossWindowOpenInternal":1,"danfossPreheatStatus":0,"danfossAdaptionRunStatus":2,"occupiedHeatingSetpoint":2200,"setpointChangeSource":2,"pIHeatingDemand":31}},"hvacUserInterfaceCfg":{"attributes":{"keypadLockout":0}},"genBasic":{"attributes":{"modelId":"eTRV0103","manufacturerName":"Danfoss\u0000","powerSource":3,"zclVersion":3,"appVersion":28,"stackVersion":8,"hwVersion":69,"dateCode":"20230928","swBuildId":"00.28.0008 00.28"}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0xMACmacMAC","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xMACmacMAC","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0xMACmacMAC","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0},{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":8,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":16402,"minRepIntval":60,"maxRepIntval":65000,"repChange":1,"manufacturerCode":4678},{"cluster":513,"attrId":16384,"minRepIntval":60,"maxRepIntval":3600,"repChange":1,"manufacturerCode":4678},{"cluster":513,"attrId":16433,"minRepIntval":60,"maxRepIntval":3600,"repChange":1,"manufacturerCode":4678},{"cluster":513,"attrId":16405,"minRepIntval":60,"maxRepIntval":65000,"repChange":1,"manufacturerCode":4678},{"cluster":513,"attrId" [... cut ...] :65000,"repChange":1,"manufacturerCode":4678},{"cluster":513,"attrId":16461,"minRepIntval":60,"maxRepIntval":3600,"repChange":1,"manufacturerCode":4678},{"cluster":513,"attrId":16463,"minRepIntval":60,"maxRepIntval":65000,"repChange":1,"manufacturerCode":4678}],"meta":{}}},"appVersion":28,"stackVersion":8,"hwVersion":69,"dateCode":"20230928","swBuildId":"00.28.0008 00.28","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1977182463},"lastSeen":1736979935944,"checkinInterval":300}
```

manuelkue, 2025-01-17:
@dommolHA do you probably use a ConBee adapter as well? E.g. a ConBee II / III?
In the next few days I will receive a SLZB-07p7 and will try if the thermostats work with it.

Temporary fix:
Remove the affected devices, start z2m and re-pair them. Sideeffect: Next restart of z2m will crash again.

dommolHA, 2025-01-18:
@manuelkue ConBee II, SLZB-06 is waiting for replacement...

>> into row with data with "id":26 - without success, FE doesn't start.

>Do you mean by that, when you just add the "nwkAddr":33800 to the previously working entry z2m won't start afterwards?

I replaced "ieeeAddr" too.

One more thing about these 3 devices - I was able to start FE, but not able to set temperature, gauges were greyed out.

I fixed it by removing all of these devices and re-pairing. FE doesn't crash after restarts.

10Thirty, 2025-01-24:
I'm also experiencing the same issue but with [Hive UK7004240](https://www.zigbee2mqtt.io/devices/UK7004240.html#hive-uk7004240) TRV`s which also show as Danfoss in the database.db.

I currently run 2 Zigbee2MQTT instances, one running the HA addon and one running via docker in unraid, both using UZG-1. The valves were originally on the addon instance a couple of weeks ago which refused to start after rebooting the Pi4 HA is running on. I removed all the valves from that instance and paired them with the one on unraid just to get the heating running again. I manually edited the database.db to remove the valves and the HA addon started without issue and I though "wierd, but ok". The unraid instance had been running fine, even surviving a few restarts due to backups and OS updates. Then last night that instance crashed and refused to start up. I removed them from the database.db and re-paired them last night and they all seemed to work fine after that, then this morning I tried restarting Zigbee2MQTT and it once again hung on startup. Rolled back to 1.42.0 and everything started up fine.

@Koenkk if you still need debug logs I should be able to supply them tomorrow assuming my house survives storm Éowyn and we still have power.

manuelkue, 2025-01-24:
> In the next few days I will receive a SLZB-07p7 and will try if the thermostats work with it.

The SLZB-07p7 arrived, I re-paired all devices including the TRVs and restarted the server multiple times since then. So far, no z2m- crashes happened since the coordinator replacement 🤷‍♂️
But as @10Thirty reported this also happening with another coordinator than the ConBee, I don't have too much trust in this setup yet.

mcneilc, 2025-01-24:
> [@mcneilc](https://github.com/mcneilc) can you try removing the entries of your zigbee database.db device by device and look when/if the z2m frontend starts again?
> 
> (backup your original database somewhere before)
> 
> Possibly you also have a device type in your database that prevents the frontend from starting like in my case the Popp thermostats, mentioned in the above comment.

I ended up moving to Z2M Edge which worked first time so I don't have any logs to provide. Very interesting that it's been narrowed down to Danfoss TRVs as I've got about 12 Hive TRVs plus the Hive thermostat and boiler control which to my knowledge are just the same as Danfoss. Whatever the issue was/ is, it's went away in Edge

HungryZebra563, 2025-01-25:
I have the same issue with Danfoss Ally TRV. Lots of frameparser errors and no frontend. However, in my case the frontend would not be accessible immediately, but somehow it became accessible after 1 day or so. Really strange. 

Moved Z2M Edge, the frontend is immediately available after a restart, but the frameparser errors persist, and the entities are unavailable in home assistant

`[2025-01-25 08:52:56] debug:    zh:deconz:frameparser: DATA_INDICATION RESPONSE - Error: Unsupported response reading for cluster ID '0'.`

spookoly, 2025-01-26:
Same problem for me with lots of errors, no frontend, and no communication with HA or node-red. Also just paired four Danfoss Ally TRVs, but I have a SLZB-06. So the issues doesn't appear to be linked to the coordinator from what others have said.

michaelsidenius, 2025-01-30:
Removed my 10 Danfoss Ally TRVs from the database.db and frontend runs now again. Seems like it stopped during the backup this night.

libussa, 2025-01-30:
SLZB-06M here, same issue.
I have 5 Danfoss TRVs, tried removing one by one, but only one ended up fixing the situation (I re added all but one in the end)

stkr22, 2025-02-01:
At least for me the issues was fixed by updating to latest-dev. I guess the fix by @Koenkk helped.

Koenkk, 2025-02-02:
Should be fixed in z2m 2.1.0 indeed!

closed 2025-02-02 by Koenkk

spookoly, 2025-02-02:
> Should be fixed in z2m 2.1.0 indeed!

Have updated to 2.1.0 and looking great so far. Many thanks!

manuelkue, 2025-02-19:
@dommolHA, @michaelsidenius, @mcneilc, @10Thirty, @spookoly, @libussa, @HungryZebra563

I've experienced the issue so far once again, but it healed itself somehow after some hours of just doing nothing - so the same as @HungryZebra563 commented.

Have you experienced it again after the z2m update?

@ahattink created an issue for another device that has been found to be a trigger for the same problem (not starting z2m, but correct start after removing faulty devices). So if some of you had the problem also once again feel free to join the other issue:
https://github.com/Koenkk/zigbee2mqtt/issues/26430

spookoly, 2025-02-20:
This problem has not reoccurred so far since the update for me. Having some other issues with the Danfoss Ally TRVs (they seem to lose connection for me after a while) but haven’t worked out what’s causing that yet.

#26430<https://github.com/Koenkk/zigbee2mqtt/issues/26430>

#26430<https://github.com/Koenkk/zigbee2mqtt/issues/26430>