Koenkk/zigbee2mqtt #25717: zigbee2mqtt error Failed to call 'Bridge' 'start'

johnzara, 2025-01-10:
### What happened?

After moving my project from a completely working environment on a RockPi4 to a Raspberry Pi 5, I suddenly get this error when running the container. The error shows up if I run the container both via docker compose or the command line.

### What did you expect to happen?

I expected zigbee2mqtt to start successfully.

### How to reproduce it (minimal and precise)

Start the docker container and see the live logging in Portainer.

# Devices

`dmesg | grep tty
[95523.834750] cdc_acm 3-2:1.0: ttyACM0: USB ACM device
[95527.738890] cdc_acm 3-2:1.0: ttyACM0: USB ACM device
[99386.528721] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[99390.448906] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[148988.253705] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[148992.509935] cdc_acm 1-2:1.0: ttyACM0: USB ACM device`

`ls /dev/serial/by-id/
usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2694991-if00`

# docker-compose.yml:

`services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest
    restart: unless-stopped
    volumes:
      - ./data:/app/data
      - /run/udev:/run/udev
    ports:
      - 8085:8080  # Exposes Zigbee2MQTT frontend on port 8085
    environment:
      - TZ=Europe/Stockholm
    devices:
      - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2694991-if00:/dev/ttyACM0
    networks:
      - mqtt-network  #  already existing network the mqtt broker is also connected to
    cap_add:
      - NET_ADMIN
      - SYS_MODULE

networks:
  mqtt-network:
    external: true`

# configuration.yaml

`homeassistant:
  enabled: false
frontend:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mqtt5:1883
  user: mqtt
  password: mqtt
  version: 4
serial:
  port: /dev/ttyACM0
advanced:
  network_key:
    - 133
    - 114
    - 247
    - 13
    - 234
    - 135
    - 186
    - 72
    - 237
    - 253
    - 183
    - 96
    - 90
    - 156
    - 144
    - 143
  pan_id: 24682
  ext_pan_id:
    - 19
    - 85
    - 101
    - 87
    - 148
    - 173
    - 201
    - 130
version: 4`

### Zigbee2MQTT version

2..0.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Setup

Plain on Docker container

### Debug log

Using '/app/data' as data directory

Starting Zigbee2MQTT without watchdog.

[2025-01-10 09:29:24] info: 	z2m: Logging to console, file (filename: log.log)

[2025-01-10 09:29:24] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a)

[2025-01-10 09:29:24] info: 	z2m: Starting zigbee-herdsman (3.2.1)

[2025-01-10 09:29:24] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE2694991","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2694991-if00","vendorId":"1cf1","productId":"0030"} => deconz: path=/dev/ttyACM0, score=4

[2025-01-10 09:29:25] info: 	z2m: zigbee-herdsman started (resumed)

[2025-01-10 09:29:25] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'

[2025-01-10 09:29:25] info: 	z2m: Currently 0 devices are joined.

[2025-01-10 09:29:25] info: 	z2m: Connecting to MQTT server at mqtt://mqtt5:1883

[2025-01-10 09:29:25] info: 	z2m: Connected to MQTT server

[2025-01-10 09:29:25] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'

[2025-01-10 09:29:25] error: 	z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'getDevice')

    at Bridge.publishInfo (/app/lib/extension/bridge.ts:692:69)

    at Bridge.start (/app/lib/extension/bridge.ts:179:9)

    at Controller.callExtensions (/app/lib/controller.ts:368:17)

    at Controller.start (/app/lib/controller.ts:186:9)

    at start (/app/index.js:161:5))

[2025-01-10 09:29:25] info: 	z2m: Started frontend on port 8080

[2025-01-10 09:29:25] info: 	z2m: Zigbee2MQTT started!

[2025-01-10 09:29:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'

johnzara, 2025-01-10:
Might be worth also mentioning that the frontend shows "unknown" for all the parameters, including the zigbee2mqtt version itself:
[image]

simsod, 2025-01-12:
Just trying to set up z2m and zigbee in general for the first time and yeah, this occurs for me too, Conbee II and z2m running in truenas scale docker.  I´ve downgraded the conbee II to the suggested firmware version from the docs.

`Starting Zigbee2MQTT without watchdog.
[2025-01-12 14:48:13] info:     z2m: Logging to console, file (filename: log.log)
[2025-01-12 14:48:13] info:     z2m: Starting Zigbee2MQTT version 2.0.0 (commit #060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a)
[2025-01-12 14:48:13] info:     z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-12 14:48:13] info:     z2m: zigbee-herdsman started (resumed)
[2025-01-12 14:48:13] info:     z2m: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
[2025-01-12 14:48:13] info:     z2m: Currently 0 devices are joined.
[2025-01-12 14:48:13] info:     z2m: Connecting to MQTT server at mqtt://192.168.1.36:1883
[2025-01-12 14:48:13] info:     z2m: Connected to MQTT server
[2025-01-12 14:48:13] info:     z2m:mqtt: MQTT publish: topic 'zigbee/bridge/state', payload '{"state":"online"}'
[2025-01-12 14:48:13] error:    z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'getDevice')
    at Bridge.publishInfo (/app/lib/extension/bridge.ts:692:69)
    at Bridge.start (/app/lib/extension/bridge.ts:179:9)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))
[2025-01-12 14:48:13] info:     z2m: Started frontend on port 0.0.0.0:31100
[2025-01-12 14:48:13] info:     z2m: Zigbee2MQTT started!
[2025-01-12 14:48:15] info:     z2m:mqtt: MQTT publish: topic 'zigbee/bridge/state', payload '{"state":"online"}'`

johnzara, 2025-01-12:
@simsod that's weird, but thanks for the update! Did you reboot or fix any permissions, or did nothing at all?

johnzara, 2025-01-14:
> Could you provide the debug log of this + your data/database.db?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Here you go. Thank you!
[database.db.txt](https://github.com/user-attachments/files/18407672/database.db.txt)
[z2m-debug.log](https://github.com/user-attachments/files/18407673/z2m-debug.log)

Koenkk, 2025-01-14:
Could you try the following:
- Stop z2m
- Delete the `database.db`
- Start z2m, provide the debug log of this

johnzara, 2025-01-15:
[_zigbee2mqtt_logs(1).txt](https://github.com/user-attachments/files/18420702/_zigbee2mqtt_logs.1.txt)

One step forward at least! Right now I'm able to press "Permit join (all)" and I see the countdown timer. However, I'm still unable to pair any zigbee device.
As a test, I've tried using the same adapter on my Rock Pi 4 (running z2m 1.42.1 on Docker) and everything works flawlessly.

johnzara, 2025-01-17:
Solved! I followed these two steps, not sure about which one really did the fix (part of it still remains a mystery to me). Either way, I did these 2 things:

- Downgraded udev to a previous version (there are known compatibility issues with ConBee II and the latest versions of Debian), following [this guide](https://forum-iobroker-net.translate.goog/topic/65017/zigbee-conbee2-funktioniert-nach-update-nicht-mehr-udev/66?_=1684661924293&_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=wapp&_x_tr_hist=true)
- Connected the ConBee II to USB 3.0 (blue ports) using an extension cord. The extension cord on USB 2.0 did not work AT ALL, and I was so desperate to try out all ports on my Raspberry, so I was really surprised to see that USB 3.0 worked.

closed 2025-01-17 by johnzara