Koenkk/zigbee2mqtt #25718: Making BAC-002-ALZB Thermostats Work Flawlessly with Zigbee2MQTT

kuposcar, 2025-01-10:
### What happened?

After a long and challenging journey, I have finally managed to get my BAC-002-ALZB thermostats working perfectly with Zigbee2MQTT. The default support was far from ideal, and finding the right solution online was like searching for a needle in a haystack.

This post outlines the main issues I faced, the solution I implemented, and acknowledges the amazing people who helped along the way. If you're struggling with these thermostats, this might be just what you need!

Problems I Faced

Programming Schedule Issues: The thermostat’s programming schedule didn't work as expected. The timings would either not save or not trigger correctly.

Heating Functionality: The heating mode didn't operate reliably, causing confusion and frustration.

Incorrect Date (Day of the Week): The day of the week displayed on the thermostat was consistently wrong, affecting time-based features like schedules.

### What did you expect to happen?

The Solution: A Custom Converter

To address these issues, I created a custom converter based on snippets and advice I found online. Here's the converter that fixed all three problems:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{
        modelID: 'TS0601',
        manufacturerName: '_TZE200_dzuqwsyg',
    }],
    model: 'BAC-002-ALZB',
    vendor: 'HKGK',
    description: 'BAC series thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetLocalTime, 
    configure: tuya.configureMagicPacket,
    exposes: [
        e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the thermostat ON/OFF'),
        e.child_lock(),
        e.climate()
            .withLocalTemperature(ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
            .withSystemMode(['cool', 'heat', 'fan_only'], ea.STATE_SET)
            .withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
            .withLocalTemperatureCalibration(-5, 5, 0.5, ea.STATE_SET),
        e.min_temperature().withValueMin(5).withValueMax(15),
        e.max_temperature().withValueMin(15).withValueMax(45),
        e.binary('manual_mode', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Select between Manual (ON) and Auto (OFF) mode'),
        e.deadzone_temperature().withValueMin(0).withValueMax(5).withValueStep(1),
        e.text('schedule', ea.STATE_SET).withDescription(
            `Weekly schedule in the format "HH:MM/TT HH:MM/TT ...".
            Example for 12 segments:
            "06:00/20 11:30/21 13:30/22 17:30/23 06:00/24 12:00/23 14:30/22 17:30/21 06:00/19 12:30/20 14:30/21 18:30/20".
            Each segment contains:
            - HH:MM: Time in 24-hour format.
            - TT: Temperature in °C.
            Ensure all 12 segments are defined and separated by spaces.`),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff],
            [2, 'system_mode', tuya.valueConverterBasic.lookup({'cool': tuya.enum(0), 'heat': tuya.enum(1), 'fan_only': tuya.enum(2)})],
            [4, 'manual_mode', tuya.valueConverter.onOff],
            [16, 'current_heating_setpoint', tuya.valueConverter.raw],
            [19, 'max_temperature', tuya.valueConverter.raw],
            [24, 'local_temperature', tuya.valueConverter.divideBy10],
            [26, 'min_temperature', tuya.valueConverter.raw],
            [27, 'local_temperature_calibration', tuya.valueConverter.localTemperatureCalibration],
            [28, 'fan_mode', tuya.valueConverterBasic.lookup({
                'low': tuya.enum(0),
                'medium': tuya.enum(1),
                'high': tuya.enum(2),
                'auto': tuya.enum(3),
            })],
            [40, 'child_lock', tuya.valueConverter.lockUnlock],
            [101, 'schedule', {
                to: (v, meta) => {
                    const regex = /((?<h>[01][0-9]|2[0-3]):(?<m>[0-5][0-9])\/(?<t>[0-3][0-9](\.[0,5]|)))/gm;
                    const matches = [...v.matchAll(regex)];
                    if (matches.length == 12) {
                        return matches.reduce((arr, m) => {
                            arr.push(parseInt(m.groups.h));
                            arr.push(parseInt(m.groups.m));
                            arr.push(parseFloat(m.groups.t) * 2);
                            return arr;
                        }, []);
                    }
                    meta.logger.warning('Ignoring invalid or incomplete schedule');
                },
                from: (v, meta) => {
                    let r = '';
                    for (let i = 0; i < 12; i++) {
                        r += `${v[i * 3].toString().padStart(2, '0')}:${v[i * 3 + 1].toString().padStart(2, '0')}/${v[i * 3 + 2] / 2}`;
                        if (i < 11) {
                            r += ' ';
                        }
                    }
                    return r;
                },
            }],
            [103, 'deadzone_temperature', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
`

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

__

### Adapter

BAC-002-ALZB

### Setup

Plain on Docker

Koenkk, 2025-01-10:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

kuposcar, 2025-01-24:
> Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Of course,  

Here I attach the Custom-Converter with some corrections.  

I will make the PR including ON/OFF within the climate to align it with the rest of Zigbee2MQTT. In this custom-converter, I’ve kept it separate because I personally prefer it that way.  

Thank you very much for all your work!  

´´´
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const { convertDecimalValueTo4ByteHexArray } = require('zigbee-herdsman-converters/lib/tuya');

// Función personalizada para sumar un día al tiempo
const onEventSetLocalTimePlusOneDay = async (type, data, device, options) => {
    const nextLocalTimeUpdate = globalStore.getValue(device, 'nextLocalTimeUpdate');
    const forceTimeUpdate = nextLocalTimeUpdate == null || nextLocalTimeUpdate < new Date().getTime();

    if ((data.type === 'commandMcuSyncTime' && data.cluster === 'manuSpecificTuya') || forceTimeUpdate) {
        globalStore.putValue(device, 'nextLocalTimeUpdate', new Date().getTime() + 3600 * 1000);
        try {
            let utcTime = Math.round(new Date().getTime() / 1000);
            let localTime = utcTime - new Date().getTimezoneOffset() * 60;

            // Sumar un día (86400 segundos)
            utcTime += 86400;
            localTime += 86400;

            const endpoint = device.getEndpoint(1);
            const payload = {
                payloadSize: 8,
                payload: [...convertDecimalValueTo4ByteHexArray(utcTime), ...convertDecimalValueTo4ByteHexArray(localTime)],
            };

            await endpoint.command('manuSpecificTuya', 'mcuSyncTime', payload, {});
        } catch (error) {
            console.error(`Error al sincronizar el tiempo: ${error.message}`);
        }
    }
};

// Definición combinada
const definition = {
    fingerprint: [{
        modelID: 'TS0601',
        manufacturerName: '_TZE200_dzuqwsyg',
    }],
    model: 'BAC-002-ALZB',
    vendor: 'HKGK',
    description: 'BAC series thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    // Lógica para seleccionar el onEvent según el dispositivo
    onEvent: async (type, data, device, options) => {
        if (device.ieeeAddr === '0x5c0272fffeecc1e9') {
            // Usa la lógica específica de sumar un día
            await onEventSetLocalTimePlusOneDay(type, data, device, options);
        } else {
            // Usa la lógica genérica
            await tuya.onEventSetLocalTime(type, data, device, options);
        }
    },
    configure: tuya.configureMagicPacket,
    exposes: [
        e.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('Turn the thermostat ON/OFF'),
[... 47 lines cut ...]
                            arr.push(parseInt(m.groups.m));
                            arr.push(parseFloat(m.groups.t) * 2);
                            return arr;
                        }, []);
                    }
                    meta.logger.warning('Ignoring invalid or incomplete schedule');
                },
                from: (v, meta) => {
                    let r = '';
                    for (let i = 0; i < 12; i++) {
                        r += `${v[i * 3].toString().padStart(2, '0')}:${v[i * 3 + 1].toString().padStart(2, '0')}/${v[i * 3 + 2] / 2}`;
                        if (i < 11) {
                            r += ' ';
                        }
                    }
                    return r;
                },
            }],
            [103, 'deadzone_temperature', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
´´´

robvanoostenrijk, 2025-01-29:
@kuposcar,

Thank you for your contribution. The existing BAC-002-AZLB converter was updated a while ago by me. 
I'm actually just in the process of improving it to primarily support both 2-pipe and 4-pipe configurations (i.e. only cool or cool/heat mode).

My device is only cooling, so I couldn't test the 4-mode.

In your code I can see the following:

- onEvent changed from `onEventSetTime` to `onEventSetLocalTime`
- Adding back in `state` for on/off. This was actually removed because Home Assistant and Matter do not understand it, they expect it to be part of `system_mode`. Did `off` in `system_mode` not work as expected?
- Adding datapoints `26` and `19` for min and max temperature respectively. I'll double check these against the service menu on the device and if they can be exposed.
- Moving deadzone_temperature to dp `103` from `26`. I'm pretty sure I checked this before, but I'll validate if there is a mistake there indeed.

Let me know if there's anything I missed.

The service menu reference I found (and used previously):

[image]

The changes I am working on can be viewed here:

https://github.com/Koenkk/zigbee-herdsman-converters/compare/master...robvanoostenrijk:zigbee-herdsman-converters:device/tuya-bac-002?expand=1

Changes made:

- Clean-up `state` and `system_mode`, only exposing `system_mode` as [`off`, `cool`, `heat`, `fan_only`].
- Added device specific setting Control sequence of operation, based on [.withControlSequenceOfOperation](https://github.com/Koenkk/zigbee-herdsman-converters/blob/8272fdba00459570534b6a6b262751988aded02d/src/lib/exposes.ts#L675C5-L675C35) with modes [`cooling_only`, `cooling_and_heating_4-pipes'`, `]
- Refactored `schedule` to a composite configuration. Also rewrote schedule parsing and formatting functions to be more readable and re-usable.
- Removed `current_cooling_setpoint`, which was an attempt to fix issues with identification of cooling capability in Matter over Matterbridge. Hopefully configurable `cooling_only` mode will resolve these issues.
- Integrated your change of `onEventSetTime` to `onEventSetLocalTime`, I never use schedules so never noticed this was off. Will double-check this.

Happy to get your feedback to ensure all improvements end upstream.

robvanoostenrijk, 2025-01-29:
I've dumped the zigbee messages from both BAC-002-ALZB  and BAC-003:

```json
{"65506":56,"65508":0,"appVersion":74}

// DP01: on/off
{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":1} // off
{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":1} // on

// DP02: system_mode
{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":2}],"seq":1} // cool
{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":2}],"seq":1} // fan_only

// DP04: preset
{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":4}],"seq":1} // auto
{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":4}],"seq":1} // manual

// DP16: current_heating_setpoint (°C)
{"dpValues":[{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":16}],"seq":1} // 22
{"dpValues":[{"data":{"data":[0,0,0,23],"type":"Buffer"},"datatype":2,"dp":16}],"seq":1} // 23
{"dpValues":[{"data":{"data":[0,0,0,23],"type":"Buffer"},"datatype":2,"dp":16}],"seq":1} // 23.5
{"dpValues":[{"data":{"data":[0,0,0,26],"type":"Buffer"},"datatype":2,"dp":16}],"seq":1} // 26

// DP19: max_temperature (°C)
{"dpValues":[{"data":{"data":[0,0,0,35],"type":"Buffer"},"datatype":2,"dp":19}],"seq":1} // 35
{"dpValues":[{"data":{"data":[0,0,0,36],"type":"Buffer"},"datatype":2,"dp":19}],"seq":1} // 36
{"dpValues":[{"data":{"data":[0,0,0,40],"type":"Buffer"},"datatype":2,"dp":19}],"seq":1} // 40

// DP24: local_temperature (°C * 10)
{"dpValues":[{"data":{"data":[0,0,0,245],"type":"Buffer"},"datatype":2,"dp":24}],"seq":1} // 24.5
{"dpValues":[{"data":{"data":[0,0,0,255],"type":"Buffer"},"datatype":2,"dp":24}],"seq":1} // 25.5

// DP26: deadzone_temperature (°C)
{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":26}],"seq":1}
{"dpValues":[{"data":{"data":[0,0,0,2],"type":"Buffer"},"datatype":2,"dp":26}],"seq":1}

// DP27: local_temperature_calibration (°C)
{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":27}],"seq":1}
{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":27}],"seq":1}

// DP28: fan_mode
{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":28}],"seq":1} // low
{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":28}],"seq":1} // medium
{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":28}],"seq":1} // high
{"dpValues":[{"data":{"data":[3],"type":"Buffer"},"datatype":4,"dp":28}],"seq":1} // auto

// DP40: child_lock
{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":40}],"seq":1} // unlocked
{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":40}],"seq":1} // locked

// DP100: schedule
// Weekdays (1-5): 1: 06:04/27C, 2: 11:30/27C, 3: 13:30/27C, 4: 17:30/27.5C
// Saturday (6): 1: 06:00/27C, 2: 12:00/27C, 3: 14:30/27C, 4: 17:30/27C
// Sunday (7): 1: 06:00/27C, 2: 12:30/27C, 3: 14:30/27C, 4: 18:30/27C
{"dpValues":[{"data":{"data":[6,5,40,11,30,42,13,30,44,17,30,46,6,0,49,12,0,46,14,30,44,17,30,42,6,0,38,12,30,40,14,30,42,18,30,40],"type":"Buffer"},"datatype":0,"dp":101}],"seq":1}
```

I do not see any DP103 traffic and I double checked DP26, it definitely is deadzone_temperature delta on both of my devices.
Interestingly enough many of the service options 12/24h clock, min_temperature and some others do not seem exposed on zigbee at all.

I added max_temperature now for good measure, before I did not because min_temperature is missing.