Koenkk/zigbee2mqtt #25720: TuYa TS0041 _TZ3000_axpdxqgu stopped working after Zigbee2mqtt update 2.0.0

dhde, 2025-01-10:
### What happened?

ll TuYa TS0041 _TZ3000_axpdxqgu stopped working after Zigbee2mqtt 2.0.0 update. My TuYa TS0041 switche are visible, can be added to Z2M and are recognized, but automation is not working anymore. Pressing switches button is not tuning on the light (light can be still switched on or off via HA) . Deleting the automation and creating new one is not resolving the issue, either deleting this device from Z2M and re-adding it. Before latest Z2M update, it was working. Same issue with the latest-dev version.

### What did you expect to happen?

Restore functionality of TuYa TS0041 _TZ3000_axpdxqgu switches.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

docker on x86-64

dhde, 2025-01-10:
There is a similar issue described here, but it occurred with an older version:
https://github.com/Koenkk/zigbee2mqtt/issues/21196

lucgrefte, 2025-01-10:
I have similar issues with three plugs (TS011F_plug_3) and (TS0001_switch_module).

It's random which device works and which doesn't. The devices devices are always visible, data is being reported live on the dashboard, however sometimes the controls simply don't respond to anything and it stays in the current states.

Koenkk, 2025-01-10:
It's likely one of the 2.0 breaking changes:

> - All action sensors are now disabled by default (`sensor.*_action` entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your `configuration.yaml`.
>   ```yaml
>   homeassistant:
>     legacy_action_sensor: true
>   ```

dhde, 2025-01-20:
Ok, works with the legacy_action_sensor.
Is there no other chance to get that working in node-red for example?

closed 2025-02-01 by dhde

gdschut, 2025-02-02:
> Ok, works with the legacy_action_sensor. Is there no other chance to get that working in node-red for example?

I changed the trigger node in node-red to a MQTT node, and bypass HA. Not pretty, but working.