Koenkk/zigbee2mqtt #25721: Firmware is newer than latest available firmware

GigiPompieru, 2025-01-10:
### What happened?

After updating to 2.0.0-2 I went to the OTA page and pressed check all.

I have 10 [Tuya TS011F_plug_1](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f-plug-1).

On 4 of them I have:
```
    "update": {
        "installed_version": 192,
        "latest_version": 80,
        "state": "idle"
    },
```
While the other 6 report correctly that:
```
    "update": {
        "installed_version": 192,
        "latest_version": 192,
        "state": "idle"
    },
```
Since all plugs are reporting the same version installed, but the version they are comparing with seems to differ, I would think this is related to Z2M and what I sees as the latest firmware.

For the 4 in questions the logs show:
```
[2025-01-10 13:40:31] warning: 	zhc:ota: [0xa4c13858738ee4ca | TS011F] Firmware is newer than latest available firmware.
[2025-01-10 13:40:31] warning: 	zhc:ota: [0xa4c13809293f023b | TS011F] Firmware is newer than latest available firmware.
[2025-01-10 13:40:31] warning: 	zhc:ota: [0xa4c1388ef8f03b4d | TS011F] Firmware is newer than latest available firmware.
[2025-01-10 13:40:31] warning: 	zhc:ota: [0xa4c1389073baa81c | TS011F] Firmware is newer than latest available firmware.
```

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

HAOS VM

PotcFdk, 2025-01-13:
This is happening to me as well:

```JSON
{
    "child_lock": "UNLOCK",
    "countdown": 0,
    "current": 0,
    "energy": 1.75,
    "indicator_mode": "off/on",
    "linkquality": 65,
    "power": 0,
    "power_outage_memory": "off",
    "state": "OFF",
    "update": {
        "installed_version": 192,
        "latest_version": 80,
        "state": "idle"
    },
    "voltage": 237
}
```

The device identifies as

```
Zigbee Model
    TS011F
Zigbee Manufacturer
    _TZ3000_fukaa7nc
```

Jeepmg, 2025-01-13:
Same for me with 1 plug of 4 ; model [Tuya TS011F_plug_1]

`{
    "child_lock": "UNLOCK",
    "countdown": 0,
    "current": 0,
    "energy": 0.08,
    "indicator_mode": "off/on",
    "last_seen": "2025-01-13T13:40:27.701Z",
    "linkquality": 182,
    "power": 0,
    "power_outage_memory": "restore",
    "state": "OFF",
    "update": {
        "installed_version": 192,
        **"latest_version": 80,**
        "state": "idle"
    },
    "voltage": 235
}`

While we're on the subject of these sockets, it seems that the Child Lock mode has not been working normally since the V2 update. Lock mode doesn't stay ON

GigiPompieru, 2025-01-14:
> Same for me with 1 plug of 4 ; model [Tuya TS011F_plug_1]
>
> `{ "child_lock": "UNLOCK", "countdown": 0, "current": 0, "energy": 0.08, "indicator_mode": "off/on", "last_seen": [...]

I just tested with one of the plugs that sees latest version as being 80 and the child lock works. When does it change, do you restart Z2M?

Jeepmg, 2025-01-14:
I'm not sure that the Lock function lasts very long without jumping alone.

Otherwise, if you press the physical button of the socket several times quickly, the lock function jumps but remains active on the Z2M frontend.

GigiPompieru, 2025-01-15:
> I'm not sure that the Lock function lasts very long without jumping alone.
> 
> Otherwise, if you press the physical button of the socket several times quickly, the lock function jumps but remains active on the Z2M frontend.

tested, it stayed on for 24 hours.

aircooledbeemer, 2025-01-24:
Same here for 2 [Tuya TS011F_plug_1]:

    "update": {
        "installed_version": 192,
        "latest_version": 80,
        "state": "idle"
    },

Regards

Mangomc, 2025-04-05:
Same here. Anyone found a fix?

asychev, 2025-04-07:
Same for me, TS011F / _TZ3000_w0qqde0g
```
    "update": {
        "installed_version": 192,
        "latest_version": 80,
        "state": "idle"
    }
```

GigiPompieru, 2025-06-07:
now the firmware is 0

[image]

stefanuytterhoeven, 2025-06-19:
is there a solution for this? My aqara presence sensor p1 cannot be updated. (model:[RTCZCGQ11LM])
current firmware:
App: 0.0 build 00
Stack: 0.0 build 3a

available firmware:
App: 0.0 build 00
Stack: 0.0 build 36

"current firmware is newer than latest available " is the message.

3a > 36 ? 
in hex value it is. But i guess in version numbering 3a should be less than 36?

any progress ????

closed 2025-08-26 by github-actions[bot]

Matthy168, 2025-10-04:
any update - resolution on this??

Matthy168, 2025-10-04:
for me the same as as above:
    "child_lock": "LOCK",
    "countdown": 0,
    "current": 0.2,
    "energy": 301.85,
    "indicator_mode": "on",
    "last_seen": "2025-10-04T06:41:17+02:00",
    "linkquality": 39,
    "power": 28,
    "power_outage_memory": "on",
    "state": "ON",
    "update": {
        "installed_version": 192,
        "latest_version": 82,
        "state": "idle"

[image]
[image]

GigiPompieru, 2025-10-04:
somehow this seems fixed (well, except for one plug, which is quite strange since they are all the same) and this is how they show in the new GUI:

[image]