Koenkk/zigbee2mqtt #25723: Philips 324131092621 no more update "counter" Z2m 2.0

Bastiencc, 2025-01-10:
### What happened?

Philips 324131092621 no more update "counter"
```
{
    "action": "",
    "action_duration": null,
    "battery": 100,
    "brightness": 255,
    "counter": 2,
    "linkquality": 29,
    "update": {
        "installed_version": 1107324829,
        "latest_version": 1107324829,
        "state": "idle"
    },
    "update_available": false,
    "duration": 0
}
```

### What did you expect to happen?

Update counter value like in 1.4

### Zigbee2MQTT version

2.0

### Adapter firmware version

20240710

### Adapter

zzh

### Setup

addon on HA with rpi5

Vroume, 2025-01-17:
Same issue here : my counter is set to 3 and remains at this value no matter what.

On my setup, I think it started to behave this way before upgrading to z2m 2.0 (so it started in version 1.39.1-1) but during the modification of z2m configuration to prepare for 2.0 breaking changes.

Vroume, 2025-01-17:
It seems there was a specific code available in zigbee2mqtt when using "legacy" mode for this device : this legacy code was counting "press" messages received from device in a given amount of time.
This code is no more present (and was thus not used in previous version when disabling "legacy" mode).

=> Now, this counter function is deactivated/lost and in fact HA is receiving all the "press" events as a dedicated MQTT message ("double-click" on dimmer => 2 zigbee messages received by Z2M => 2 MQTT messages received by HA).
The "counter" value we see in events payload is probably just a remnant value, never updated anymore.

I guess we could do the counting in our automations or in some kind of intermediate automation, script, etc.
I don't know if it could be possible (and a good idea) to restore counter behavior, or to directly publish MQTT messages with sub-type, like "on_press_double", "on_press_triple", etc ?

JohnnyM84, 2025-01-26:
> It seems there was a specific code available in zigbee2mqtt when using "legacy" mode for this device : this legacy code was counting "press" messages received from device in a given amount of time. [...]

I have tried it with node red and the node-red-contrib-timed-counter, but the normal 2 times fast click for a double press does not work I get count 1. If I click 2 times slower then I get a count 2, but most of the time I get 1 because of the timing. So this is not working for me. 
Maybe direct with an automation in ha but have to look into this how to get that working. 
Hopefully legacy is coming back.

closed 2025-04-04 by github-actions[bot]