Koenkk/zigbee2mqtt #25725: [Wrong device]: Tuya TS0201

FredrikM97, 2025-01-10:
### Link

https://www.zigbee2mqtt.io/devices/TS0201.html#tuya-ts0201

### Model

ZTH01Y

### Description

Temperature & Humidity

### Vendor

Tuya

### Picture (link)

https://www.zigbee2mqtt.io/devices/ZTH01.html

### Database entry

{"id":26,"type":"EndDevice","ieeeAddr":"0xa4c138998aa48c04","nwkAddr":56219,"manufId":4098,"manufName":"_TZ3000_v1w2k9dd","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1,1026,1029],"outClusterList":[3,25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0201","manufacturerName":"_TZ3000_v1w2k9dd","powerSource":3,"zclVersion":3,"appVersion":90,"stackVersion":2,"hwVersion":1,"dateCode":""}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2360}},"msRelativeHumidity":{"attributes":{"measuredValue":4600}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":90,"stackVersion":2,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1736516736649}

flavioxavier, 2025-01-22:
Same here a ZTH01 gets recognized as TS0201

My db:
> ``` {"id":4,"type":"EndDevice","ieeeAddr":"0xa4c1387c837943c5","nwkAddr":42745,"manufId":4098,"manufName":"_TZ3000_v1w2k9dd","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1,1026,1029],"outClusterList":[3,25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0201","manufacturerName":"_TZ3000_v1w2k9dd","dateCode":"","zclVersion":3,"appVersion":90,"powerSource":3}},"msTemperatureMeasurement":{"attributes":{"measuredValue":1690}},"msRelativeHumidity":{"attributes":{"measuredValue":6500}},"genPowerCfg":{"attributes":{"batteryVoltage":25,"batteryPercentageRemaining":46}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":90,"stackVersion":2,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1737556644324} ```

eretromicin, 2025-01-27:
Same issue with 2 new units. Both ZTH01's both show up as TS0201

**Unit 1:**
`{"id":12,"type":"EndDevice","ieeeAddr":"0xa4c13856fa21cbf8","nwkAddr":48167,"manufId":4098,"manufName":"_TZ3000_v1w2k9dd","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1,1026,1029],"outClusterList":[3,25,10],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2250}},"msRelativeHumidity":{"attributes":{"measuredValue":4500}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}},"genBasic":{"attributes":{"65534":0,"hwVersion":1,"dateCode":"","manufacturerName":"_TZ3000_v1w2k9dd","zclVersion":3,"appVersion":90,"modelId":"TS0201","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":90,"stackVersion":2,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1737955416796}`
**Unit 2:**
`{"id":13,"type":"EndDevice","ieeeAddr":"0xa4c1380bb4323fe7","nwkAddr":191,"manufId":4098,"manufName":"_TZ3000_v1w2k9dd","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1,1026,1029],"outClusterList":[3,25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0201","manufacturerName":"_TZ3000_v1w2k9dd","powerSource":3,"zclVersion":3,"appVersion":90,"stackVersion":2,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":90,"stackVersion":2,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1737956509692}`

batmule, 2025-02-17:
My ZTH01's also are detected as TS0201

FREE2WIN2, 2025-02-17:
Also my ZTH01 is recognized as TS0601 with the manufacturerName '_TZE284_yjjdcqsq'.

GrasshopperNL, 2025-03-12:
Same issue here

FREE2WIN2, 2025-03-13:
In this case this could help: https://github.com/Koenkk/zigbee2mqtt/issues/26412

closed 2025-05-21 by github-actions[bot]