Koenkk/zigbee2mqtt #25729: ERS-10TZBVK-AA - HA event has not action step/transition/rate

DubhAd, 2025-01-10:
### What happened?

The old (legacy) HA sensors have details on the amount of rotation (step size, transition time, and rate) that aren't exposed with the new events.

### What did you expect to happen?

I expected this information to also be in the event.

### How to reproduce it (minimal and precise)

Enable experimental events and rotate the dial.

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20220726

### Adapter

Sonoff P

### Setup

Docker

### Debug log

N/A

mundschenk-at, 2025-01-10:
Can you show an example screenshot of such a state with attributes in the legacy sensors (via the HA developer tools) with 2.0.0 please?

DubhAd, 2025-01-13:
This is what the new event looks like:
[image]

mundschenk-at, 2025-01-13:
Sorry, I meant the legacy `sensor`. I know that the new `event` entity does not have any of those properties.

DubhAd, 2025-01-14:
Ah, yes... I don't have that any more as I disabled it all.

Fortunately it turns out I don't have to restart Z2M to re-enable that
[image]
[image]
[image]
[image]

and when I rotate it the logbook shows:
[image]

mundschenk-at, 2025-01-14:
Is that screenshot from Z2M 2.0.0? Because to me that looks like the attribute definitions from the now gone `homeassistant_legacy_entity_attributes` setting (which dumped the Z2M complete state of the device into each and every entity).

DubhAd, 2025-01-15:
As mentioned in the issue that's 1.42.0 - I haven't had time to update to 2.0.0

closed 2025-03-25 by github-actions[bot]