Koenkk/zigbee2mqtt #25730: [Wrong device]: Tuya ZTH02 detected as TS0201

dynasticorpheus, 2025-01-10:
### Link

https://nl.aliexpress.com/item/1005007662967612.html

### Model

ZTH02

### Description

Temperature and humidity sensor

### Vendor

Tuya

### Picture (link)

https://ae01.alicdn.com/kf/Se32322edee0a44eca20ec3bb12c4f5acV.jpg_640x640q90.jpg

### Database entry

{"id":6,"type":"EndDevice","ieeeAddr":"0xa4c138aee88fee06","nwkAddr":26130,"manufId":4098,"manufName":"_TZ3000_v1w2k9dd","powerSource":"Battery","modelId":"TS0201","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1,1026,1029],"outClusterList":[3,25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0201","manufacturerName":"_TZ3000_v1w2k9dd","powerSource":3,"zclVersion":3,"appVersion":86,"stackVersion":2,"hwVersion":1,"dateCode":""}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2170}},"msRelativeHumidity":{"attributes":{"measuredValue":5000}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":86,"stackVersion":2,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1736522339135}

### Notes

Tuya ZTH02 detected as TS0201

[image]

dynasticorpheus, 2025-01-10:
@Koenkk  Happy to create a PR but not sure if I should add to existing model [ZTH01/ZTH02](https://www.zigbee2mqtt.io/devices/ZTH01_ZTH02.html) or clone [TS0201](https://www.zigbee2mqtt.io/devices/TS0201.html#tuya-ts0201) towards new model ZY-ZTH02 ?

Koenkk, 2025-01-11:
@dynasticorpheus you can add it as a whitelabel

dynasticorpheus, 2025-01-11:
@Koenkk  Thanks, done [#8591](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8591)

closed 2025-01-12 by dynasticorpheus