Koenkk/zigbee2mqtt #25732: Update to 2.0.0 fails at git pull --no-rebase

GForce2010, 2025-01-10:
### What happened?

I tried following the instructions to upgrade to V2. 

I have it installed of a Debain VM.

I installed pnpm, moved my config file but when I try to run git pull --no-rebase after stopping Zigbee2MQTT it errors and lists a load of files it says will be overwritten.

deb@debian:/opt/zigbee2mqtt$ git pull --no-rebase
Updating ffc2ff1..060ae99
error: Your local changes to the following files would be overwritten by merge:
        .eslintignore
        .eslintrc.js
        .github/FUNDING.yml
        .github/ISSUE_TEMPLATE/config.yml
        .github/ISSUE_TEMPLATE/feature_request.yaml
        .github/ISSUE_TEMPLATE/new_device_support.yaml
        .github/ISSUE_TEMPLATE/problem_report.yaml
        .github/ISSUE_TEMPLATE/wrong_device.yaml
        .github/dependabot.yml
        .github/workflows/ci.yml
        .github/workflows/codeql.yaml
        .github/workflows/deps_review.yaml
        .github/workflows/ghcr_cleanup.yml
        .github/workflows/merge_master_to_dev.yml
        .github/workflows/release_please.yml
        .github/workflows/stale.yml
        .github/workflows/update_dep.yml
        .github/workflows/update_deps.yml
        .gitignore
        .release-please-manifest.json
        CHANGELOG.md
        CODE_OF_CONDUCT.md
        CONTRIBUTING.md
        README.md
        babel.config.js
        docker/Dockerfile
        docker/docker-compose.yml
        index.js
        lib/controller.ts
        lib/eventBus.ts
        lib/extension/availability.ts
        lib/extension/bridge.ts
        lib/extension/configure.ts
        lib/extension/extension.ts
        lib/extension/externalConverters.ts
        lib/extension/externalExtension.ts
        lib/extension/frontend.ts
        lib/extension/groups.ts
        lib/extension/homeassistant.ts
        lib/extension/legacy/bridgeLegacy.ts
        lib/extension/legacy/deviceGroupMembership.ts
        lib/extension/legacy/report.ts
        lib/extension/legacy/softReset.ts
        lib/extension/networkMap.ts
        lib/extension/onEvent.ts
        lib/extension/otaUpdate.ts
        lib/extension/publish.ts
        lib/model/device.ts
        lib/model/group.ts
        lib/mqtt.ts
        lib/state.ts
        lib/types/mkdir-recursive.ts
        lib/types/types.d.ts
        lib/types/zigbee2mqtt-frontend.d.ts
        lib/util/data.ts
        lib/util/logger.ts
        lib/util/settings.schema.json
        lib/util/settings.ts
        lib/util/utils.ts
        lib/util/yaml.ts
        lib/zigbee.ts
        package.json
        release-please-config.json
        scripts/generateChangelog.js
        scripts/install.sh
        scripts/testExternalConverter.js
        scripts/zStackEraseAllNvMem.js
        test/assets/exampleExtension.js
        test/assets/mock-external-converter-multiple.js
        test/assets/mock-external-converter.js
        test/availability.test.js
        test/bind.test.js
        test/bridge.test.js
        test/configure.test.js
        test/controller.test.js
        test/data.test.js
        test/externalConverters.test.js
        test/externalExtension.test.js
        test/frontend.test.js
        test/group.test.js
        test/homeassistant.test.js
        test/legacy/bridgeLegacy.test.js
        test/legacy/report.test.js
        test/lib/flushPromises.js
        test/logger.test.js
        test/networkMap.test.js
        test/onEvent.test.js
        test/otaUpdate.test.js
        test/publish.test.js
        test/settings.test.js
        test/stub/data.js
        test/stub/logger.js
        test/stub/mqtt.js
        test/stub/sleep.js
        test/stub/zigbeeHerdsman.js
        test/utils.test.js
        tsconfig.json
Please commit your changes or stash them before you merge.
Aborting

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

Texas Instruments LAUNCHXL-CC26X2R1

### Setup

Virtual Debian machine running on Proxmox.

lux73, 2025-01-10:
https://github.com/Koenkk/zigbee2mqtt/issues/19286

GForce2010, 2025-01-11:
After a bit of googling I found a solution.

I ran `git reset --hard HEAD`

The pull command and update script then worked as shown and I am now on V2.