Koenkk/zigbee2mqtt #25736: ElectSmart PRO-900Z needs hvac_action based on "valve_state"

Nick-Sills, 2025-01-10:
### What happened?

Apologies, I may have opened this issue in the wrong repo previously:
https://github.com/Koenkk/zigbee2mqtt/issues/25706

The thermostat card in Home Assistant expects an `hvac_action` of `heating` or `idle` to visually implement the current operational state of a thermostat. 

ElectSmart PRO-900Z implements a `valve_state` binary sensor with an `open` or `closed` state. This should be used to provide the MQTT `running_state` state  of `heat` when `closed` and `hvac_mode` is `heat`, otherwise `idle` 

See below:
This should be `idle` with a non active climate icon as the set temperature is below the current temperature:
[image]

https://www.zigbee2mqtt.io/devices/PRO-900Z.html#electsmart-pro-900z

### What did you expect to happen?

I expected the thermostat state to change to `Heating` when the set temperature is above the current temperature, and `idle` when the set temp is below the hysteresis temperature or the thermostat is `off`. 

### How to reproduce it (minimal and precise)

Change the set temp to below or above the current temperature using the Home Assistant climate card. 

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20210708

### Adapter

SONOFF Universal Zigbee 3.0 USB Dongle

### Setup

Docker - Ras Pi 3B+

Nick-Sills, 2025-01-10:
PR created:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/8588
https://github.com/Koenkk/zigbee2mqtt.io/pull/3417

Tested and working correctly.

Heating:
[image]
[image]
Idle:
[image]
[image]

Nick-Sills, 2025-01-10:
Apologies, this issue is duplicated: https://github.com/Koenkk/zigbee2mqtt/issues/25706

nickeeey, 2025-02-09:
hi, an other Problem for me, was that there is no sensor roomtemp readout, that can be fixed by yourself, if you need it for some programms like betterthermostat.

My Problem with that Device is ATM, that it is very chatty and send a info mqtt toppic every 1-5sec. I want to reduce it, to have less traffic on my zigbee network. Do you know any tricks?

closed 2025-04-19 by github-actions[bot]