Koenkk/zigbee2mqtt #25738: Z2M crashes and does not restart if a tcp adapter is used

Hull9k, 2025-01-10:
### What happened?

If the network connection between the Home assistant server and the SLS gateway Zigbee bridge is lost for a while, then Z2M crashes and does not restart automatically. If there are no problems with the connection, then Z2M works fine.

### What did you expect to happen?

Z2M should restart automatically after the connection is restored. Since watchdog is enabled in both Z2M and Home assistant.

### How to reproduce it (minimal and precise)

Disconnect the network connection with the adapter approx for 10-20 minutes.

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20240710

### Adapter

EBYTE E72-2G4M20S1E module, which is connected to an SLS gateway that operates in Zigbee bridge mode

### Setup

Addon on Home Assistant OS on VMware Workstation virtual machine

### Debug log

[2025-01-10 19:36:47] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-01-10 19:36:47] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-01-10 19:36:47] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-01-10 19:36:47] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-01-10 19:36:47] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-10 19:36:47] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,62,248]
[2025-01-10 19:36:47] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,62,248]
[2025-01-10 19:36:47] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,62] - 248
[2025-01-10 19:36:47] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":62}
[2025-01-10 19:36:47] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-10 19:37:13] error: 	zh:zstack:znp: Socket error Error: read ETIMEDOUT
[2025-01-10 19:37:13] info: 	zh:zstack:znp: Port closed
[2025-01-10 19:37:13] debug: 	zh:controller: Adapter disconnected
[2025-01-10 19:37:13] info: 	zh:zstack:znp: closing
[2025-01-10 19:37:13] error: 	z2m: Adapter disconnected, stopping
[2025-01-10 19:37:13] debug: 	z2m: Saving state to file /config/zigbee2mqtt_gar/state.json
[2025-01-10 19:37:13] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt_gar/bridge/state', payload '{"state":"offline"}'
[2025-01-10 19:37:13] info: 	z2m: Disconnecting from MQTT server
[2025-01-10 19:37:13] info: 	z2m: Stopping zigbee-herdsman...
[2025-01-10 19:37:13] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt_gar/database.db'
[2025-01-10 19:37:13] info: 	z2m: Stopped zigbee-herdsman
[2025-01-10 19:37:13] info: 	z2m: Stopped Zigbee2MQTT
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
[2025-01-10 19:37:15] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt_gar/log/2025-01-10.14-04-48'
[2025-01-10 19:37:15] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-10 19:37:15] debug: 	z2m: Loaded state from file /config/zigbee2mqtt_gar/state.json
[2025-01-10 19:37:15] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-10 19:37:15] debug: 	z2m: sd-notify loaded
[2025-01-10 19:37:15] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-10 19:37:15] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":44236,\"extendedPanID\":[47,136,137,143,55,49,6,140],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt_gar/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt_gar/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt_gar/coordinator_backup.json\",\"serialPort\":{\"path\":\"tcp://192.168.3.2:8881\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false,\"transmitPower\":20}}"'
[2025-01-10 19:37:15] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":44236,"extendedPanID":[47,136,137,143,55,49,6,140],"channelList":[11]},"serialPort":{"path":"tcp://192.168.3.2:8881","adapter":"zstack"},"adapter":{"disableLED":false,"transmitPower":20},"databasePath":"/config/zigbee2mqtt_gar/database.db","databaseBackupPath":"/config/zigbee2mqtt_gar/database.db.backup","backupPath":"/config/zigbee2mqtt_gar/coordinator_backup.json"}'
[2025-01-10 19:37:15] info: 	zh:zstack:znp: Opening TCP socket with 192.168.3.2:8881
[2025-01-10 19:38:52] error: 	z2m: Not connected to MQTT server!
[2025-01-10 19:38:52] error: 	z2m: Cannot send message: topic: 'zigbee2mqtt_gar/IR-sensor', payload: '{"battery":1,"battery_low":false,"keep_time":30,"last_seen":"2025-01-10T16:36:47.549Z","linkquality":127,"occupancy":false,"sensitivity":"medium","tamper":false,"voltage":2500}
[2025-01-10 19:39:30] error: 	zh:zstack:znp: Socket error Error: connect ETIMEDOUT 192.168.3.2:8881
[2025-01-10 19:39:30] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-10 19:39:30] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-10 19:39:30] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-10 19:39:30] error: 	z2m: Exiting...
[... 89 lines cut ...]
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/.pnpm/winston@3.17.0/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:198:25)
    at Logger.info (/app/lib/util/logger.ts:211:14)
    at Znp.onPortClose (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:88:16)
    at Object.onceWrapper (node:events:633:26)
    at Socket.emit (node:events:518:28)
    at TCP.<anonymous> (node:net:343:12)
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/.pnpm/winston@3.17.0/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:198:25)
    at Logger.info (/app/lib/util/logger.ts:211:14)
    at Znp.onPortClose (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:88:16)
    at Object.onceWrapper (node:events:633:26)
    at Socket.emit (node:events:518:28)
    at TCP.<anonymous> (node:net:343:12)

_----------------------------
After that, nothing happens._

causefx, 2025-01-10:
i have the same error.

ESP4Ever, 2025-01-11:
Hi!

I have a similar problem with ESPHomeD ZB-GW03 gateway causing random z2m restarts but until now I seen no permanent restart failure.

My setup:
```
> HAOS on RPI5
> Core 2025.1.2
> Supervisor 2024.12.3
> Operating System 14.1
> Frontend 20250109.0

> Zigbee2MQTT Current version: 2.0.0-2
> Coordinator type EmberZNet
> Coordinator revision 7.4.3 [GA]
> Frontend version 0.9.4
> zigbee-herdsman-converters version 21.12.0
> zigbee-herdsman version 3.2.1
```

z2m add-on log:

```
.........
[12:05:41] INFO: Preparing to start...
[12:05:41] INFO: Socat not enabled
[12:05:41] INFO: Enabled Zigbee2MQTT watchdog with value 'default'
[12:05:41] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
WATCHDOG: Waiting 0.03333333333333333min before next start try.
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
Error: Unsupported response reading for cluster ID '260'.
    at Function.readResponse (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:1581:23)
    at EmberAdapter.onZDOResponse (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:514:36)
    at Ezsp.emit (node:events:518:28)
    at Ezsp.ezspIncomingMessageHandler (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5323:18)
    at Ezsp.callbackDispatch (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:780:26)
    at Ezsp.onAshFrame (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:396:22)
    at UartAsh.emit (node:events:518:28)
    at Immediate.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1171:45)
    at processImmediate (node:internal/timers:491:21)
[16:53:35] INFO: Preparing to start...
[16:53:35] INFO: Socat not enabled
[16:53:35] INFO: Enabled Zigbee2MQTT watchdog with value 'default'
[16:53:35] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
Error: Unsupported response reading for cluster ID '260'.
    at Function.readResponse (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:1581:23)
    at EmberAdapter.onZDOResponse (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:514:36)
    at Ezsp.emit (node:events:518:28)
    at Ezsp.ezspIncomingMessageHandler (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5323:18)
    at Ezsp.callbackDispatch (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:780:26)
    at Ezsp.onAshFrame (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:396:22)
    at UartAsh.emit (node:events:518:28)
    at Immediate.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1171:45)
    at processImmediate (node:internal/timers:491:21)
[17:12:47] INFO: Preparing to start...
[17:12:47] INFO: Socat not enabled
[17:12:47] INFO: Enabled Zigbee2MQTT watchdog with value 'default'
[17:12:47] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT with watchdog (2000,60000,300000,900000,1800000,3600000).
Error: Unsupported response reading for cluster ID '260'.
[... 307 lines cut ...]
[2025-01-11 08:01:58] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-01-11 08:01:58] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-01-11 08:01:58] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-01-11 08:01:58] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-01-11 08:01:58] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-01-11 08:01:58] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-01-11 08:01:58] info: 	zh:ember:uart:ash:   Retry dupes     : RX=7
[2025-01-11 08:01:58] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-01-11 08:01:58] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-01-11 08:01:58] debug: 	zh:ember:ezsp: <=x= [CBFRAME: ID=42:"CLEAR_BINDING_TABLE" Seq=66 Len=3] Invalid, status=ERROR_TRUNCATED.
[2025-01-11 08:01:58] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2025-01-11 08:01:58] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2025-01-11 08:01:58] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2025-01-11 08:01:58] error: 	z2m: Adapter disconnected, stopping
[2025-01-11 08:01:58] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json
[2025-01-11 08:01:58] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-01-11 08:01:58] info: 	z2m: Disconnecting from MQTT server
[2025-01-11 08:01:58] info: 	z2m: Stopping zigbee-herdsman...
[2025-01-11 08:01:58] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2025-01-11 08:01:58] info: 	z2m: Stopped zigbee-herdsman
[2025-01-11 08:01:58] info: 	z2m: Stopped Zigbee2MQTT

```

At the same time there is nothing in ESPHome logs

etiausegg, 2025-01-11:
Did you read 
https://github.com/Koenkk/zigbee2mqtt/discussions/24198 and
https://github.com/Koenkk/zigbee2mqtt/discussions/24364

probably you have to add this to your configuration:

serial:
  adapter: zstack # <-- Add this line
  port: /dev/ttyACM0

Hull9k, 2025-01-11:
@etiausegg, Yes, this is done in my configuration.
I will add that I use two copies of Z2M. The second copy works with a USB zstack adapter and it works fine.

closed 2025-03-27 by github-actions[bot]

gmirz, 2025-11-28:
I have same problem