Koenkk/zigbee2mqtt #25741: Aqara Knob H1 (Wireless) ZNXNKG02LM - Battery values not available

TDCroPower, 2025-01-10:
### What happened?

Unfortunately, no battery values are currently output
[image]

### What did you expect to happen?

The Battery and Voltage read out values contain

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

7.4.5 [GA]

### Adapter

SMLIGHT SLZB-06M (Core: v2.6.8.dev26) - ember

### Setup

Z2M: 2.0.0 as Docker Container Host: Raspberry Pi 4 8GB with DietPi v9.8.0 (Bookworm)

hrustiks, 2025-01-16:
The same problem
[image]

m0t0rh3ad, 2025-04-24:
I've got the smae issues with same device

hrustiks, 2025-04-27:
I've got the same to!