Koenkk/zigbee2mqtt #25748: failed (value.slice is not a function) at BuffaloZdo.writeIeeeAddr

pawelzny, 2025-01-11:
### What happened?

When trying to join PIR sensor I've got 3 failed attempts with errors pointing to code issue. After all the PIR sensor joined the network, but it does not report it's state to the network. I tried 2 brand new sensors and both had the same issue.

I have third sensor that was added to the network before zigbee2mqtt 2.0.0 and it works fine.

```
2025-01-11 13:55:26z2m: Failed to configure '0xa4c1389f5fd46d60', attempt 1 (TypeError: Bind 0xa4c1389f5fd46d60/1 genPowerCfg from '28/1' failed (value.slice is not a function) at BuffaloZdo.writeIeeeAddr (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:238:41) at BuffaloZdo.buildBindRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:1035:22) at Function.buildRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:796:32) at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:550:44) at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/reporting.ts:42:24) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/devices/tuya.ts:1823:29) at Configure.configure (/app/lib/extension/configure.ts:124:37) at /app/lib/extension/configure.ts:81:62 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:23) at EventEmitter.emit (node:events:530:35))
2025-01-11 13:55:32z2m: Failed to configure '0xa4c1389f5fd46d60', attempt 2 (TypeError: Bind 0xa4c1389f5fd46d60/1 genPowerCfg from '28/1' failed (value.slice is not a function) at BuffaloZdo.writeIeeeAddr (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:238:41) at BuffaloZdo.buildBindRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:1035:22) at Function.buildRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:796:32) at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:550:44) at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/reporting.ts:42:24) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/devices/tuya.ts:1823:29) at Configure.configure (/app/lib/extension/configure.ts:124:37) at /app/lib/extension/configure.ts:82:62 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:23) at EventEmitter.emit (node:events:530:35))
2025-01-11 13:55:32z2m: Failed to configure '0xa4c1389f5fd46d60', attempt 3 (TypeError: Bind 0xa4c1389f5fd46d60/1 genPowerCfg from '28/1' failed (value.slice is not a function) at BuffaloZdo.writeIeeeAddr (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:238:41) at BuffaloZdo.buildBindRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:1035:22) at Function.buildRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/zspec/zdo/buffaloZdo.ts:796:32) at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:550:44) at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/reporting.ts:42:24) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/devices/tuya.ts:1823:29) at Configure.configure (/app/lib/extension/configure.ts:124:37) at /app/lib/extension/configure.ts:82:62 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:23) at EventEmitter.emit (node:events:530:35))
```

Device:

- Zigbee Model: TS0202
- Zigbee Manufacturer: _TZ3000_lf56vpxj
- Description: Motion sensor
- Support status: Supported
- IEEE Address: 0xa4c1389f5fd46d60
- Network address: 0xA4FB / 42235
- Manufacturer: [Tuya](https://www.zigbee2mqtt.io/supported-devices/#v=Tuya)
- Model: [ZP01](https://www.zigbee2mqtt.io/devices/ZP01.html#tuya-zp01)
- Power: Battery OK
- Interview completed: True

The device state always shows:

```
{
    "battery_low": false,
    "linkquality": 255,
    "occupancy": false,
    "tamper": false
}
```

### How to reproduce it (minimal and precise)

1. Enable "Permit join (All)
2. Press the physical "reset" button on the device
3. Wait for device interview

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

0x26720700

### Adapter

deconz

### Setup

Add-on on Home Assistant OS

jpagand, 2025-01-29:
Exact same issue here but with Zigbee2MQTT version 2.0.0-2

Koenkk, 2025-02-14:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

gintsmurans, 2025-02-14:
Here is the log file. After restarting zigbee2mqtt, it seems that I can call reconfigure device if I quickly press the button on the sensor. And also I am not getting same error I was getting. If I don't press the button on the sensor it just says it timed out. But it also never changes Occupancy state - its always clear.

[image]

[log.log](https://github.com/user-attachments/files/18805740/log.log)

Koenkk, 2025-02-17:
@gintsmurans I guess this is a different issue, try to force remove the device, restart z2m and re-pair the device.

closed 2025-04-27 by github-actions[bot]

pawelzny, 2025-06-10:
@Koenkk @gintsmurans Upgrade to version 2.4.0 fixed the issue. The PIR device has to be deleted first and rejoin again. Below is the log and the last 5 rows shows the occupancy reacts to motion detection.

```
[2025-06-10 16:59:33] z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c1389f5fd46d60/occupancy/config', payload ''
[2025-06-10 16:59:33] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0xa4c1389f5fd46d60"},"status":"ok","transaction":"u60c8-1"}'
[2025-06-10 16:59:33] z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c1389f5fd46d60/battery_low/config', payload ''
[2025-06-10 16:59:33] z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c1389f5fd46d60/battery/config', payload ''
[2025-06-10 16:59:33] z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c1389f5fd46d60/voltage/config', payload ''
[2025-06-10 16:59:33] z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c1389f5fd46d60/tamper/config', payload ''
[2025-06-10 16:59:33] z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c1389f5fd46d60/linkquality/config', payload ''
[2025-06-10 16:59:56] zh:controller: Interview for '0xa4c1389f5fd46d60' started
[2025-06-10 16:59:56] z2m: Device '0xa4c1389f5fd46d60' joined
[2025-06-10 16:59:56] z2m: Starting interview of '0xa4c1389f5fd46d60'
[2025-06-10 16:59:56] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1389f5fd46d60","ieee_address":"0xa4c1389f5fd46d60"},"type":"device_joined"}'
[2025-06-10 16:59:56] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1389f5fd46d60","ieee_address":"0xa4c1389f5fd46d60","status":"started"},"type":"device_interview"}'
[2025-06-10 16:59:59] zh:controller: Succesfully interviewed '0xa4c1389f5fd46d60'
[2025-06-10 16:59:59] z2m: Successfully interviewed '0xa4c1389f5fd46d60', device has successfully been paired
[2025-06-10 16:59:59] z2m: Device '0xa4c1389f5fd46d60' is supported, identified as: Tuya Motion sensor (ZP01)
[2025-06-10 16:59:59] z2m: Configuring '0xa4c1389f5fd46d60'
[2025-06-10 16:59:59] z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c1389f5fd46d60/occupancy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c1389f5fd46d60/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c1389f5fd46d60"],"manufacturer":"Tuya","model":"Motion sensor","model_id":"ZP01","name":"0xa4c1389f5fd46d60","via_device":"zigbee2mqtt_bridge_0x00212effff09ae49"},"device_class":"occupancy","object_id":"0xa4c1389f5fd46d60_occupancy","origin":{"name":"Zigbee2MQTT","sw":"2.4.0","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xa4c1389f5fd46d60","unique_id":"0xa4c1389f5fd46d60_occupancy_zigbee2mqtt","value_template":"{{ value_json.occupancy }}"}'
[2025-06-10 16:59:59] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Motion sensor","exposes":[{"access":1,"description":"Indicates whether the device detected occupancy","label":"Occupancy","name":"occupancy","property":"occupancy","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","description":"Indicates if the battery of this device is almost empty","label":"Battery low","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","description":"Remaining battery in %, can take up to 24 hours before reported","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"category":"diagnostic","description":"Voltage of the battery in millivolts","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"description":"Indicates whether the device is tampered","label":"Tamper","name":"tamper","property":"tamper","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZP01","options":[],"supports_ota":false,"vendor":"Tuya"},"friendly_name":"0xa4c1389f5fd46d60","ieee_address":"0xa4c1389f5fd46d60","status":"successful","supported":true},"type":"device_interview"}'
[2025-06-10 16:59:59] z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c1389f5fd46d60/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c1389f5fd46d60/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c1389f5fd46d60"],"manufacturer":"Tuya","model":"Motion sensor","model_id":"ZP01","name":"0xa4c1389f5fd46d60","via_device":"zigbee2mqtt_bridge_0x00212effff09ae49"},"device_class":"battery","entity_category":"diagnostic","object_id":"0xa4c1389f5fd46d60_battery_low","origin":{"name":"Zigbee2MQTT","sw":"2.4.0","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xa4c1389f5fd46d60","unique_id":"0xa4c1389f [... cut ...]
[2025-06-10 17:00:28] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1389f5fd46d60', payload '{"battery":100,"battery_low":false,"linkquality":255,"occupancy":false,"tamper":false,"voltage":3000}'
[2025-06-10 17:00:38] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1389f5fd46d60', payload '{"battery":100,"battery_low":false,"linkquality":255,"occupancy":true,"tamper":false,"voltage":3000}'
[2025-06-10 17:01:13] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1389f5fd46d60', payload '{"battery":100,"battery_low":false,"linkquality":255,"occupancy":false,"tamper":false,"voltage":3000}'
```