Koenkk/zigbee2mqtt #25753: [New device support]: Tuya TS0601_temperature_humidity_sensor_2 _TZE204_jygvp6fk

peterfido, 2025-01-11:
### Link

https://www.emos-shop.at/gosmart-digitales-drathloses-thermometer-egs0101-zigbee

### Database entry

{"id":28,"type":"EndDevice","ieeeAddr":"0xa4c138b8b21c0b91","nwkAddr":33479,"manufId":4417,"manufName":"_TZE204_jygvp6fk","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_jygvp6fk","powerSource":3,"zclVersion":3,"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1013852590},"lastSeen":1736616314856}

### Zigbee2MQTT version

iobroker v1.10.14 and v1.11.2

### Comments

The device is runnin under others like this one.

i have added three lines of code in this area:

1.         fingerprint: tuya.fingerprint('TS0601', [
2.             '_TZE200_yjjdcqsq',
3.             '_TZE200_9yapgbuv',
4.             '_TZE200_utkemkbs',
5.             '_TZE204_utkemkbs',
6.             '_TZE204_9yapgbuv',
7.             '_TZE204_upagmta9',
8.             '_TZE200_cirvgep4',
9.             '_TZE200_upagmta9',
10.             '_TZE204_yjjdcqsq',
11.             '_TZE204_cirvgep4',
12. 			'_TZE200_jygvp6fk',
13. 			'_TZE204_jygvp6fk',
14.         ]),
15.         model: 'TS0601_temperature_humidity_sensor_2',
16.         vendor: 'Tuya',
17.         description: 'Temperature and humidity sensor',
18.         fromZigbee: [tuya.fz.datapoints],
19.         toZigbee: [tuya.tz.datapoints],
20.         onEvent: tuya.onEvent({ queryOnDeviceAnnounce: true }),
21.         configure: async (device, coordinatorEndpoint) => {
22.             await tuya.configureMagicPacket(device, coordinatorEndpoint);
23.             // Required to get the device to start reporting
24.             await device.getEndpoint(1).command('manuSpecificTuya', 'dataQuery', {});
25.         },
26.         exposes: [e.temperature(), e.humidity(), tuya.exposes.batteryState(), tuya.exposes.temperatureUnit()],
27.         meta: {
28.             tuyaDatapoints: [
29.                 [1, 'temperature', tuya.valueConverter.divideBy10],
30.                 [2, 'humidity', tuya.valueConverter.raw],
31.                 [3, 'battery_state', tuya.valueConverter.batteryState],
32.                 [9, 'temperature_unit', tuya.valueConverter.temperatureUnitEnum],
33.             ],
34.         },
35.         whiteLabel: [
36.             tuya.whitelabel('Tuya', 'ZTH01', 'Temperature and humidity sensor', ['_TZE200_yjjdcqsq', '_TZE204_yjjdcqsq']),
37.             tuya.whitelabel('Tuya', 'SZTH02', 'Temperature and humidity sensor', ['_TZE200_utkemkbs', '_TZE204_utkemkbs']),
38.             tuya.whitelabel('Tuya', 'ZTH02', 'Temperature and humidity sensor', ['_TZE200_9yapgbuv', '_TZE204_9yapgbuv']),
39.             tuya.whitelabel('Tuya', 'ZTH05', 'Temperature and humidity sensor', ['_TZE204_upagmta9', '_TZE200_upagmta9']),
40.             tuya.whitelabel('Tuya', 'ZTH08-E', 'Temperature and humidity sensor', ['_TZE200_cirvgep4', '_TZE204_cirvgep4']),
41. 			tuya.whitelabel('Tuya', 'EGS0101', 'Temperature and humidity sensor', ['_TZE200_jygvp6fk', '_TZE204_jygvp6fk']),
42.         ],
43. 

the lines are 12,13 and 41.

So, the device is running fine.

### External definition

```shell
i don´t have one.
```

### What does/doesn't work with the external definition?

My changes above make the device working.

Koenkk, 2025-01-12:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2025-01-12 by Koenkk

peterfido, 2025-01-13:
Thank you very much.